openwrt/packages #19166: tang: Unable to infer signing algorithm

nilsmeyer, 2022-08-14:
Maintainer: @neheb 
Environment: OpenWRT 22.03.0-rc5, x86_64 

Description:
When using tang on 22.03.0-rc5 it seems that encrypting / signing no longer works as there are no algorithms available, with files generated on another machine I get the following error in the tangd log:

> <unknown> GET /adv../lib/jws.c:49:JOSE_CFG_ERR_ALG_NOINFER:Unable to infer signing algorithm
Error trying to jose_jws_sign
 => 404 (../src/tangd.c:64)

In addition, it seems jose no longer lists any supported algorithms, calling `jose alg` returns nothing (on a regular Ubuntu system this returns a list of algorithms), the problem may be related to this.

jilroy, 2022-09-14:
**Environment**
OpenWrt 22.03.0, platform armv7l
Package: jose Version: 11-3
Package: tang Version: 11-1

**Description**
I have the same issue. Tang does not work anymore.

> /usr/libexec/tangd-keygen  /usr/share/tang/db/
> JWK generation failed!

because:

> jose jwk gen -i '{"alg":"ES512"}'
> JWK generation failed!

Also, jose alg gives no output contrary to what I get on a test Debian system for comparison.
I tried and strace on jose with no obvious error to me in the output.

Can someone have a look at it ? @neheb ?

neheb, 2022-09-17:
If I had to guess, the issue is libjose being built statically. jose uses

```
static void __attribute__((constructor))
```

which might not work as a static library

neheb, 2022-09-19:
Pushed and backported to 22.03.

closed 2022-09-19 by neheb

jilroy, 2022-09-20:
**Environment**
OpenWrt 22.03.0, platform armv7l
Package: jose Version: 11-4
Package: tang Version: 11-1

**Description**
Jose is fixed but tang is still not working.

First, I think the /etc/xinetd.d/tangdx has to be modified with this line since there is no cache directory anymore:
`server_args     = /usr/share/tang/db`

Secondly, even after modifying this file, tang is still not working. I get this error in /var/log/tangd.log:
`<unknown> GET /adv../lib/jws.c:49:JOSE_CFG_ERR_ALG_NOINFER:Unable to infer signing algorithm`

I did an ldd on /usr/libexec/tangd and found that there is no link to libcrypto nor to libz. @neheb , could it be the same issue with tang as it was with jose ?

neheb, 2022-09-21:
Probably just needs a PKG_RELEASE bump.

neheb, 2022-09-22:
Yeah...static library is a bad idea. Reverted and pushed.

jilroy, 2022-09-24:
Thanks. Tang works now.