openwrt/packages #19206: iperf3 service script

slogen, 2022-08-18:
Maintainer: @jamesmacwhite
Environment: OpenWrt 22.03-rc6 (all modelse)

Description:
It would be nice to have a "service" (/etc/init.d) script for automatically starting iperf3.

Here is a simple proposal. It could, of course, be nice with more uci-options and a luci-app-iperf3. Unfortunately I can't really offer to do a pull-request, but at least here is some simple inspiration.

/etc/init.d/iperf3:
```
#!/bin/sh /etc/rc.common

START=50
USE_PROCD=1

NAME=usteer
PROG=/usr/bin/iperf3

service_triggers() {
        procd_add_reload_trigger iperf3
}

start_service()
{
        ENABLED="$(uci -q get iperf3.@iperf3[-1].enabled)"
        ENABLED="${ENABLED:-0}"

        [ "$ENABLED" -gt 0 ] || return
        procd_open_instance iperf3
        procd_set_param command "$PROG"
        procd_append_param command -s
        procd_append_param command -i 0
        #procd_add_jail iperf3 ronly requirejail
        procd_set_param stdout 1
        procd_set_param stderr 1
        procd_set_param respawn
        procd_close_instance
}
```

feel free to simply close the issue as "wont-fix" :)

stangri, 2022-08-18:
https://github.com/openwrt/packages/pull/16282

slogen, 2022-08-22:
Superceded by #16282, which seems ready to merge and with lots more features (and a lot more complicated, nice that someone is willing to put in all that effort!)

closed 2022-08-22 by slogen