openwrt/packages #19210: stress-ng miss the dependency for libjpeg

BKPepe, 2022-08-19:
Maintainer: @commodo 
Environment: Turris MOX, mvebu/cortexa53, OpenWrt 22.03

Description:

While running full build, I noticed that the package stress-ng can not be compiled due to this error

```
Package stress-ng is missing dependencies for the following libraries:
libjpeg.so.62
```

I can provide more details if necessary.

commodo, 2022-08-19:
wow, that got added in April (recently)

i guess we'll add libjpeg to the dep list of stress-ng

will try to do this in the next couple of thanks

thanks for reporting this :)

commodo, 2022-08-23:
see https://github.com/openwrt/packages/pull/19235

dreirund, 2022-08-23:
I see the same problem in branch 22.03, latest `make targetclean; git pull && ./scripts/feeds update -a && ./scripts/feeds install -a` 2022-08-23, ca. 08:15 UTC:

```
Package stress-ng is missing dependencies for the following libraries:
libjpeg.so.62
```

on [&rarr;&hairsp;this `.config`](http://ix.io/48ok):  
* `grep -i stress .config`:  
  ```
  CONFIG_PACKAGE_stress=m
  CONFIG_PACKAGE_stress-ng=m
  ```
* grep -i libjpeg .config:  
  ```
  CONFIG_PACKAGE_libjpeg-turbo=y
  CONFIG_PACKAGE_libjpeg-turbo-utils=m
  CONFIG_LCD4LINUX_CUSTOM_NEEDS_libjpeg=y
  ```

*@commodo wrote:*  
> *see #19235*

I can confirm that manually applying the change from [this diff](https://github.com/openwrt/packages/pull/19235/files#diff-ede4dd7aef3461d6312fe2a49858a8d272269e781213194c817f8f40770b21eb) solves the reported problem for me.

closed 2022-08-23 by BKPepe

commodo, 2022-08-24:
Thanks @dreirund 
Was not sure whether this needs to be in 22.03 as well.

Will backport.

commodo, 2022-08-24:
Ah, nevermind. I see it was backported already

Thanks @BKPepe