openwrt/packages #19222: collectd-mod-cpufreq: ExtraItems does not seem to collect any data

fantom-x, 2022-08-22:
Maintainer: @hnyman 
Environment: 22.04-SNAPSHOT, x86-64

Setting ExtraItems to 1 does not generate any data and the UI displays what seems to be broken links.

hnyman, 2022-08-22:
Works for me.
To my knowledge there has been no change regarding that for years.
The "extra items" is not about collecting more data, but about showing it in LuCI.

[image]

fantom-x, 2022-08-22:
THis could be why:
```
cpufreq plugin: File /sys/devices/system/cpu/cpu0/cpufreq/stats/time_in_state not exists or no access. P-State statistics will not be reported. Check if `cpufreq-stats' kernel module is loaded.
```

kofec, 2022-08-22:
Is it possible to enable this feature for all raspberry ?

fantom-x, 2022-08-22:
And here is the explanation: https://github.com/openwrt/openwrt/issues/7620

hnyman, 2022-08-22:
ipq806x (like my R7800) has that CONFIG_CPU_FREQ_STAT enabled in kernel config:

```
  73 CONFIG_CPU_FREQ=y
  74 CONFIG_CPU_FREQ_DEFAULT_GOV_ONDEMAND=y
  75 # CONFIG_CPU_FREQ_DEFAULT_GOV_PERFORMANCE is not set
  76 CONFIG_CPU_FREQ_GOV_ATTR_SET=y
  77 CONFIG_CPU_FREQ_GOV_COMMON=y
  78 # CONFIG_CPU_FREQ_GOV_CONSERVATIVE is not set
  79 CONFIG_CPU_FREQ_GOV_ONDEMAND=y
  80 CONFIG_CPU_FREQ_GOV_PERFORMANCE=y
  81 # CONFIG_CPU_FREQ_GOV_POWERSAVE is not set
  82 CONFIG_CPU_FREQ_GOV_SCHEDUTIL=y
  83 # CONFIG_CPU_FREQ_GOV_USERSPACE is not set
  84 CONFIG_CPU_FREQ_STAT=y
```

But x86_64 does not ???

But I will close this issue, as it is apparently not about a bug in the collectd package or LuCI (that actually defines and uses the ExtraItems options), but about the kernel config that is used for x86.

closed 2022-08-22 by hnyman

fantom-x, 2022-08-24:
It is all enabled for x86_64. 

It looks like `intel_pstate` driver is used instead of the old `cpufreq` and the decision is made at runtime based on CPU capabilities. I suspect that if I disable pstates in BIOS, this collectd plugin will work again. If that is the case, then this plugin is permanently half-broken on x86_64, because they all come with pstates enabled by default.

https://forums.fedoraforum.org/showthread.php?294933-CPU-Governors-where-is-ONDEMAND

fantom-x, 2022-08-24:
If I add `intel_pstate=disable` to `/boot/grub/grub.cfg` then I get the old behaviour back. 

The collectd data collection plugin assumes that the old `cpufreq` driver is used, while it could be `intel_pstate` instead and the "new" one (~10 year old or more) does not expose the data (or it exposes it differently).

Is there a collectd version that is compatible with `intel_pstate`?

fantom-x, 2022-08-24:
One option is to not show `Extra items` or it can be graded out if `/sys/devices/system/cpu/cpu0/cpufreq/stats/time_in_state` or similar is missing.

hnyman, 2022-08-24:
> The collectd data collection plugin assumes that the old cpufreq driver is used

Yeah, the upstream plugin is called "cpufreq" for a reason.

> Is there a collectd version that is compatible with intel_pstate?

You might check with the upstream collectd.
Looks like the cpufreq recognizes that the pstate drive does not offer all data: 
https://github.com/collectd/collectd/blob/main/src/cpufreq.c#L56-L74

Based on https://github.com/collectd/collectd/issues/624 the issues regarding the pstate driver has been around for some time  (although at least the basic data is now obtained).

Regarding collectd here, we basically offer the modules that the upstream has created.