openwrt/packages #19251: libgd: Warning about recursive dependency in 22.03.

dreirund, 2022-08-25:
### Maintainer:

@jow-

### Environment:

* OpenWrt git branch `openwrt-22.03`, latest `git pull && ./scripts/feeds update -a && ./scripts/feeds install -a` on 2022-08-25, ca. 08:00 UTC
* Configuration: [&rarr;&hairsp;`.config`](http://ix.io/48s6)
* `grep -i libgd .config`:  
  ```
  CONFIG_PACKAGE_libgd=y
  CONFIG_LIBGD_TIFF=y
  CONFIG_LIBGD_FREETYPE=y
  CONFIG_PACKAGE_libgd-full=m
  CONFIG_PACKAGE_libgdbm=y
  CONFIG_LCD4LINUX_CUSTOM_NEEDS_libgd=y
  ```

### Description:

After a `make clean` and `git pull && ./scripts/feeds update -a && ./scripts/feeds install -a`, a  
`make oldconfig`

prints out

```
tmp/.config-package.in:54634:error: recursive dependency detected!
tmp/.config-package.in:54634:	symbol PACKAGE_libgd is selected by PACKAGE_libgd-full
tmp/.config-package.in:54668:	symbol PACKAGE_libgd-full depends on PACKAGE_libgd
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"
```

(and further on, succeeds).

BKPepe, 2022-08-25:
It is just a warning, which was already reported in comment https://github.com/openwrt/packages/commit/42b36b7180a1859502d72a42dcd6e9ef80519c55#commitcomment-82052792, but yeah, I agree that it should not be there. I will look at it.

BKPepe, 2022-10-06:
Fixed by https://github.com/openwrt/packages/pull/19372

closed 2022-10-06 by BKPepe