openwrt/packages #19270: strongswan: Function not implemented with chacha20poly1305

lowjoel, 2022-08-28:
Maintainer: @pprindeville @Thermi (find it by checking history of the package Makefile)
Environment: Linksys E8450 (aarch64)/Linksys WRT1900ACS (armv7), OpenWrt 22.03-rc6
Description:

@xwang1498 in #18192 and openwrt/openwrt#9951, together with @pprindeville in #18654 implemented support for chacha20poly1305 in master. The kmod change and was backported to 22.03-rc5 (openwrt/openwrt@3cfe050c4a683ecef25d6ded05b1d240921a121a), but not #18192. Regardless of #18192 being backported, it should be sufficient to manually install kmod-crypto-chacha20poly1305:

```
$ opkg list-installed | grep cha
kmod-crypto-chacha20poly1305 - 5.10.134-1
kmod-crypto-echainiv - 5.10.134-1
strongswan-charon - 5.9.5-8
strongswan-mod-chapoly - 5.9.5-8
```

However, when trying to negotiate chacha20poly1305, the kernel still returns Function not implemented:

```
Sun Aug 28 09:21:08 2022 daemon.info ipsec: 16[KNL] received netlink error: Function not implemented (38)
Sun Aug 28 09:21:08 2022 daemon.info ipsec: 16[KNL] unable to add SAD entry with SPI XXXXXXXX (FAILED)
Sun Aug 28 09:21:09 2022 daemon.info ipsec: 16[KNL] received netlink error: Function not implemented (38)
Sun Aug 28 09:21:09 2022 daemon.info ipsec: 16[KNL] unable to add SAD entry with SPI YYYYYYYY (FAILED)
Sun Aug 28 09:21:09 2022 daemon.info ipsec: 16[IKE] unable to install inbound and outbound IPsec SA (SAD) in kernel
Sun Aug 28 09:21:09 2022 daemon.info ipsec: 16[IKE] failed to establish CHILD_SA, keeping IKE_SA
Sun Aug 28 09:21:09 2022 daemon.info ipsec: 16[ENC] generating CREATE_CHILD_SA response 0 [ N(NO_PROP) ]
```

I'm not sure if this is a problem with my config or if it's a problem elsewhere:

/var/swanctl/swanctl.conf:
```
...
    children {
      .... {
        ....
        esp_proposals = chacha20poly1305-curve448,aes256gcm128-curve448,aes256gcm128-modp4096
        ....
      }
    }
...
```

Removing `esp_chacha20poly1305_curve448` does allow phase2 to be negotiated and the tunnel is set up (using aes). Did I miss anything here?

Thermi, 2022-08-28:
currently running kernel and kernel the module is installed for might be different. Check that.

lowjoel, 2022-08-28:
I think this means that they are the same?

```
#  uname -a
Linux ZZZZZZZZ 5.10.134 #0 SMP Sun Jul 31 15:12:47 2022 armv7l GNU/Linux
#  opkg list-installed | grep cha
kmod-crypto-chacha20poly1305 - 5.10.134-1
kmod-crypto-echainiv - 5.10.134-1
strongswan-charon - 5.9.5-8
strongswan-mod-chapoly - 5.9.5-8
```

The module also seems to be loaded:
```
#  lsmod | grep chacha
chacha20poly1305       12288  0
```

Thermi, 2022-08-28:
Does it appear in /proc/crypto? You need `rfc7539esp(chacha20,poly1305)`.

lowjoel, 2022-08-28:
OK, that's weird:

```
#  grep rfc7539 /proc/crypto | wc -l
0
```

So the module is loaded but doesn't appear in `/proc/crypto`. This is happening on two different pieces of hardware (on two different architectures: armv7 and aarch64)

Maybe there's a dependency somewhere that's not included?

Thermi, 2022-08-28:
Maybe. Probably a kernel config issue.

lowjoel, 2022-08-29:
@Thermi I did find that maybe CONFIG_CRYPTO_CHACHA20 and CONFIG_CRYPTO_POLY1305 is needed. I tried adding it in openwrt/openwrt#10551, and I get this:

```
#  grep -E 'rfc|chacha|poly1305' /proc/crypto
name         : xchacha12
driver       : xchacha12-neon
module       : chacha_neon
name         : xchacha20
driver       : xchacha20-neon
module       : chacha_neon
name         : chacha20
driver       : chacha20-neon
module       : chacha_neon
name         : poly1305
driver       : poly1305-neon
module       : poly1305_neon
name         : poly1305
driver       : poly1305-generic
module       : poly1305_generic
name         : xchacha12
driver       : xchacha12-generic
module       : chacha_generic
name         : xchacha20
driver       : xchacha20-generic
module       : chacha_generic
name         : chacha20
driver       : chacha20-generic
module       : chacha_generic
name         : seqiv(rfc4106(gcm(aes)))
driver       : seqiv(rfc4106(gcm-aes-ce))
name         : rfc4106(gcm(aes))
driver       : rfc4106(gcm-aes-ce)
```

So it's getting closer, but no `rfc7539esp(chacha20,poly1305)`. 😢

lowjoel, 2022-08-30:
@Thermi the crypto module seems to be loaded on demand. I pushed through the config and with openwrt/openwrt#10551 I can get it established (needed to compile for two architectures):

```
  child2: #2, reqid 2, INSTALLED, TUNNEL, ESP:CHACHA20_POLY1305/CURVE_448
```

Let me try to figure out if I'm submitting the kernel change correctly.