openwrt/packages #19277: boinc: This project doesn't support computers of type x86_64-openwrt-linux-musl

fantom-x, 2022-08-28:
Maintainer: @N30dG @smoe
Environment: 22.03-SNAPSHOT x86_64 (Intel I5-7400)

I can silence this error with below, but then the jobs fail promptly.

```
<cc_config>
<options>
<alt_platform>x86_64-pc-linux-gnu</alt_platform>
</options>
</cc_config>
```

smoe, 2022-08-29:
The binary that fails is directly provided by the project you are contributing to. It is not provided by OpenWrt. If it fails then it fails because of a technical insufficiency of that project, not of this BOINC package or of OpenWrt. What happens is that the project is not perfectly statically linked but still depends on the glibc. At the time we found that installing the OpenWrt glibc package does not help that situation, but you may want to give it another try.

Which project were you contributing to? TN-Grid should work.

fantom-x, 2022-08-29:
@smoe , are you saying that the workaround I applied for the computer type is correct? I did not find it documented anywhere.

smoe, 2022-08-29:
You did everything right from what I see. We crafted https://volunteercomputinghelp.github.io/ a while back.

fantom-x, 2022-08-29:
Ok, I wish that page came up in the searches I did. Looks like things have changed since the page was written:

Rosetta@Home needs VirtualBox `VirtualBox is not installed`
World Community Grid has GPU jobs only (at this time?)

Have not tried the third one yet. Amicable Numbers works though, but it seems to be a waste of resources.

In the meantime, I put the app into a jail: it downloads executables from random sources :) What I am not sure yet is if the child processes remain in the same jail...

```diff -u /rom/etc/init.d/boinc-client ../boinc-client 
--- /rom/etc/init.d/boinc-client	2022-08-28 10:56:01.000000000 -0400
+++ ../boinc-client	2022-08-29 12:55:09.715594775 -0400
@@ -4,7 +4,7 @@
 USE_PROCD=1
 
 BOINCEXE_NAME=boinc_client
-BOINCDIR=/opt/boinc/
+BOINCDIR=/opt/boinc
 PRESETDIR=/usr/share/boinc
 BOINCUSR=boinc
 BOINCEXE_OPTS="--check_all_logins --redirectio --dir $BOINCDIR"
@@ -41,7 +41,7 @@
    # now use procd to start boinc
    procd_open_instance $BOINCEXE_NAME
 
-   procd_set_param command $BOINCEXE_NAME
+   procd_set_param command $(which $BOINCEXE_NAME)
    procd_append_param command $BOINCEXE_OPTS
    procd_set_param user $BOINCUSR
    procd_set_param limits core="unlimited"
@@ -49,5 +49,13 @@
    procd_set_param stderr 1
    procd_set_param pidfile $PID_FILE
 
+   procd_add_jail $BOINCEXE_NAME log requirejail
+   procd_add_jail_mount /etc/TZ
+   procd_add_jail_mount /proc/cpuinfo /proc/meminfo
+   procd_add_jail_mount /etc/ssl/certs/ca-certificates.crt
+   procd_add_jail_mount $PRESETDIR
+   procd_add_jail_mount_rw $(dirname $PID_FILE)
+   procd_add_jail_mount_rw $BOINCDIR
+
    procd_close_instance
 }
```

fantom-x, 2022-09-01:
Thx, but I was looking for a fun weekend project, not a long term commitment. Feel free to use the patch above if you would like, but I think a Docker container would work better in this case and make it compatible with more projects. I will give it another shot once Docker works in OpenWrt again.

closed 2022-09-01 by fantom-x