openwrt/packages #19281: Docker: ipv6 NAT network not working

kuoruan, 2022-08-30:
Maintainer: @G-M0N3Y-2503 
Environment: OpenWrt 22.03 (aarch64_cortex-a53)

Description:

I modified the dockerd file:

```patch
--- a/dockerd
+++ b/dockerd
@@ -184,6 +184,8 @@
        config_get ip globals ip ""
        config_get fixed_cidr globals fixed_cidr ""
        config_get fixed_cidr_v6 globals fixed_cidr_v6 ""
+       config_get experimental globals experimental ""
+       config_get ip6tables globals ip6tables ""

        . /usr/share/libubox/jshn.sh
        json_init
@@ -205,6 +207,8 @@
        [ -z "${ip}" ] || json_add_string "ip" "${ip}"
        [ -z "${fixed_cidr}" ] || json_add_string "fixed-cidr" "${fixed_cidr}"
        [ -z "${fixed_cidr_v6}" ] || json_add_string "fixed-cidr-v6" "${fixed_cidr_v6}"
+       [ -z "${experimental}" ] || json_add_boolean "experimental" "${experimental}"
+       [ -z "${ip6tables}" ] || json_add_boolean "ip6tables" "${ip6tables}"
        json_dump > "${DOCKERD_CONF}"

        [ "${iptables}" -eq "1" ] && config_foreach iptables_add_blocking_rule firewall
```

and config:

```
option fixed_cidr_v6 'fc00:100:1::/80'
option ipv6 '1'
option experimental '1'
option ip6tables '1'
```

The `daemon.json` is like

```
{ "data-root": "\/opt\/docker\/", "log-level": "warn", "iptables": true, "ipv6": true, "fixed-cidr-v6": "fc00:100:1::\/80", "experimental": true, "ip6tables": true }
```

My new container has IPv6 address:

```
[
    {
        "Name": "bridge",
        "Id": "bdc1c7dc065d4bae95a43d3e9cd47be91422f527581f4062c94892252532600a",
        "Created": "2022-08-30T13:13:32.01903987+08:00",
        "Scope": "local",
        "Driver": "bridge",
        "EnableIPv6": true,
        "IPAM": {
            "Driver": "default",
            "Options": null,
            "Config": [
                {
                    "Subnet": "172.17.0.0/16",
                    "Gateway": "172.17.0.1"
                },
                {
                    "Subnet": "fc00:100:1::/80",
                    "Gateway": "fc00:100:1::1"
                }
            ]
        },
        "Internal": false,
        "Attachable": false,
        "Ingress": false,
        "ConfigFrom": {
            "Network": ""
        },
        "ConfigOnly": false,
        "Containers": {
            "31b3d129e7bc60d753c431fe1fd2590303bca947cf5a8f69790cd9f96696268e": {
                "Name": "clash",
                "EndpointID": "acbf5741b3bbaa4709bb5f78e33641664cbecf36994eb550548d3ed891cab8ef",
                "MacAddress": "02:42:ac:11:00:02",
                "IPv4Address": "172.17.0.2/16",
                "IPv6Address": "fc00:100:1::242:ac11:2/80"
            }
        },
        "Options": {
            "com.docker.network.bridge.default_bridge": "true",
            "com.docker.network.bridge.enable_icc": "true",
            "com.docker.network.bridge.enable_ip_masquerade": "true",
            "com.docker.network.bridge.host_binding_ipv4": "0.0.0.0",
            "com.docker.network.bridge.name": "docker0",
            "com.docker.network.driver.mtu": "1500"
        },
        "Labels": {}
    }
]
```

but I can't ping outside in the container.

G-M0N3Y-2503, 2022-08-31:
Only looking quickly, the docker iptables features don't work with nftables/fw4 whitch I believe OpenWrt 22.03 uses by default.
Last I checked you can replace fw4 with fw3 and this might work. Otherwise, with the newer firewall I think you have to manage it manually via OpenWrt and the docker bridges.

G-M0N3Y-2503, 2024-07-02:
https://github.com/openwrt/packages/issues/17766