openwrt/packages #19290: p910nd: only available over IPv4, doesn't bind on IPv6

notgood, 2022-09-01:
Maintainer: @pkerling
Environment: OpenWRT 22.03 latest

Description:
p910nd only binds on IPv4 address 0.0.0.0
Should bind on both IPv4 and IPv6 ::/0

```
# netstat -tulpn | grep p910
tcp        0      0 0.0.0.0:9100            0.0.0.0:*               LISTEN      27250/p9100d

# netstat -tulpn | grep uhtt
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      2233/uhttpd
tcp        0      0 0.0.0.0:443             0.0.0.0:*               LISTEN      2233/uhttpd
tcp        0      0 :::80                   :::*                    LISTEN      2233/uhttpd
tcp        0      0 :::443                  :::*                    LISTEN      2233/uhttpd

cat /etc/config/p910nd

config p910nd
        option device '/dev/usb/lp0'
        option port '0'
        option bidirectional '0'
        option enabled '1'

```

pkerling, 2022-09-03:
Hi, thanks for reporting the issue, but as I am no longer using p910nd for multiple years now, I will not look into it further. It should be pretty easy to fix though if p910nd supports binding to IPv6 at all.

neheb, 2022-09-15:
Relevant: https://dev.archive.openwrt.org/changeset/22432.html

neheb, 2022-09-15:
Upstream was moved to https://github.com/kenyapcomau/p910nd

Maybe report there.