openwrt/packages #19292: lxc-auto: errors during an Image Builder build

fantom-x, 2022-09-02:
Maintainer: @ratkaj
Environment: 22.03-SNAPSHOT x86_64

Getting these errors when using Image Builder.

```
Finalizing root filesystem...
/build/openwrt-imagebuilder/build_dir/target-x86_64_musl/root-x86/etc/init.d/lxc-auto: line 3: /lib/functions.sh: No such file or directory
Enabling boot
Enabling cgroupfs-mount
Enabling collectd
Enabling cron
Enabling ddns
Enabling dnsmasq
Enabling done
Enabling dropbear
Enabling firewall
Enabling fstab
Enabling gpio_switch
Enabling led
Enabling lm-sensors
Enabling log
Enabling luci_statistics
./etc/init.d/lxc-auto: line 3: /lib/functions.sh: No such file or directory
Enabling lxc-auto
Enabling network
Enabling nlbwmon
Enabling odhcpd
Enabling rpcd
Enabling sqm
Enabling sysctl
Enabling sysfixtime
Enabling sysfsutils
Enabling sysntpd
Enabling system
Enabling ucitrack
Enabling uhttpd
Enabling umount
Enabling urandom_seed
Enabling urngd
```

BKPepe, 2022-09-03:
Which exact commit hash are you using for packages repository? This was already fixed here, so I am thinking that you are just using older HEAD.

fantom-x, 2022-09-03:
I do not know the commit for the package repo, but `OPENWRT_RELEASE="OpenWrt 22.03-SNAPSHOT r19677-ae1786e543"`

Looking at the latest source code in both master & 22.03 and they both have the same issue:

https://github.com/openwrt/packages/blob/master/utils/lxc/files/lxc-auto.init
https://github.com/openwrt/packages/blob/openwrt-22.03/utils/lxc/files/lxc-auto.init

closed 2022-11-04 by neheb