openwrt/packages #19352: Bandwidthd  version 2.0.1-35-5 sending SIGSEGV to bandwidthd for invalid read access from 00000018

frankgrant, 2022-09-13:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: 
arch MediaTek MT7621 ver:1 eco:3
model Ubiquiti EdgeRouter X
OpenWrt version  OpenWrt 22.03.0 r19685-512e76967f / LuCI openwrt-22.03 branch git-22.245.77528-487e58a

Description:
Getting kernel error messages:
do_page_fault(): sending SIGSEGV to bandwidthd for invalid read access from 00000018
epc = 77e76534 in libc.so[77dea000+a9000]
ra  = 77e76410 in libc.so[77dea000+a9000]

No graphs are being produced

padre-lacroix, 2022-09-18:
OK: I have to check on a 32 bit virtual OpenWRT router (or if I can get my jand on a Ubiquiti Edge router) that the fix #19392 fixes that problem.  Unless somehow has confirmed that it works properly.

frankgrant, 2022-09-28:
Any outlook when this fix will be added to the openwrt packages?

padre-lacroix, 2022-09-29:
Sorry,  I have been pretty sick int he last 10 days.  The opened pull request (19392) is for the Master branch.  I will approve the pull request and it will be available in the nightly build.

Fort the stable version (openwrt-22.03) another pull request will have to be made in that branch to be available in the packages in the stable version of openwrt.

closed 2022-09-29 by neheb

frankgrant, 2022-10-01:
Updated to the latest  version 2.0.1-35-6 and still get the error
do_page_fault(): sending SIGSEGV to bandwidthd for invalid read access from 00000018
epc = 77e9e534 in libc.so[77e12000+a9000]
ra  = 77e9e410 in libc.so[77e12000+a9000]

padre-lacroix, 2022-10-01:
Sorry that there is still a bug!
I will have to investigate this a little bit further (obviously).
I will try to replicate the bug in a router (I have a couple siting on a shelf) and then I will try to fix it when I will be able to replicate it.

As I mentionned previously, I have been pretty sick lately, and I am still a little bit sick.  So bear with me if it takes some time.

padre-lacroix, 2022-10-01:
OK, I replicated the bug in a Archer C7 v2.
Now to fix it!  I will start investigating the problem this weekend.