openwrt/packages #19361: It would be missing a package that enables the radios by default

castillofrancodamian, 2022-09-14:
That is, that package detects that if the wireless configuration file is written in such a way, the radios are enabled.

feckert, 2022-09-14:
What exactly you mean!
The configuration is created via a  script, if that's what you mean.
https://github.com/openwrt/openwrt/blob/master/package/kernel/mac80211/files/lib/wifi/mac80211.sh

castillofrancodamian, 2022-09-14:
As by default the radios are disabled, it would be nice if when installing a package they are enabled on the first factory boot. It would be better for the new online compiler or ImageBuilder to have as "official" as possible without having to enter configuration files. I think many devices that are out of reach would benefit.

GeorgeSapkin, 2025-11-15:
This is better suited for the official forum: https://forum.openwrt.org/t/community-question-what-do-you-want-to-see-in-openwrt/216706/245

closed 2025-11-15 by GeorgeSapkin