openwrt/packages #19366: https-dns-proxy: dmesg page fault

systemcrash, 2022-09-14:
Maintainer: @stangri 

-- | Value
-- | --
Model | TP-Link Archer C7 v2
Architecture | Qualcomm Atheros QCA9558 ver 1 rev 0
Target Platform | ath79/generic
Firmware Version | OpenWrt 22.03.0 r19685-512e76967f / LuCI openwrt-22.03 branch git-22.245.77528-487e58a
Kernel Version | 5.10.138

Description:

added in https-dns-proxy into imagebuilder [here](https://firmware-selector.openwrt.org/). At boot, the kernel log shows:

```
[   78.258834] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 63222510
[   78.268317] epc = 77e72df4 in libc.so[77df4000+a9000]
[   78.273488] ra  = 77e74b04 in libc.so[77df4000+a9000]
...
[   89.156259] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 6322251a
[   89.165861] epc = 77dc2dc0 in libc.so[77d44000+a9000]
[   89.171129] ra  = 77dc4b04 in libc.so[77d44000+a9000]
...
[   94.439188] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 63222520
[   94.448917] epc = 77e2cdf4 in libc.so[77dae000+a9000]
[   94.454103] ra  = 77e2eb04 in libc.so[77dae000+a9000]
[   99.763575] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 63222525
[   99.773390] epc = 77e66dc0 in libc.so[77de8000+a9000]
[   99.778544] ra  = 77e68b04 in libc.so[77de8000+a9000]
[  104.985969] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 6322284c
[  104.995583] epc = 77e4adf4 in libc.so[77dcc000+a9000]
[  105.000758] ra  = 77e4cb04 in libc.so[77dcc000+a9000]
[  129.174846] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 63222865
[  129.184505] epc = 77e78dc0 in libc.so[77dfa000+a9000]
[  129.189687] ra  = 77e7ab04 in libc.so[77dfa000+a9000]
[  134.335382] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 6322286a
[  134.344791] epc = 77e4adc0 in libc.so[77dcc000+a9000]
[  134.350131] ra  = 77e4cb04 in libc.so[77dcc000+a9000]
[  139.404103] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 6322286f
[  139.413861] epc = 77e28dc0 in libc.so[77daa000+a9000]
[  139.419098] ra  = 77e2ab04 in libc.so[77daa000+a9000]
[  144.494286] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 63222874
[  144.503690] epc = 77de2df4 in libc.so[77d64000+a9000]
[  144.508995] ra  = 77de4b04 in libc.so[77d64000+a9000]
[  149.580321] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 63222879
[  149.589720] epc = 77eb2dc0 in libc.so[77e34000+a9000]
[  149.595016] ra  = 77eb4b04 in libc.so[77e34000+a9000]
[  154.672733] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 6322287e
[  154.682129] epc = 77eb4dc0 in libc.so[77e36000+a9000]
[  154.687367] ra  = 77eb6b04 in libc.so[77e36000+a9000]
```

stangri, 2022-09-15:
Can you please post the output of `opkg list-installed`.

systemcrash, 2022-09-15:
Here, sorry about that. 

```
6in4 - 28
ath10k-board-qca988x - 20220411-1
ath10k-firmware-qca988x-ct - 2020-11-08-1
base-files - 1490-r19685-512e76967f
busybox - 1.35.0-3
ca-bundle - 20211016-1
cgi-io - 2022-08-10-901b0f04-21
ddns-scripts - 2.8.2-25
ddns-scripts-services - 2.8.2-25
dnsmasq - 2.86-14
dropbear - 2022.82-2
firewall4 - 2022-09-01-f5fcdcf2-1
fstools - 2022-06-02-93369be0-2
fwtool - 2019-11-12-8f7fe925-1
getrandom - 2021-08-03-205defb5-2
hostapd-common - 2022-01-16-cff80b4f-12
https-dns-proxy - 2021-11-22-3
iw - 5.16-1
iwinfo - 2022-08-19-0dad3e66-1
jansson4 - 2.13.1-2
jshn - 2022-05-15-d2223ef9-1
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 5.10.138-1-abf0c66378f3d0588b20489662c12426
kmod-ath - 5.10.138+5.15.58-1-1
kmod-ath10k-ct - 5.10.138+2022-05-13-f808496f-1
kmod-ath9k - 5.10.138+5.15.58-1-1
kmod-ath9k-common - 5.10.138+5.15.58-1-1
kmod-cfg80211 - 5.10.138+5.15.58-1-1
kmod-crypto-aead - 5.10.138-1
kmod-crypto-ccm - 5.10.138-1
kmod-crypto-cmac - 5.10.138-1
kmod-crypto-crc32c - 5.10.138-1
kmod-crypto-ctr - 5.10.138-1
kmod-crypto-gcm - 5.10.138-1
kmod-crypto-gf128 - 5.10.138-1
kmod-crypto-ghash - 5.10.138-1
kmod-crypto-hash - 5.10.138-1
kmod-crypto-hmac - 5.10.138-1
kmod-crypto-kpp - 5.10.138-1
kmod-crypto-lib-chacha20 - 5.10.138-1
kmod-crypto-lib-chacha20poly1305 - 5.10.138-1
kmod-crypto-lib-curve25519 - 5.10.138-1
kmod-crypto-lib-poly1305 - 5.10.138-1
kmod-crypto-manager - 5.10.138-1
kmod-crypto-null - 5.10.138-1
kmod-crypto-rng - 5.10.138-1
kmod-crypto-seqiv - 5.10.138-1
kmod-crypto-sha256 - 5.10.138-1
kmod-gpio-button-hotplug - 5.10.138-3
kmod-hwmon-core - 5.10.138-1
kmod-iptunnel - 5.10.138-1
kmod-iptunnel4 - 5.10.138-1
kmod-lib-crc-ccitt - 5.10.138-1
kmod-lib-crc32c - 5.10.138-1
kmod-mac80211 - 5.10.138+5.15.58-1-1
kmod-nf-conntrack - 5.10.138-1
kmod-nf-conntrack6 - 5.10.138-1
kmod-nf-flow - 5.10.138-1
kmod-nf-log - 5.10.138-1
kmod-nf-log6 - 5.10.138-1
kmod-nf-nat - 5.10.138-1
kmod-nf-reject - 5.10.138-1
kmod-nf-reject6 - 5.10.138-1
kmod-nfnetlink - 5.10.138-1
kmod-nft-core - 5.10.138-1
kmod-nft-fib - 5.10.138-1
kmod-nft-nat - 5.10.138-1
kmod-nft-offload - 5.10.138-1
kmod-nls-base - 5.10.138-1
kmod-phy-ath79-usb - 5.10.138-1
kmod-ppp - 5.10.138-1
kmod-pppoe - 5.10.138-1
kmod-pppox - 5.10.138-1
kmod-sit - 5.10.138-1
kmod-slhc - 5.10.138-1
kmod-udptunnel4 - 5.10.138-1
kmod-udptunnel6 - 5.10.138-1
kmod-usb-core - 5.10.138-1
kmod-usb-ehci - 5.10.138-1
kmod-usb-ledtrig-usbport - 5.10.138-1
kmod-usb2 - 5.10.138-1
kmod-wireguard - 5.10.138-1
libblobmsg-json20220515 - 2022-05-15-d2223ef9-1
libc - 1.2.3-4
libcap-ng - 0.8.2-1
libcares - 1.18.1-1
libcurl4 - 7.83.1-3
libev - 4.33-1
libgcc1 - 11.2.0-4
libiwinfo-data - 2022-08-19-0dad3e66-1
libiwinfo-lua - 2022-08-19-0dad3e66-1
libiwinfo20210430 - 2022-08-19-0dad3e66-1
libjson-c5 - 0.15-2
libjson-script20220515 - 2022-05-15-d2223ef9-1
liblua5.1.5 - 5.1.5-10
liblucihttp-lua - 2022-07-08-6e68a106-1
liblucihttp0 - 2022-07-08-6e68a106-1
libmbedtls12 - 2.28.1-1
libmnl0 - 1.0.5-1
libnftnl11 - 1.2.1-1
libnghttp2-14 - 1.44.0-1
libnl-tiny1 - 2021-11-21-8e0555fb-1
libpcap1 - 1.10.1-3
libpthread - 1.2.3-4
librt - 1.2.3-4
libubox20220515 - 2022-05-15-d2223ef9-1
libubus-lua - 2022-06-01-2bebf93c-1
libubus20220601 - 2022-06-01-2bebf93c-1
libuci-lua - 2021-10-22-f84f49f0-6
libuci20130104 - 2021-10-22-f84f49f0-6
libuclient20201210 - 2021-05-14-6a6011df-1
libucode20220812 - 2022-08-29-344fa9e6-1
libustream-wolfssl20201210 - 2022-01-16-868fd881-1
libuuid1 - 2.37.3-3
libwolfssl5.4.0.ee39414e - 5.4.0-stable-5
logd - 2021-08-03-205defb5-2
lua - 5.1.5-10
luci-app-ddns - git-22.193.49501-e0f0502
luci-app-firewall - git-22.089.67563-7e3c1b4
luci-app-https-dns-proxy - 2021-09-27-3
luci-app-opkg - git-22.154.41881-28e92e3
luci-app-p910nd - git-20.108.38431-8f34e10
luci-app-upnp - git-22.250.64651-a61c041
luci-app-wireguard - git-21.322.66896-8ae208d
luci-base - git-22.245.77528-487e58a
luci-compat - git-22.069.45071-03bb0e2
luci-lib-base - git-20.232.39649-1f6dc29
luci-lib-ip - git-20.250.76529-62505bd
luci-lib-ipkg - git-18.318.71164-4bbe325
luci-lib-jsonc - git-22.097.61921-7513345
luci-lib-nixio - git-20.234.06894-c4a4e43
luci-mod-admin-full - git-19.253.48496-3f93650
luci-mod-network - git-22.244.54818-b13d8c7
luci-mod-status - git-22.189.48501-6731190
luci-mod-system - git-22.140.66206-02913be
luci-proto-ipv6 - git-21.148.48881-79947af
luci-proto-wireguard - git-22.213.40001-18e6040
luci-theme-bootstrap - git-22.141.59265-d8ecf48
miniupnpd-nftables - 2022-08-31-68c8ec50-1
mtd - 26
netifd - 2022-08-25-76d2d41b-1
nftables-json - 1.0.2-2.1
odhcp6c - 2022-08-05-7d21e8d8-18
odhcpd-ipv6only - 2022-03-22-860ca900-1
openwrt-keyring - 2022-03-25-62471e69-3
opkg - 2022-02-24-d038e5b6-1
p910nd - 0.97-9
ppp - 2.4.9.git-2021-01-04-3
ppp-mod-pppoe - 2.4.9.git-2021-01-04-3
procd - 2022-06-01-7a009685-1
procd-seccomp - 2022-06-01-7a009685-1
procd-ujail - 2022-06-01-7a009685-1
rpcd - 2022-08-24-82904bd4-1
rpcd-mod-file - 2022-08-24-82904bd4-1
rpcd-mod-iwinfo - 2022-08-24-82904bd4-1
rpcd-mod-luci - 20210614
swconfig - 12
tcpdump - 4.9.3-4
uboot-envtools - 2022.01-31
ubox - 2021-08-03-205defb5-2
ubus - 2022-06-01-2bebf93c-1
ubusd - 2022-06-01-2bebf93c-1
uci - 2021-10-22-f84f49f0-6
uclient-fetch - 2021-05-14-6a6011df-1
ucode - 2022-08-29-344fa9e6-1
ucode-mod-fs - 2022-08-29-344fa9e6-1
ucode-mod-ubus - 2022-08-29-344fa9e6-1
ucode-mod-uci - 2022-08-29-344fa9e6-1
uhttpd - 2022-08-12-e3395cd9-1
urandom-seed - 3
urngd - 2020-01-21-c7f7b6b6-1
usign - 2020-05-23-f1f65026-1
wireguard-tools - 1.0.20210424-3
wireless-regdb - 2022.06.06-1
wpad-wolfssl - 2022-01-16-cff80b4f-12
```

stangri, 2022-09-15:
> Here, sorry about that.
> 

I've had a wild theory that maybe the image was somehow built with some dependencies missing, but everything seems to be present. 

I don't have a technical expertise to troubleshoot it further, I'd like to escalate this to the upstream to seek guidance, it could be beneficial to be on the latest version of libcurl for that.

I've [created a PR](https://github.com/openwrt/packages/pull/19377) for the 7.85.0 version of curl in master, I'll cherry-pick it for the release branches within a few days unless there are issues reported for the updated version in a master branch.

Ohfalderal, 2022-09-27:
The segmentation fault (22.03) only occurs when logging is enabled and verbosity > 0.

N.B. - setting logging / verbosity for 21.02 does not page fault

stangri, 2022-09-27:
> The segmentation fault (22.03) only occurs when logging is enabled and verbosity > 0.
> 
> N.B. - setting logging / verbosity for 21.02 does not page fault

Thanks for adding this. Just so I have the complete information for upstream, can you please update curl/ibcurl to 7.85.0 and include the output for: `curl --version`?

Ohfalderal, 2022-09-27:
opkg list-installed *curl*
libcurl4 - 7.83.1-3

opkg install libcurl
Upgrading libcurl4 on root from 7.83.1-3 to 7.85.0-1...

vi /etc/config/https-dns-proxy

service https-dns-proxy restart

Tue Sep 27 18:00:31 2022 user.notice https-dns-proxy: Starting service ✓✓✓✓
Tue Sep 27 18:00:31 2022 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Tue Sep 27 18:00:31 2022 kern.info kernel: [52874.728769] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 63331e1f
Tue Sep 27 18:00:31 2022 kern.info kernel: [52874.738717] epc = 77e0cdc0 in libc.so[77d8e000+a9000]
Tue Sep 27 18:00:31 2022 kern.info kernel: [52874.743959] ra  = 77e0eb04 in libc.so[77d8e000+a9000]

stangri, 2022-09-28:
I can't reproduce this issue with setting verbosity to 2.

Ohfalderal, 2022-09-28:
Verbosity setting altered from '1' to '2' with no change to the observed result (SIGSEGV).

Kindly post your /etc/config/https-dns-proxy configuration.

Ohfalderal, 2022-09-28:
FYI - the device is TP-Link Archer C6 v2

Ohfalderal, 2022-10-03:
So the trail ends here?

systemcrash, 2022-10-04:
Well, it doesn't seem critical, since it works normally after bootup, when the system settles down. Maybe before config is loaded, the internal default settings trigger the crash?

stangri, 2022-10-04:
I need a config from anyone experiencing this issue to try to reproduce this.

systemcrash, 2022-10-04:
output of  `cat /etc/config/https-dns-proxy`
```

config main 'config'
	option update_dnsmasq_config '*'
	option force_dns '1'
	list force_dns_port '53'
	list force_dns_port '853'

config https-dns-proxy
	option listen_addr '127.0.0.1'
	option user 'nobody'
	option group 'nogroup'
	option bootstrap_dns '74.82.42.42,2001:470:20::2'
	option resolver_url 'https://ordns.he.net/dns-query'
	option listen_port '5054'

config https-dns-proxy
	option bootstrap_dns '116.202.176.26,1.1.1.1'
	option resolver_url 'https://doh.libredns.gr/dns-query'
	option listen_port '5055'

config https-dns-proxy
	option resolver_url 'https://cloudflare-dns.com/dns-query'
	option bootstrap_dns '1.1.1.1,1.0.0.1,2606:4700:4700::1111,2606:4700:4700::1001'

```

Ohfalderal, 2022-10-04:
Uncomment both 'logfile' and 'verbosity' options of any one of the four sections -

```
config main 'config'
	option update_dnsmasq_config '*'
	option force_dns '1'
	list force_dns_port '53'
	list force_dns_port '853'
# ports listed below are used by some
# of the dnscrypt-proxy v1 resolvers
#	list force_dns_port '553'
#	list force_dns_port '1443'
#	list force_dns_port '4343'
#	list force_dns_port '4434'
#	list force_dns_port '5443'
#	list force_dns_port '8443'

config https-dns-proxy
	option bootstrap_dns '2a01:3a0:53:53::'
	option resolver_url 'https://unicast.censurfridns.dk/dns-query'
	option use_ipv6_resolvers_only 1
#	option logfile '/var/log/DoH_unicast.censurfridns.dk.log' 
#	option verbosity 1 

config https-dns-proxy
	option bootstrap_dns '2a07:a8c0::84:5e57,2a07:a8c1::84:5e57'
	option resolver_url 'https://dns.nextdns.io/845e57'
	option use_ipv6_resolvers_only 1
#	option logfile '/var/log/DoH_dns.nextdns.io.log' 
#	option verbosity 1

config https-dns-proxy
	option bootstrap_dns '2001:67c:28a4::'
	option resolver_url 'https://anycast.uncensoreddns.org/dns-query'
	option use_ipv6_resolvers_only 1
#	option logfile '/var/log/DoH_anycast.censurfridns.dk.log' 
#	option verbosity 1

config https-dns-proxy
	option bootstrap_dns '2a07:a8c0::84:5e57,2a07:a8c1::84:5e57'
	option resolver_url 'https://secure.avastdns.com/dns-query'
	option use_ipv6_resolvers_only 1
#	option logfile '/var/log/DoH_secure.avastdns.com.log' 
#	option verbosity 1
```

Ohfalderal, 2022-10-04:
> output of `cat /etc/config/https-dns-proxy`
>
> ```
>
> config main 'config'
> 	option update_dnsmasq_config '*'
> 	option force_dns '1'
> 	list force_dns_port '53'
> 	list force_dns_port '853'
>
> config [...]

Review the arguments of the spawned processes. Without the 'use_ipv6_resolvers_only' option set, only the IPV4 addresses are passed and using the option, only IPV6 addresses are passed.

stangri, 2022-10-04:
> #	option logfile '/var/log/DoH_unicast.censurfridns.dk.log'

I believe `/var/log` is owned by root and other users don't have the write permissions there, so nobody/nogroup can't write there. Could the crash be because of that?

Ohfalderal, 2022-10-05:
No, the process(es) is spawned under 'root' and during initialisation, updates its credentials, as does DNSmasq.

```
root@pWPW10-2:~# ls -l /var/log/
-rw-r-----    1 root     root        251521 Oct  5 10:33 DoH_anycast.censurfridns.dk_IPV6.log
-rw-r-----    1 root     root         36170 Oct  5 10:33 DoH_dns.digitale-gesellschaft.ch_IPV6.log
-rw-r-----    1 root     root         49522 Oct  5 10:33 DoH_dns.nextdns.io_IPV6.log
-rw-r-----    1 root     root        251378 Oct  5 10:33 DoH_secure.avastdns.com_IPV6.log
-rw-r-----    1 root     root        249875 Oct  5 10:33 DoH_unicast.censurfridns.dk_IPV6.log

 4614 nobody    4180 S    /usr/sbin/https-dns-proxy -r https://unicast.censurfridns.dk/dns-query -a 127.0.0.1 -p 5053 -b 2a01:3a0:53
 4615 nobody    5352 S    /usr/sbin/https-dns-proxy -r https://dns.nextdns.io/845e57 -a 127.0.0.1 -p 5054 -b 2a07:a8c0::84:5e57,2a07
 4616 nobody    4180 S    /usr/sbin/https-dns-proxy -r https://anycast.uncensoreddns.org/dns-query -a 127.0.0.1 -p 5055 -b 2001:67c:
 4617 nobody    4676 S    /usr/sbin/https-dns-proxy -r https://dns.digitale-gesellschaft.ch/dns-query -a 127.0.0.1 -p 5056 -b 2606:4
 4618 nobody    4180 S    /usr/sbin/https-dns-proxy -r https://secure.avastdns.com/dns-query -a 127.0.0.1 -p 5057 -b 2001:67c:28a4::

```
Under 21.03, this config with logging does not SIGSEGV.

stangri, 2022-10-05:
> No, the process(es) is spawned under 'root' and during initialisation, updates its credentials, as does DNSmasq.
>
> ```
> root@pWPW10-2:~# ls -l /var/log/
> -rw-r-----    1 root     root        251521 [...]

Sorry, I'm confused, I thought you said that SIGSEGV happens at boot, yet there are non-zero log files. Could you please clarify the conditions it crashes and conditions it doesn't crash and fills up logs?

stangri, 2022-10-05:
> I see. This config is generated by using luci GUI. I would have expected that both v4/6 are used from the config without specific intervention. @stangri can there be middle ground option for both v4/6? (don't wish to derail this bug report tho)

Yes, please open a separate issue.

stangri, 2022-10-07:
I see. I'll need to test this myself, I'll update this issue on the progress.

stangri, 2022-10-11:
I can't reproduce the `https-dns-proxy` crash with the following config:

```
config https-dns-proxy
	option bootstrap_dns '149.112.121.10,149.112.122.10,2620:10A:80BB::10,2620:10A:80BC::10'
	option resolver_url 'https://private.canadianshield.cira.ca/dns-query'
	option verbosity  '2'
	option logfile '/var/log/DoH_canadianshield.log'
```

@Ohfalderal can you re-test with the shorted file names for the logfile? If we can narrow down the source of the problem before reporting upstream, that'd be helpful.

Ohfalderal, 2022-10-14:
@stangri 
I notice the update is available in the mirror repository that is configured in the AP and will pull and test it this coming weekend.

Ohfalderal, 2022-10-15:
@stangri 
Uhh, no -
```
root@pWPW10-0:~# vi /etc/config/https-dns-proxy
root@pWPW10-0:~# service https-dns-proxy restart
Stopping https-dns-proxy 2022-08-12-1 ✓
Starting https-dns-proxy 2022-08-12-1 ✓✓✓
root@pWPW10-0:~# logread -l20
Sat Oct 15 09:32:55 2022 daemon.info dnsmasq[1]: using only locally-known addresses for local
Sat Oct 15 09:32:55 2022 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Sat Oct 15 09:32:55 2022 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Sat Oct 15 09:32:55 2022 daemon.info dnsmasq[1]: using only locally-known addresses for mask-h2.icloud.com
Sat Oct 15 09:32:55 2022 daemon.info dnsmasq[1]: using only locally-known addresses for mask.icloud.com
Sat Oct 15 09:32:55 2022 daemon.info dnsmasq[1]: using 1 more local addresses
Sat Oct 15 09:32:55 2022 daemon.info dnsmasq[1]: read /etc/hosts - 4 addresses
Sat Oct 15 09:32:55 2022 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Sat Oct 15 09:32:55 2022 kern.info kernel: [22419.664161] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 634a6227
Sat Oct 15 09:32:55 2022 kern.info kernel: [22419.673903] epc = 77eb0dc0 in libc.so[77e32000+a9000]
Sat Oct 15 09:32:55 2022 kern.info kernel: [22419.679387] ra  = 77eb2b04 in libc.so[77e32000+a9000]
Sat Oct 15 09:33:00 2022 kern.info kernel: [22424.768945] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 634a622c
Sat Oct 15 09:33:00 2022 kern.info kernel: [22424.778701] epc = 77e7adf4 in libc.so[77dfc000+a9000]
Sat Oct 15 09:33:00 2022 kern.info kernel: [22424.784060] ra  = 77e7cb04 in libc.so[77dfc000+a9000]
Sat Oct 15 09:33:05 2022 kern.info kernel: [22429.810433] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 634a6231
Sat Oct 15 09:33:05 2022 kern.info kernel: [22429.820151] epc = 77e16dc0 in libc.so[77d98000+a9000]
Sat Oct 15 09:33:05 2022 kern.info kernel: [22429.825650] ra  = 77e18b04 in libc.so[77d98000+a9000]
Sat Oct 15 09:33:11 2022 kern.info kernel: [22434.848670] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 634a6236
Sat Oct 15 09:33:11 2022 kern.info kernel: [22434.858382] epc = 77eb6dc0 in libc.so[77e38000+a9000]
Sat Oct 15 09:33:11 2022 kern.info kernel: [22434.863713] ra  = 77eb8b04 in libc.so[77e38000+a9000]
root@pWPW10-0:~# vi /etc/config/https-dns-proxy
root@pWPW10-0:~# service https-dns-proxy restart
Stopping https-dns-proxy 2022-08-12-1 ✓
Starting https-dns-proxy 2022-08-12-1 ✓✓✓
root@pWPW10-0:~# logread -l20
Sat Oct 15 09:34:13 2022 user.notice https-dns-proxy: Starting service ✓✓✓
Sat Oct 15 09:34:13 2022 daemon.info dnsmasq[1]: started, version 2.86 cachesize 150
Sat Oct 15 09:34:13 2022 daemon.info dnsmasq[1]: DNS service limited to local subnets
Sat Oct 15 09:34:13 2022 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
Sat Oct 15 09:34:13 2022 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Sat Oct 15 09:34:13 2022 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5055
Sat Oct 15 09:34:13 2022 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5053
Sat Oct 15 09:34:13 2022 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5054
Sat Oct 15 09:34:13 2022 daemon.info dnsmasq[1]: using only locally-known addresses for test
Sat Oct 15 09:34:13 2022 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Sat Oct 15 09:34:13 2022 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Sat Oct 15 09:34:13 2022 daemon.info dnsmasq[1]: using only locally-known addresses for local
Sat Oct 15 09:34:13 2022 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Sat Oct 15 09:34:13 2022 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Sat Oct 15 09:34:13 2022 daemon.info dnsmasq[1]: using only locally-known addresses for mask-h2.icloud.com
Sat Oct 15 09:34:13 2022 daemon.info dnsmasq[1]: using only locally-known addresses for mask.icloud.com
Sat Oct 15 09:34:13 2022 daemon.info dnsmasq[1]: using 1 more local addresses
Sat Oct 15 09:34:13 2022 daemon.info dnsmasq[1]: read /etc/hosts - 4 addresses
Sat Oct 15 09:34:13 2022 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Sat Oct 15 09:34:27 2022 user.info : luci: accepted login on / for root from 192.168.179.198
root@pWPW10-0:~#
```
N.B. - 'vi' commands are:

1. reduce length of logfile name, increase verbosity level '2'  and uncomment
2. comment logfile and verbosity

systemcrash, 2022-10-18:
Problem remains in 22.03.2

```kernel logs
[   86.418247] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 634ebf2b
[   86.427647] epc = 77de2dc0 in libc.so[77d64000+a9000]
[   86.432809] ra  = 77de4b04 in libc.so[77d64000+a9000]
...
[   88.751136] br-lan: port 3(wlan0) entered blocking state
[   88.756655] br-lan: port 3(wlan0) entered disabled state
[   88.762531] device wlan0 entered promiscuous mode
[   89.926855] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   89.933560] br-lan: port 2(wlan1) entered blocking state
[   89.938957] br-lan: port 2(wlan1) entered forwarding state
[   91.949680] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 634ebf31
[   91.959038] epc = 77e4cdc0 in libc.so[77dce000+a9000]
[   91.964227] ra  = 77e4eb04 in libc.so[77dce000+a9000]
[   97.198637] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 634ebf36
[   97.208262] epc = 77eb4dc0 in libc.so[77e36000+a9000]
[   97.213451] ra  = 77eb6b04 in libc.so[77e36000+a9000]
[  102.401394] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 634ebfa4
[  102.410796] epc = 77dcedf4 in libc.so[77d50000+a9000]
[  102.415934] ra  = 77dd0b04 in libc.so[77d50000+a9000]
[  107.647482] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 634ebfa9
[  107.656919] epc = 77dc6dc0 in libc.so[77d48000+a9000]
[  107.662082] ra  = 77dc8b04 in libc.so[77d48000+a9000]
[  112.748180] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 634ebfae
[  112.757920] epc = 77e0edc0 in libc.so[77d90000+a9000]
[  112.763103] ra  = 77e10b04 in libc.so[77d90000+a9000]
```

systemcrash, 2022-10-18:
Here is the build log for the firmware:
```log
Build successful
STDERR

Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/22.03.2/targets/ath79/generic/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.2/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.2/ath79/generic/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/22.03.2/targets/ath79/generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.2/packages/mips_24kc/base/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.2/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.2/ath79/generic/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/22.03.2/packages/mips_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.2/packages/mips_24kc/luci/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.2/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.2/ath79/generic/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/22.03.2/packages/mips_24kc/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.2/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.2/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.2/ath79/generic/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/22.03.2/packages/mips_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.2/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.2/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.2/ath79/generic/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/22.03.2/packages/mips_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.2/packages/mips_24kc/telephony/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.2/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.2/ath79/generic/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/22.03.2/packages/mips_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.2/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.2/ath79/generic/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Pseudo file "/dev" exists in source filesystem "/home/aparcar/asu/worker1/cache/22.03.2/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/dev".
Ignoring, exclude it (-e/-ef) to override.
[mktplinkfw] rootfs offset aligned to 0x2244644
[mktplinkfw] firmware file "/home/aparcar/asu/worker1/cache/22.03.2/ath79/generic/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-22.03.2-77bb28ffdcd3-ath79-generic-tplink_archer-c7-v2-squashfs-sysupgrade.bin.new" completed
[mktplinkfw] rootfs offset aligned to 0x2244644
[mktplinkfw] firmware file "/home/aparcar/asu/worker1/cache/22.03.2/ath79/generic/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-22.03.2-77bb28ffdcd3-ath79-generic-tplink_archer-c7-v2-squashfs-factory.bin.new" completed
[mktplinkfw] rootfs offset aligned to 0x2244644
[mktplinkfw] firmware file "/home/aparcar/asu/worker1/cache/22.03.2/ath79/generic/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-22.03.2-77bb28ffdcd3-ath79-generic-tplink_archer-c7-v2-squashfs-factory-us.bin.new" completed
[mktplinkfw] rootfs offset aligned to 0x2244644
[mktplinkfw] firmware file "/home/aparcar/asu/worker1/cache/22.03.2/ath79/generic/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-22.03.2-77bb28ffdcd3-ath79-generic-tplink_archer-c7-v2-squashfs-factory-eu.bin.new" completed

STDOUT

Building images for ath79 - TP-Link Archer C7 v2
Packages: 6in4 ath10k-firmware-qca988x-ct base-files busybox ca-bundle ddns-scripts ddns-scripts-services dnsmasq dropbear firewall4 fstools https-dns-proxy kmod-ath10k-ct kmod-ath9k kmod-gpio-button-hotplug kmod-nft-offload kmod-usb-ledtrig-usbport kmod-usb2 kmod-wireguard libustream-wolfssl logd luci-app-ddns luci-app-firewall luci-app-https-dns-proxy luci-app-opkg luci-app-p910nd luci-app-uhttpd luci-app-upnp luci-app-wireguard luci-proto-ipv6 luci-proto-wireguard luci-theme-bootstrap mtd netifd nftables odhcp6c odhcpd-ipv6only opkg p910nd ppp ppp-mod-pppoe procd procd-seccomp procd-ujail swconfig tcpdump uboot-envtools uci uclient-fetch uhttpd urandom-seed urngd wireguard-tools wpad-wolfssl ath10k-firmware-qca988x-ct base-files busybox ca-bundle dnsmasq dropbear firewall4 fstools kernel kmod-ath10k-ct km [... cut ...]
	20.19% of uncompressed inode table size (59897 bytes)
Directory table size 17886 bytes (17.47 Kbytes)
	43.42% of uncompressed directory table size (41194 bytes)
Number of duplicate files found 187
Number of inodes 1823
Number of files 1458
Number of fragments 36
Number of symbolic links  221
Number of device nodes 1
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 143
Number of ids (unique uids + gids) 1
Number of uids 1
	root (0)
Number of gids 1
	qroot (0)

Calculating checksums...
```

stangri, 2023-01-16:
Sorry for neglecting this. Can you still reproduce with `2022-10-15-10`?

systemcrash, 2023-01-16:
No worries. You've a lot on your plate.

It crashes, but now not only at boot, but also later during uptime.

stangri, 2023-01-17:
It may be the mips_24kc exclusive problem, I can't (for the life of me) reproduce on x86_64. I don't even know who I can ask to have a look at it, as I don't think upstream devs use the same platform and I don't know enough about libc to troubleshoot.

systemcrash, 2023-01-18:
Something interesting:

when I restart the daemon via the GUI restart button, I consistently get a crash:
```
[604017.161423] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 63c73e68
[604017.170910] epc = 77dfedf4 in libc.so[77d80000+a9000]
[604017.176307] ra  = 77e00b04 in libc.so[77d80000+a9000]
[604373.787103] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 63c73fcc
[604373.796612] epc = 77e92df4 in libc.so[77e14000+a9000]
[604373.801841] ra  = 77e94b04 in libc.so[77e14000+a9000]
[604378.877891] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 63c73fd1
[604378.887685] epc = 77e58dc0 in libc.so[77dda000+a9000]
[604378.892931] ra  = 77e5ab04 in libc.so[77dda000+a9000]
```

The nature of the crash looks a bit better from previous openwrt builds. 

When I run an equivalent command line that I see via `ps`, e.g.
`/usr/sbin/https-dns-proxy -r https://ordns.he.net/dns-query -a 127.0.0.1 -p 5054 -b 74.82.42.42 -4 -u nobody -g nogroup` 
no crash......

My google fu shows these kinds of crashes are not exclusive to my mips platform, but possibly the platform is more sensitive to things that run OK on x86_64.

systemcrash, 2023-01-18:
If there's an equivalent version built with debug symbols and/or it runs with strace, we may get somewhere further.

systemcrash, 2023-01-18:
Just a thought: could it be the `/etc/init.d/https-dns-proxy` script which triggers this?

stangri, 2023-01-18:
I haven't looked into details on how PROCD daemonizes processes, actually out of all PROCD-using packages I maintain this is the only one with the true daemon. I'll double check the code.

stangri, 2023-03-11:
Is this still happening with the new code to kill/respawn instances on interface restarts?

DianP, 2023-03-20:
I didn't mean to bump this, but I'm also getting this error on the same devices & platform.
Even after some fresh installation.

[image]

|                         |                                                                                        |
|-------------------------|----------------------------------------------------------------------------------------|
| Model                   | TP-Link Archer C7 v2                                                                   |
| Architecture            | Qualcomm Atheros QCA9558 ver 1 rev 0                                                   |
| Target Platform         | ath79/generic                                                                          |
| Firmware Version        | OpenWrt 22.03.3 r20028-43d71ad93e / LuCI openwrt-22.03 branch git-22.361.69894-438c598 |
| Kernel Version          | 5.10.161                                                                               |
| https-dns-proxy Version | 2022-10-15-11                                                                          |

stangri, 2023-03-21:
@aarond10 @baranyaib90 any insight on what may be happening on this specific platform? There have not been any reports of this on other platforms.

aarond10, 2023-03-30:
That's really weird. Invalid read access and libc.so (if not a red herring) makes me wonder if it's a failed malloc() that isn't being handled properly (i.e. null pointer returned and then dereferenced) but the location looks valid.

I found a similar bug for another package, also Archer C7v2: https://github.com/openwrt/packages/issues/19352
Could it be a platform-wide issue?

systemcrash, 2023-03-30:
It's not inconceivable, although I cannot say I've noticed any I'll
effects. That said it's only on more recent major releases. 18 and 19 were
unaffected. Guess: compiler related.

aarond10, 2023-03-30:
That would be my guess too -- From the data I can find on QCA9558, it's a single core SoC so memory races are likely not an issue but it seems quite a coincidence that it's the same sort of problem on the same hardware.

systemcrash, 2023-03-30:
One possibility I've considered is that the memory or flash is dying. But
not having experienced any other issues, I'm doubtful that's it.

A memtest would be nice but it's an involved process.

I'm fairly sure my flash has been to hell and back tho.

Ohfalderal, 2023-04-01:
@aarond10 @systemcrash the C7 v5 is also afflicted.

systemcrash, 2023-04-16:
> @aarond10 @systemcrash the C7 v5 is also afflicted.

Probably the same underlying hardware platform. 

In any case, the symptom continues on 22.03.4. But everything is otherwise working fine. 

```
[   78.696834] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 643561b1
[   78.706235] epc = 77e00dc0 in libc.so[77d82000+a9000]
[   78.711408] ra  = 77e02b04 in libc.so[77d82000+a9000]
...
[   84.318139] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 643561b7
[   84.327535] epc = 77e34dc0 in libc.so[77db6000+a9000]
[   84.332698] ra  = 77e36b04 in libc.so[77db6000+a9000]
...
[   89.725811] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 643561bc
[   89.735310] epc = 77e38df4 in libc.so[77dba000+a9000]
[   89.740491] ra  = 77e3ab04 in libc.so[77dba000+a9000]
...
[   98.284561] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 643561c5
[   98.293977] epc = 77dc4dc0 in libc.so[77d46000+a9000]
[   98.299378] ra  = 77dc6b04 in libc.so[77d46000+a9000]
...
[  103.500905] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 643561ca
[  103.510325] epc = 77e22dc0 in libc.so[77da4000+a9000]
[  103.515466] ra  = 77e24b04 in libc.so[77da4000+a9000]
[  108.684551] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 643b3fd9
[  108.694295] epc = 77e7adc0 in libc.so[77dfc000+a9000]
[  108.699531] ra  = 77e7cb04 in libc.so[77dfc000+a9000]
[  113.766460] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 643b3fde
[  113.776216] epc = 77e5cdc0 in libc.so[77dde000+a9000]
[  113.781404] ra  = 77e5eb04 in libc.so[77dde000+a9000]
[  118.846343] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 643b3fe3
[  118.855804] epc = 77e4edc0 in libc.so[77dd0000+a9000]
[  118.861817] ra  = 77e50b04 in libc.so[77dd0000+a9000]
```

systemcrash, 2023-04-16:
So I looked at [the other linked issue](https://github.com/openwrt/packages/issues/19510#issuecomment-1283148510), and apparently:

```It was indeed related to the change in time_t (32 bits to 64 bits on a 32 bit system) with musl.```

@stangri - perhaps there is some useful info in [the commit](https://github.com/openwrt/packages/commit/f7d592607f876a64d4e641e57d6300ab99d1e9e9)? Hint: change time types e.g. from `long int` to `time_t`  (altho I don't immediately see those in the source).

Ohfalderal, 2023-04-17:
@systemcrash - logging *generally* includes timestamps. Doh!

stangri, 2023-04-18:
> So I looked at [the other linked issue](https://github.com/openwrt/packages/issues/19510#issuecomment-1283148510), and apparently:
>
> `It was indeed related to the change in time_t (32 bits to 64 bits [...]

@aarond10 I'm out of my depth here, anything useful in the link above?

aarond10, 2023-04-30:
I'm still quite confused. There is no use of time_t in https_dns_proxy. I
did a bit of a cursory search on github for libev and libcares but can't
find direct references to time_t in either of those as well.

schmaller, 2023-05-22:
After compiling the package with warning output one can see the relevant fprintf calls.
Fixing this warnings by switching to %lld and casting the parameter to (long long) solved the issue for me.

```
/usr/src/OpenWRT/openwrt/build_dir/target-mips_24kc_musl/https-dns-proxy-2022-10-15/src/logging.c: In function '_log':
/usr/src/OpenWRT/openwrt/build_dir/target-mips_24kc_musl/https-dns-proxy-2022-10-15/src/logging.c:80:24: warning: format '%ld' expects argument of type 'long int', but argument 4 has type 'time_t' {aka 'long long int'} [-Wformat=]
   80 |   fprintf(logf, "%s %8ld.%06ld %s:%d ", SeverityStr[severity], tv.tv_sec,
      |                     ~~~^                                       ~~~~~~~~~
      |                        |                                         |
      |                        long int                                  time_t {aka long long int}
      |                     %8lld
/usr/src/OpenWRT/openwrt/build_dir/target-mips_24kc_musl/https-dns-proxy-2022-10-15/src/logging.c:80:30: warning: format '%ld' expects argument of type 'long int', but argument 5 has type 'suseconds_t' {aka 'long long int'} [-Wformat=]
   80 |   fprintf(logf, "%s %8ld.%06ld %s:%d ", SeverityStr[severity], tv.tv_sec,
      |                          ~~~~^
      |                              |
      |                              long int
      |                          %06lld
   81 |           tv.tv_usec, file, line);
      |           ~~~~~~~~~~
      |             |
      |             suseconds_t {aka long long int}
```

schmaller, 2023-05-23:
@aarond10 
Is my fix also suitable for upstream patching of your (great :wink:)) project ?
If so, do you want to patch it yourself to be conform with all the guidelines, or should I try to provide a PR ?

aarond10, 2023-05-23:
Ah! Of course! It's the logging code... 
Thanks for the fix. I think it'll need a little more conditional handling though for cases where time_t is still 32-bit.

systemcrash, 2023-05-24:
Just cast everything to 64 bit?

aarond10, 2023-05-24:
Yeah, that would do it. :)

aarond10, 2023-05-25:
Done.

systemcrash, 2023-06-08:
Hmm - still get a few of these when I restart the proxy on 22.03.4

```
[4657909.666372] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 64825294
[4657909.676309] epc = 77e0cdf4 in libc.so[77d8e000+a9000]
[4657909.681655] ra  = 77e0eb04 in libc.so[77d8e000+a9000]
[4657914.709335] do_page_fault(): sending SIGSEGV to https-dns-proxy for invalid read access from 64825299
[4657914.719361] epc = 77de6dc0 in libc.so[77d68000+a9000]
[4657914.724743] ra  = 77de8b04 in libc.so[77d68000+a9000]
```

I ran:
```
opkg update
opkg upgrade https-dns-proxy

Upgrading https-dns-proxy on root from 2022-10-15-12 to 2023-05-25-1...
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/mips_24kc/packages/https-dns-proxy_2023-05-25-1_mips_24kc.ipk
Stopping https-dns-proxy 2022-10-15-12 ✓
Configuring https-dns-proxy.
Starting https-dns-proxy 2023-05-25-1 ✓✓✓
```

Thought perhaps a libc upgrade might also help, but:
```
opkg upgrade libc
Not upgrading package libc which is marked hold (flags=0x302).
```

stangri, 2023-06-11:
@aarond10 any insight?

Ohfalderal, 2023-07-02:
@stangri - updated a 22.03.0 TP-Link C7 and raised the verbosity... and immediately see segmentation faults logged.

systemcrash, 2023-10-16:
So, I just spent a while getting my hands dirty. And basically, on master, the patch for this fix simply does not take effect. 

@stangri https-dns-proxy in master produces SIGSEGV every time. 

@schmaller's proposed fix *does* actually work, when it is *applied*. 

Running this:
```
strace -yy -v -t /usr/sbin/https-dns-proxy -v debug -b 1.1.1.1,1.0.0.1,2606:4700:4700::1111,2606:4700:4700::1001 -u nobody -g nobody -p 5053 -r https://ordns.he.net -q
```
without the patch produces:
```
getuid()                                = 0
ioctl(1</dev/pts/0<char 136:0>>, TIOCGWINSZ, 0x7f813e08) = 0
writev(1</dev/pts/0<char 136:0>>, [{iov_base="", iov_len=0}, {iov_base="----------------------------\nWAR"..., iov_len=102}], 2----------------------------
WARNING: Running as root without dropping privileges is NOT recommended.
) = 102
writev(1</dev/pts/0<char 136:0>>, [{iov_base="----------------------------", iov_len=28}, {iov_base="\n", iov_len=1}], 2----------------------------
) = 29
nanosleep({tv_sec=1, tv_nsec=0}, 0x7f813e50) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x77e30000
fcntl64(1</dev/pts/0<char 136:0>>, F_GETFL) = 0x200a (flags O_RDWR|O_APPEND|O_LARGEFILE)
ioctl(1</dev/pts/0<char 136:0>>, TIOCGWINSZ, 0x7f813ec8) = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x652c61ef} ---
+++ killed by SIGSEGV +++
Segmentation fault
```

with the patch produces:
```
getuid()                                = 0
ioctl(1</dev/pts/0<char 136:0>>, TIOCGWINSZ, 0x7fb107f0) = 0
writev(1</dev/pts/0<char 136:0>>, [{iov_base="", iov_len=0}, {iov_base="----------------------------\nWAR"..., iov_len=102}], 2----------------------------
WARNING: Running as root without dropping privileges is NOT recommended.
) = 102
writev(1</dev/pts/0<char 136:0>>, [{iov_base="----------------------------", iov_len=28}, {iov_base="\n", iov_len=1}], 2----------------------------
) = 29
nanosleep({tv_sec=1, tv_nsec=0}, 0x7fb10838) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x77e13000
fcntl64(1</dev/pts/0<char 136:0>>, F_GETFL) = 0x200a (flags O_RDWR|O_APPEND|O_LARGEFILE)
ioctl(1</dev/pts/0<char 136:0>>, TIOCGWINSZ, 0x7fb108a8) = 0
writev(1</dev/pts/0<char 136:0>>, [{iov_base="[W] 1697419512.446313 main.c:236"..., iov_len=75}, {iov_base="\n", iov_len=1}], 2[W] 1697419512.446313 main.c:236 HTTP/3 is not supported by current libcurl
) = 76
getuid()                                = 0
geteuid()                               = 0
getgid()                                = 0
getegid()                               = 0
...
```

A compile from a fresh checkout of master still produces:

```
/home/user/openwrt/build_dir/target-mips_24kc_musl/https-dns-proxy-2023-05-25/src/logging.c: In function '_log':
/home/user/openwrt/build_dir/target-mips_24kc_musl/https-dns-proxy-2023-05-25/src/logging.c:81:24: warning: format '%lu' expects argument of type 'long unsigned int', but argument 4 has type 'long long unsigned int' [-Wformat=]
   81 |   fprintf(logf, "%s %8lu.%06lu %s:%d ", SeverityStr[severity],
      |                     ~~~^
      |                        |
      |                        long unsigned int
      |                     %8llu
   82 |           (uint64_t)tv.tv_sec,
      |           ~~~~~~~~~~~~~~~~~~~
      |           |
      |           long long unsigned int
/home/user/openwrt/build_dir/target-mips_24kc_musl/https-dns-proxy-2023-05-25/src/logging.c:81:30: warning: format '%lu' expects argument of type 'long unsigned int', but argument 5 has type 'long long unsigned int' [-Wformat=]
   81 |   fprintf(logf, "%s %8lu.%06lu %s:%d ", SeverityStr[severity],
      |                          ~~~~^
      |                              |
      |                              long unsigned int
      |                          %06llu
   82 |           (uint64_t)tv.tv_sec,
   83 |           (uint64_t)tv.tv_usec, file, line);
      |           ~~~~~~~~~~~~~~~~~~~~
      |           |
      |           long long unsigned int
```

So putting @schmaller's patch into play is a fix. Or @aarond10 fixing the `logging.c` code to the following is also a fix:

```c
  fprintf(logf, "%s %8llu.%06llu %s:%d ", SeverityStr[severity],
          (uint64_t)tv.tv_sec,
          (uint64_t)tv.tv_usec, file, line);
```

stangri, 2023-10-16:
@aarond10 @baranyaib90 Can you fix the code upstream or do I need to apply patch when making OpenWrt binaries?

stangri, 2023-10-16:
Thanks @schmaller @systemcrash I've created a patch in my repo: https://github.com/stangri/source.openwrt.melmac.net/commit/6c4bad94c9be2bdcd4ce62c658f4a7b78adc9d30, if upstream can't merge the PR with the fix, I'll merge the patch to OpenWrt repo.

systemcrash, 2023-10-17:
So, just my $0.02, I think that we can take the 030 patch file in the meantime. And when https-dns-proxy eventually patches up if they choose, the patch will become redundant (and can be removed).

PR #21177 (2023-05-26): https-dns-proxy: update to 2023-05-25-1
Maintainer: me
Compile tested: x86_64, Sophos XG-135r3, OpenWrt 22.03.5
Run tested: x86_64, Sophos XG-135r3, OpenWrt 22.03.5, test name resolution

Description:
* update to a new upstream commit, fixes #19366
* update patches/010-cmakelists-remove-cflags.patch as upstream file was update
* remove patches/020-cmakelists-add-version.patch as version is now set elsewhere
* add patches/020-src-options.c-add-version.patch to set the version information
* adjust PROCD START time to 95
commit eb40aa1b (2023-05-26): https-dns-proxy: update to 2023-05-25-1

* update to a new upstream commit, fixes #19366
* update patches/010-cmakelists-remove-cflags.patch as upstream file was update
* remove patches/020-cmakelists-add-version.patch as version is now set elsewhere
* add patches/020-src-options.c-add-version.patch to set the version information
* adjust PROCD START time to 95
PR #21179 (2023-05-26): [23.05] https-dns-proxy: update to 2023-05-25-1
Maintainer: me
Compile tested: x86_64, Sophos XG-135r3, OpenWrt 22.03.5
Run tested: x86_64, Sophos XG-135r3, OpenWrt 22.03.5, test name resolution

Description:
* update to a new upstream commit, fixes #19366
* update patches/010-cmakelists-remove-cflags.patch as upstream file was update
* remove patches/020-cmakelists-add-version.patch as version is now set elsewhere
* add patches/020-src-options.c-add-version.patch to set the version information
* adjust PROCD START time to 95

(cherry picked from commit eb40aa1b5dc6b5945b8b53cd5a2cc06303a3d9eb [... cut]
commit 25883f71 (2023-05-26): https-dns-proxy: update to 2023-05-25-1

* update to a new upstream commit, fixes #19366
* update patches/010-cmakelists-remove-cflags.patch as upstream file was update
* remove patches/020-cmakelists-add-version.patch as version is now set elsewhere
* add patches/020-src-options.c-add-version.patch to set the version information
* adjust PROCD START time to 95

(cherry picked from commit eb40aa1b5dc6b5945b8b53cd5a2cc06303a3d9eb)