openwrt/packages #19376: transmission-daemon: Couldn't save file "/tmp/transmission/settings.json": Resource busy (variant.c:1226)

idarek, 2022-09-15:
`<package-name>: transmission-daemon`

# Issue

```
Couldn't save file "/tmp/transmission/settings.json": Resource busy (variant.c:1226)
```

When starting daemon.

Removing `procd-ujail` from system fix it temporarely but when added torrent to download it crashes daemon and autostart it on its own.

neheb, 2022-09-17:
Yep. All this jail garbage has been causing issues since introduction. Maintainer is @dangowrt

dangowrt, 2022-09-18:
The error message
```
daemon.err transmission-daemon[16422]: [2022-09-15 19:28:11.861] Couldn't save temporary file "/tmp/transmission/settings.json.tmp.XXXXXX": Permission denied (variant.c:1235)
```
should not prevent transmission from working (and it doesn't for me). It merely means that transmission could not make changes to it's the configuration, which on OpenWrt is generated by UCI and should be managed there.
You shouldn't need to deactivate all `procd-ujail` based isolation to get transmission working.
Which problems (apart from the log entry) do you observe? Do you see the transmission process running in `ps` output? Do you see it listening on control, dht and torrent ports in `netstat -lnp` output?

@neheb If you prefer running transmission on your home device, potentially handling other sensitive data as well, only relying on a little bit of discrete UNIX access control, ok, go ahead and uninstall `procd-ujail` or comment-out the relevant parts of the transmission init script. Having it sit in it's own filesystem namespace, isolated from other processes and guarded by seccomp BPF filter definitely makes me feel a bit better, given that it's typically available for unsolicited TCP/UDP connections from the outside and/or uses UDP hole punching to communicate with random peers suggested by the DHT even if you don't give it an open port on a public address.

neheb, 2022-09-18:
I don't run transmission.

Anyway, my point was that ever since its introduction to transmission, I've seen multiple people report issues.

dangowrt, 2022-09-23:
`procd-ujail` is a OpenWrt-specific way to isolate processes from the rest of the system using standard Linux Kernel interfaces designed for that purpose. In case of transmission it is used to hide unrelated filesystems away from it (using a filesystem namespace and bind mounts, ie. `procd_add_jail_mount`). In addition to that, seccomp is used to strictly limit the allowed interaction of the transmission process with the operating system (seccomp is like a firewall between an app and the OS).

Downloading files should work fine also with `ujail`. If you are trying to download into a directory other than the configured `download_dir` this won't work. As a hack, and to find out if this is the problem, add another `procd_add_jail_mount_rw` line to /etc/init.d/transmission for the directory you want to give transmission write access to. As many people seem to have the need to give transmission read-only or even read-write access to additional directories I will add UCI options in /etc/config/transmission for that purpose.

Regarding storing configuration: Letting transmission edit it's own configuration is intentionally not permitted as in OpenWrt the configuration is **anyway generated again from what is in /etc/config/transmission** on every restart. Temporary configuration changes, such as a temporary change of speed limits, do work despite the file not being written.

idarek, 2022-09-23:
Thank you @dangowrt for explanation. Appreciated.

Will need to investigate why, despite, that though settings, the temporary directory shall have jail RW access, why is returning "Not enough disk space". There is something interesting happening here.

jly77, 2022-11-05:
> 

Same here. Any suggestion or workaound available?

alpha7504, 2022-11-19:
@jly77 hi
Could you please explain how you solved this issue?

geser2008, 2022-11-30:
I have added these lines to /etc/config/transmission

        procd_add_jail_mount_rw "$config_dir/incomplete"
        procd_add_jail_mount_rw "$config_dir/autoload"
        procd_add_jail_mount_rw "$config_dir/dht.dat"

In the config there is also a directory for blacklist files. I think it should be added to somehow.

geser2008, 2022-12-03:
Transmission can't also write "dht.dat" ans "stats.json" files. Also "watch_dir" and "icompltete_dir" are not available and when I've activated them, transmission has stooped working. So I've modified /etc/init.d/transmission a little bit. Not sure that the result is completely correct and secure, but now it works without warnings and errors concerning file/folder permissions.

alpha7504, 2022-12-03:
Thanks for the response... I've checked that using your init.d file all the errors disappeared but this seems like a temporary solution. I wanna ask, how do I implement this in my own transmission init.d file?

geser2008, 2022-12-03:
I've added this block

###----start adding watch dir----###
 
    local watch_dir
    config_get watch_dir "$cfg" 'watch_dir' '/var/etc/transmission'
    local watch_dir_enabled
    config_get watch_dir_enabled "$cfg" 'watch_dir_enabled' 0
 
###----end adding watch dir----###

and modified this one

procd_add_jail transmission log
    procd_add_jail_mount_rw "$config_dir"
#   procd_add_jail_mount "$config_file"
#   procd_add_jail_mount_rw "$config_dir/resume"
#   procd_add_jail_mount_rw "$config_dir/torrents"
#   procd_add_jail_mount_rw "$config_dir/blocklists"
#   procd_add_jail_mount_rw "$config_dir/stats.json"
    procd_add_jail_mount_rw "$incomplete_dir"
    procd_add_jail_mount_rw "$watch_dir"
    procd_add_jail_mount_rw "$download_dir"
    web_home="${web_home:-/usr/share/transmission/web}"
    [ -d "$web_home" ] && procd_add_jail_mount "$web_home"
    [ -f "$ca_bundle_file" ] && procd_add_jail_mount "$ca_bundle_file"
    procd_close_instance
 
}

Ulysses1960, 2023-01-15:
Thank you for this helpful informations. As I wanted to be able change the download location from the Web user interface for some kind of torrents, I had to add another line to my /etc/init.d/transmission file:

**procd_add_jail_mount_rw /media/MyDisk/share/Music**

The address must be hardcoded, so there is no flexibility for the user.

I would be thankful if these issues could be fixed for the 2023 release of OpenWrt/transmission.

primoitt83, 2023-01-29:
hey guys..

I've installed transmission 3.00-18 on Openwrt 22.03.2 and my daemon wont work... looks like permission issue...

Here is my config briefly:

````
cat /etc/config/transmission
config transmission
	option enabled 1
	option config_dir '/etc/transmission'
	option config_overwrite '1'
	option user 'transmission'
	option group 'transmission'
````
Here is my workaround:

````
/etc/init.d/transmission stop
rm -Rf /etc/transmission
mkdir /etc/transmission
transmission-daemon -g /etc/transmission -f --log-debug
````
Waiting some minutes...then stop

````
cd /etc
chown transmission:224 transmission/ -R
chmod 775 transmission/ -R
/etc/init.d/transmission start
````

Now transmission-daemon goes up..

On logread still got this messages but transmission is working nicely

````
daemon.err transmission-daemon[2300]: [2023-01-28 07:53:59.535] Couldn't save temporary file "/etc/transmission/stats.json.tmp.XXXXXX": Permission denied (variant.c:1235)
````