openwrt/packages #19382: curl add http/3 support by default

user8446, 2022-09-16:
Maintainer: @stangri 

Hi,

[cURL OpenWrt package](https://github.com/openwrt/packages/tree/openwrt-21.02/net/curl) currently does not support HTTP/3 support but should. This protocol is now widely used everywhere, is [supported in all major browsers](https://caniuse.com/http3), and has many advantages over HTTPS/2 including speed and privacy.

For example [https-dns-proxy](https://github.com/openwrt/packages/tree/openwrt-21.02/net/https-dns-proxy) can now use HTTP/3 for DNS resolvers that use it. DNS is extremely latency sensitive in which a faster protocol would benefit it and be noticeable on pageload and responsiveness on OpenWrt connected devices.

stangri, 2022-09-16:
@user8446 Thanks for bringing it up.

Looking at [support for HTTP/3 in curl](https://curl.se/docs/http3.html) it is experimental and requires:
1. patched OpenSSL library.
2. two more 3rd party libraries: libngtcp2 and libnghttp3.

Which brings the following concerns to mind:
1. OpenSSL is rather large and by default curl now depends on mbedtls (used to on WolfSSL, used by wpad too) due to the space constraints on most devices OpenWrt is available for.
2. Also space concerns over two extra libraries.

So with the current state I don't think it would be accepted as a default curl build option to include support for HTTP/3.

However if you or anyone else wants to contribute the PR to enable building curl with HTTP/3 support as an option for any of the existing SSL libraries in OpenWrt, I'd gladly test it and advocate for merging said PR.

user8446, 2022-09-16:
Hi @stangri thanks for the reply. Not sure if it would better but there is also [LibreSSL](https://www.libressl.org/).

I don't have the knowledge to add HTTP/3 so the best I could do is donate to fund adding it.

sbwml, 2023-08-19:
quictls + libngtcp2 + libnghttp3, it works well

[image]

user8446, 2023-08-19:
@sbwml That is great! How did you build this? Was hoping openssl3 in master has quic but it doesn't:

```
root@OpenWrt:~# openssl version
OpenSSL 3.0.10 1 Aug 2023 (Library: OpenSSL 3.0.10 1 Aug 2023)
```

sbwml, 2023-08-19:
```
 -----------------------------------------------------
 OpenWrt 23.05.0-rc2, r23228-cd17d8df2a
 -----------------------------------------------------
[root@OpenWrt ~]# openssl version
OpenSSL 3.0.10+quic 1 Aug 2023 (Library: OpenSSL 3.0.10+quic 1 Aug 2023)
[root@OpenWrt ~]#
[root@OpenWrt ~]# openssl version -a
OpenSSL 3.0.10+quic 1 Aug 2023 (Library: OpenSSL 3.0.10+quic 1 Aug 2023)
built on: Mon Jun 26 11:20:39 2023 UTC
platform: linux-aarch64-openwrt
options:  bn(64,64)
compiler: aarch64-openwrt-linux-musl-gcc -fPIC -pthread -Wa,--noexecstack -Wall -O3 -Wl,--gc-sections -pipe -march=armv8-a+crc+crypto -fno-caller-saves -fno-plt -fhonour-copts -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Ofast -DPIC -fPIC -pipe -march=armv8-a+crc+crypto -fno-caller-saves -fno-plt -fhonour-copts -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -Ofast -fPIC -znow -zrelro -DOPENSSL_USE_NODELETE -DOPENSSL_PIC -DOPENSSL_BUILDING_OPENSSL -DNDEBUG -D_FORTIFY_SOURCE=1 -DPIC -DDEVRANDOM="\"/dev/hwrng\""
OPENSSLDIR: "/etc/ssl"
ENGINESDIR: "/usr/lib/engines-3"
MODULESDIR: "/usr/lib/ossl-modules"
Seeding source: os-specific
CPUINFO: OPENSSL_armcap=0x3d
[root@OpenWrt ~]#
```

sbwml, 2023-08-19:
Because I built OpenWrt for myself, I don't need to follow the code specifications. I replaced openssl pkg with the quitls source.

user8446, 2023-08-19:
Thank you for sharing!

sbwml, 2023-08-19:
If you're willing to break the OpenWrt source code, you can do it.

```shell
# get openwrt source
git clone https://github.com/openwrt/openwrt --depth=1 -b v23.05.0-rc2
cd openwrt
./scripts/feeds update -a
./scripts/feeds install -a

# openssl -> quictls
rm -rf package/libs/openssl
git clone https://github.com/sbwml/package_libs_openssl package/libs/openssl

# nghttp3
git clone https://github.com/sbwml/package_libs_nghttp3 package/libs/nghttp3

# ngtcp2
git clone https://github.com/sbwml/package_libs_ngtcp2 package/libs/ngtcp2

# curl - http3/quic
rm -rf feeds/packages/net/curl
git clone https://github.com/sbwml/feeds_packages_net_curl feeds/packages/net/curl
```

Add this to `.config`
```
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_libcurl=y
CONFIG_LIBCURL_NGHTTP3=y
CONFIG_LIBCURL_NGTCP2=y
CONFIG_LIBCURL_OPENSSL=y
# CONFIG_LIBCURL_GNUTLS is not set
# CONFIG_LIBCURL_MBEDTLS is not set
# CONFIG_LIBCURL_NOSSL is not set
# CONFIG_LIBCURL_WOLFSSL is not set
```

At this point, you can build a curl that supports h3/quic.

user8446, 2023-08-19:
I haven't done this before but I will research. Thank you for the info!

stangri, 2023-08-19:
> If you're willing to break the OpenWrt source code, you can do it.

Would you consider sending PRs for existing openssl and curl packages and the new libnghttp3 and libngtcp2 packages?

sbwml, 2023-08-19:
> > If you're willing to break the OpenWrt source code, you can do it.
> 
> @sbwml Would you consider sending PRs for existing openssl and curl packages and the new libnghttp3 and libngtcp2 packages?

😛 It's too difficult for me, I don't know English and I'm not familiar with using git, these are the biggest obstacles

-- Translated from Youdao

stangri, 2023-08-19:
@sbwml thanks for your prompt reply. Would you be OK with me submitting PRs crediting you for the code/testing?

sbwml, 2023-08-19:
> @sbwml thanks for your prompt reply. Would you be OK with me submitting PRs crediting you for the code/testing?

Sure 👍

user8446, 2023-08-19:
Thank you @stangri for looking into it, http3 will definitely be a benefit with DNS being so sensitive to latency

stangri, 2023-08-21:
Sorry, just noticed this uses a fork of OpenSSL, I'm pretty sure this will not be accepted to replace the official OpenSSL in OpenWrt. I'll try to reach to maintainer to see if this can be added as a variant.

stangri, 2023-09-25:
@sbwml have you tested/made updates to work with 23.05.0-rc3 and curl 8.3.0?

sbwml, 2023-09-26:
> @sbwml have you tested/made updates to work with 23.05.0-rc3 and curl 8.3.0?

I am currently using OpenWrt-23.05-rc3 and curl/8.2.1.

```
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05.0-rc3, r23389-5deed175a5
 -----------------------------------------------------
[root@OpenWrt ~]# curl --version
curl 8.2.1 (aarch64-openwrt-linux-gnu) libcurl/8.2.1 OpenSSL/3.0.10 nghttp2/1.51.0 ngtcp2/0.18.0 nghttp3/0.14.0
Release-Date: 2023-07-26
Protocols: file ftp ftps http https mqtt
Features: alt-svc HSTS HTTP2 HTTP3 HTTPS-proxy IPv6 Largefile SSL threadsafe UnixSockets
[root@OpenWrt ~]#
```

sbwml, 2023-09-26:
https://x86.cooluc.com/minimal/openwrt-23.05/v23.05.0-rc3/manifest.txt?preview

stangri, 2023-09-26:
@sbwml following your instructions I get the error on building curl with ngtcp2 with 23.05.0-rc3. Any idea on how to overcome this?

```
checking for nghttp2_session_get_stream_local_window_size in -lnghttp2... yes
checking for nghttp2/nghttp2.h... yes
configure: Added /home/stangri/development/Builder/openwrt/staging_dir/target-x86_64_musl/usr/lib to CURL_LIBRARY_PATH
checking for libngtcp2 options with pkg-config... found
configure: -l is -lngtcp2
configure: -I is -I/home/stangri/development/Builder/openwrt/staging_dir/target-x86_64_musl/usr/include
configure: -L is -L/home/stangri/development/Builder/openwrt/staging_dir/target-x86_64_musl/usr/lib
checking for ngtcp2_conn_client_new_versioned in -lngtcp2... yes
checking for ngtcp2/ngtcp2.h... yes
configure: Added  to CURL_LIBRARY_PATH
checking for libngtcp2_crypto_quictls options with pkg-config... no
configure: error: --with-ngtcp2 was specified but could not find ngtcp2_crypto_quictls pkg-config file.
make[3]: *** [Makefile:192: /home/stangri/development/Builder/openwrt/build_dir/target-x86_64_musl/curl-8.2.1/.configured_73e8a40fec0d702ff96a3dc5ec6098da] Error 1
make[3]: Leaving directory '/home/stangri/development/Builder/openwrt/feeds/packages/net/curl'
time: package/feeds/packages/curl/compile#41.67#1.51#43.63
    ERROR: package/feeds/packages/curl failed to build.
make[2]: *** [package/Makefile:120: package/feeds/packages/curl/compile] Error 1
make[2]: Leaving directory '/home/stangri/development/Builder/openwrt'
make[1]: *** [package/Makefile:114: /home/stangri/development/Builder/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/stangri/development/Builder/openwrt'
make: *** [/home/stangri/development/Builder/openwrt/include/toplevel.mk:232: world] Error 2
```

sbwml, 2023-09-26:
> configure: error: --with-ngtcp2 was specified but could not find ngtcp2_crypto_quictls pkg-config file.

ngtcp2 also needs to be based on quicssl header files and libraries. It cannot be built directly in the original openwrt source code because it requires https://github.com/quictls/openssl .

I was able to build it because of tampering with openwrt's openssl package. https://github.com/sbwml/package_libs_openssl

sbwml, 2023-09-26:
There are also examples of using quickls from the ngtcp2 readme file. https://github.com/ngtcp2/ngtcp2#build-from-git

stangri, 2023-09-26:
> ngtcp2 also needs to be based on quicssl header files and libraries. It cannot be built directly in the original openwrt source code because it requires https://github.com/quictls/openssl .
>
> I was [...]

Yes, I was using libs/curl from your repo as per your comment above: https://github.com/openwrt/packages/issues/19382#issuecomment-1685022455. I'm still getting an error when building. I wonder if there are steps you've left out in the instructions.

sbwml, 2023-09-26:
> Yes, I was using libs/curl from your repo as per your comment above: [#19382 (comment)](https://github.com/openwrt/packages/issues/19382#issuecomment-1685022455). I'm still getting an error when building. I wonder if there are steps you've left out in the instructions.

Nothing was missing. from your error logs, it doesn't seem to use quictls when compiling ngtcp2, because it can't find libngtcp2_crypto_quictls.pc. I guess the source code of your build package is not cloned from https://github.com/openwrt/openwrt

It's more like using the OpenWrt sdk.

sbwml, 2023-09-26:
If you use openwrt sdk to test compile it, you should replace openssl from feeds.

```shell
rm -rf feeds/base/package/libs/openssl
git clone https://github.com/sbwml/package_libs_openssl feeds/base/package/libs/openssl
```

abelian424, 2023-12-16:
The only custom package needed now is quictls i.e.
`# openssl -> quictls
rm -rf package/libs/openssl
git clone https://github.com/sbwml/package_libs_openssl package/libs/openssl`

Unfortunately the default settings don't allow it to build. If you use `git checkout dev` and edit the Makefile you can successfully build openssl/quictls:
`PKG_NAME:=openssl
PKG_VERSION:=3.0.12
PKG_RELEASE:=1
PKG_BUILD_FLAGS:=no-mips16 gc-sections no-lto

PKG_BUILD_PARALLEL:=1

PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
PKG_SOURCE_PROTO:=git
PKG_SOURCE_URL:=https://github.com/sbwml/openssl.git
PKG_SOURCE_VERSION:=9e001c32f3b0a22ce9968ef27850c30e8a8e3ee1
PKG_MIRROR_HASH:=f93c9e8edde5e9166119de31755fc87b4aa34863662f67ddfcba14d0b6b69b61`

Leaving at default branch and changing out the quictls package source url to the official versions possibly works but I haven't tried. It's easy enough to get the sha256 hashes for each available release though.

sbwml, 2023-12-16:
> The ~only~ EDIT: patched curl and quictls are custom package needed now i.e.
> 

I am currently patching the openssl package of openwrt to implement quic support. https://github.com/sbwml/r4s_build_script/tree/master/openwrt/patch/openssl/quic

Also I cannot directly use nghttp3 and ngtcp2 provided by openwrt feeds.

https://github.com/sbwml/r4s_build_script/blob/master/openwrt/scripts/00-prepare_base.sh#L206
https://github.com/sbwml/r4s_build_script/blob/master/openwrt/scripts/00-prepare_base.sh#L210

```
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05.2, r23630-842932a63d
 -----------------------------------------------------
[root@OpenWrt ~]# 
[root@OpenWrt ~]# openssl version -a
OpenSSL 3.0.12+quic 24 Oct 2023 (Library: OpenSSL 3.0.12+quic 24 Oct 2023)
built on: Tue Nov 14 13:38:11 2023 UTC
platform: linux-aarch64-openwrt
options:  bn(64,64)
compiler: aarch64-openwrt-linux-musl-gcc -fPIC -pthread -Wa,--noexecstack -Wall -O3 -Wl,--gc-sections -pipe -march=armv8-a+crc+crypto -fno-caller-saves -fno-plt -fhonour-copts -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Ofast -DPIC -fPIC -pipe -march=armv8-a+crc+crypto -fno-caller-saves -fno-plt -fhonour-copts -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -Ofast -fPIC -znow -zrelro -DOPENSSL_USE_NODELETE -DOPENSSL_PIC -DOPENSSL_BUILDING_OPENSSL -DNDEBUG -D_FORTIFY_SOURCE=1 -DPIC -DDEVRANDOM="\"/dev/hwrng\""
OPENSSLDIR: "/etc/ssl"
ENGINESDIR: "/usr/lib/engines-3"
MODULESDIR: "/usr/lib/ossl-modules"
Seeding source: os-specific
CPUINFO: OPENSSL_armcap=0x3d
[root@OpenWrt ~]# 
[root@OpenWrt ~]# curl --version
curl 8.5.0 (aarch64-openwrt-linux-gnu) libcurl/8.5.0 quictls/3.0.12 nghttp2/1.57.0 ngtcp2/1.1.0 nghttp3/1.1.0
Release-Date: 2023-12-06
Protocols: file ftp ftps http https mqtt
Features: alt-svc HSTS HTTP2 HTTP3 HTTPS-proxy IPv6 Largefile SSL threadsafe UnixSockets
[root@OpenWrt ~]# 
[root@OpenWrt ~]# curl -I --http3-only https://tls.cooluc.com
HTTP/3 200 
server: nginx/1.25.3 (Red Hat Enterprise Linux release 9.2 (Plow))
date: Sat, 16 Dec 2023 10:11:25 GMT
content-type: text/html; charset=utf-8
content-length: 5633
last-modified: Fri, 16 Dec 2022 21:22:34 GMT
vary: accept-encoding
etag: "639ce19a-1601"
alt-svc: h3=":443"; ma=2592000,h3-29=":443"; ma=2592000
x-quic: h3
x-xss-protection: 1
x-xss-protection: mod=block
strict-transport-security: max-age=63072000
public-key-pins: pin-sha256="BbrVIhEYvvBL6FiyC7nzVKLLDU3GPYdqHWAfk0ev/80="; pin-sha256="amMeV6gb9QNx0Zf7FtJ19Wa/t2B7KpCF/1n2Js3UuSU="; max-age=31536000; includeSubDomains
accept-ranges: bytes

[root@OpenWrt ~]#
```

stangri, 2023-12-16:
> Also I cannot directly use nghttp3 and ngtcp2 provided by openwrt feeds.

Why? Have you looked into patching the sources from openwrt feeds?

sbwml, 2023-12-16:
> > Also I cannot directly use nghttp3 and ngtcp2 provided by openwrt feeds.
> 
> Why? Have you looked into patching the sources from openwrt feeds?

因为你使用了cmake，而它又缺少Dev

sbwml, 2023-12-16:
> > Also I cannot directly use nghttp3 and ngtcp2 provided by openwrt feeds.
> 
> Why? Have you looked into patching the sources from openwrt feeds?

因为 cmake

abelian424, 2023-12-17:
> I am currently patching the openssl package of openwrt to implement quic support. https://github.com/sbwml/r4s_build_script/tree/master/openwrt/patch/openssl/quic
>
> Also I cannot directly use nghttp3 [...]

I find only superficial differences between curl, nghttp3, and ngtcp2:

https://github.com/sbwml/feeds_packages_net_curl/blob/main/Config.in
https://github.com/openwrt/packages/blob/master/net/curl/Config.in

https://github.com/sbwml/package_libs_nghttp3/blob/main/Makefile
https://github.com/openwrt/packages/blob/master/libs/nghttp3/Makefile

https://github.com/sbwml/package_libs_ngtcp2/blob/main/Makefile
https://github.com/openwrt/packages/blob/master/libs/ngtcp2/Makefile

It is a case of OpenWrt catching up to your packages (and in the case of curl, outpacing by being 8.5.0 instead of 8.4.0)?

sbwml, 2023-12-17:
> It is a case of OpenWrt catching up to your packages (and in the case of curl, outpacing by being 8.5.0 instead of 8.4.0)?

curl 的版本不是重点，我在GitHub Action 为自己构建 openwrt 固件时始终取自 openwrt/packages main 分支的 curl，它的版本是 [8.5.0](https://github.com/openwrt/packages/commit/6501290c17fb2c65e0fb589da94d121ff89e7d5e)。

> What is due to cmake?

在编译 curl 并启用 ngtcp2 库支持时，使用 cmake 编译 ngtcp2 & nghttp3，那么 curl 将会编译失败（我没有保留任何日志，因为已经过去三个月了，你可以自己试试），我第一次让 curl 支持 http3 时就是使用的 cmake 去编译 ngtcp2 & nghttp3 ，但是当时 curl 编译失败了，最后我使用默认的 automake 编译成功了。毕竟我的需求只是让 curl 实现 quic/http3 的支持，而不是使用什么方式去编译它，所以并没有进一步的去了解原因。

PS：来自手机的回复，没有任何方式将其翻译成英文

sbwml, 2023-12-17:
> What is due to cmake?

印象中好像是会让 curl 无法正确找到 ngtcp2 & nghttp3 的头文件（header file）

sbwml, 2023-12-17:
> I'm going to try to compile without this patch but with [all these patches](https://github.com/sbwml/r4s_build_script/tree/master/openwrt/patch/openssl/quic) and see what happens.

这已经足够了！

```diff
--- a/VERSION.dat
+++ b/VERSION.dat
@@ -4,4 +4,4 @@ PATCH=10
 PRE_RELEASE_TAG=
 RELEASE_DATE="1 Aug 2023"
-SHLIB_VERSION=81.3
+SHLIB_VERSION=3
```

该补丁仅仅是改变了 shared library 的版本命名，因为 quictls 上游的仓库会将 SHLIB 版本更改为 81.3 （即 `libssl.so.81.3`），这样会导致 openwrt 其它依赖 openssl 库的程序出现问题，因为他们需要的是 `libssl.so.3`。

而 [quic](https://github.com/sbwml/r4s_build_script/tree/master/openwrt/patch/openssl/quic) 这里的补丁并没有更改openssl 本身的版本信息，所以不需要它。

abelian424, 2023-12-17:
> > What is due to cmake?
> 
> 印象中好像是会让 curl 无法正确找到 ngtcp2 & nghttp3 的头文件（header file）
> 
> 我确定文件被拷贝到 `STAGING_DIR` 中了，因为那时候我有定义 `CMAKE_INSTALL:=1`。
> 
> 如果你把 openwrt feeds 中提供的 ngtcp2 & nghttp3 作为独立的库去构建，它是没有任何问题的，但是作为 curl 或其它软件的构建依赖，cmake 确实会让这个过程出现一点毛病

```
When compiling curl and enabling ngtcp2 library support, 
use cmake to compile ngtcp2 & nghttp3, then curl will fail to compile 
(I did not keep any logs, because it has been three months, you can try it yourself), 
the first time I let When curl supports http3, cmake is used to compile ngtcp2 & nghttp3, 
but curl compilation failed at that time. Finally, I used the default automake to compile successfully. 
After all, my requirement is just to enable curl to implement quic/http3 support, 
not how to compile it, so I don’t have any further understanding of the reason.

PS: Reply from mobile phone, no way to translate it into English
```

I see. I'm starting to suspect that the inclusion of quictls as a replacement for openssl in OpenWrt is the only thing really in the way. I've been unable to compile your version of libcurl BTW, probably due to the differences in `Config.in` - @stangri is possibly having the same issue.

sbwml, 2023-12-17:
> [I can just edit this file to download quictls](https://raw.githubusercontent.com/openwrt/openwrt/openwrt-23.05/package/libs/openssl/Makefile):

你不需要改变 openwrt 的 openssl ，只需要将这些补丁下载到 openssl 的patches 目录中，它就已经实现 quic 的支持 https://github.com/sbwml/r4s_build_script/blob/master/openwrt/scripts/00-prepare_base.sh#L151

abelian424, 2023-12-17:
> 而 [quic](https://github.com/sbwml/r4s_build_script/tree/master/openwrt/patch/openssl/quic) 这里的补丁并没有更改openssl 本身的版本信息，所以不需要它。

`The patch here in quic does not change the version information of openssl itself, so it is not needed.`

So these are cherry-picked commits from the quictls repo made by [@tmshort](https://github.com/tmshort) one of the maintainers of quictls.

sbwml, 2023-12-17:
> > 而 [quic](https://github.com/sbwml/r4s_build_script/tree/master/openwrt/patch/openssl/quic) 这里的补丁并没有更改 openssl 本身的版本信息，所以不需要它。
> 
> `The patch here in quic does not change the version information of openssl itself, so it is not needed.`
> 
> 所以这些是 quictls 维护者之一 [@tmshort](https://github.com/tmshort) 从 quictls 存储库中精心挑选的提交。您是否向后移植 quictls 版本的提交 > 3.0.12-quic1？

事实上它与 openssl-3.0.12-quic1 保持一致，只是移除了一些文档的说明（避免因为版本升级后文档的改变导致补丁失败）

sbwml, 2023-12-17:
> But couldn't I could just symlink these libraries after a firmware update?

它会影响你的一些依赖openssl 库的程序编译

abelian424, 2023-12-17:
> > But couldn't I could just symlink these libraries after a firmware update?
> 
> 它会影响你的一些依赖openssl 库的程序编译

Well, here's a modified `Makefile` excerpt and patch for posterity (the weirdness of variables is all due to the doubled naming scheme of the tar.gz):
```
PKG_NAME:=openssl
PKG_VERSION:=openssl-3.0.12-quic1
PKG_RELEASE:=1
PKG_BUILD_FLAGS:=no-mips16 gc-sections no-lto

PKG_BUILD_PARALLEL:=1

PKG_SOURCE:=$(PKG_VERSION).tar.gz
PKG_SOURCE_URL:= https://github.com/quictls/openssl/archive/refs/tags/

PKG_HASH:=ca0e148f5e50c38198a25559d52cc06781345f5bff15de526ff65b0cc623d1bc
```
`./patches/999-hack-version.patch`
```
--- a/VERSION.dat
+++ b/VERSION.dat
@@ -7,1 +7,1 @@
-SHLIB_VERSION=81.3
+SHLIB_VERSION=3
```
@stangri should this issue be closed until `quictls` is officially ported to OpenWrt?

abelian424, 2023-12-20:
It's possible to build quictls using the official release tarballs:
```
PKG_NAME:=openssl
PKG_VERSION:=3.0.12-quic1
PKG_RELEASE:=1
PKG_BUILD_FLAGS:=no-mips16 gc-sections no-lto
PKG_BUILD_DIR:=$(BUILD_DIR_HOST)/openssl-openssl-$(PKG_VERSION)
PKG_BUILD_PARALLEL:=1

PKG_SOURCE:=openssl-$(PKG_VERSION).tar.gz
PKG_SOURCE_URL:=https://codeload.github.com/quictls/openssl/tar.gz/openssl-$(PKG_VERSION)?dummy=/
PKG_HASH:=ca0e148f5e50c38198a25559d52cc06781345f5bff15de526ff65b0cc623d1bc
```
Some caveats: 
- Depending on `openssl` configuration the patches included in the `patches` directory may cause an error regarding `devcrypto.` A lot of the patches can be deleted, especially regarding ones about unused hardware encryption engines, but turning on `NO_DEPRECATED` option may lead to a compilation error. I edited `999-hack-version.patch` so that I could use it with `quicktls!=3.0.10' without having to edit it for every different version:
```
--- a/VERSION.dat
+++ b/VERSION.dat
@@ -7,1 +7,1 @@
-SHLIB_VERSION=81.3
+SHLIB_VERSION=3
```

- `curl` took a lot of trial and error to compile. The built in `toolchain` doesn't have all the libraries to complete without error. Make sure `gdb` is turned on in `menuconfig` before trying so that the config.log is created; it has a lot of useful information about exactly which libraries are missing. I ended up running an [Alpine Linux Docker container ](https://openwrt.org/docs/guide-user/virtualization/obtain.firmware.docker) and `make` only failed once for `expat` dependency that is not in the downloadable `containerfile`.

**The takeaway here is that `quictls` is the only out-of-tree code. You can build `curl` with HTTP/3 support in order to select `nghttp3` and `ngtcp2` as dependencies.**

stangri, 2023-12-21:
@abelian424 thank you for posting this. Have you tried building it with quictls 3.1.4?

abelian424, 2023-12-21:
@stangri Yeah, I actually built with 3.1.4 but posted for 3.0.12 as that is the current version on OpenWrt. The sha256sum `PKG_HASH` for 3.1.4 is `4bf990243d6aa39b8befa0c399834415842912ef67f88bef98e74dc619469618`

sbwml, 2023-12-21:
> ```
> --- a/VERSION.dat
> +++ b/VERSION.dat
> @@ -7,1 +7,1 @@
> -SHLIB_VERSION=81.3
> +SHLIB_VERSION=3
> ```

How is this diff generated for a single line of content?

abelian424, 2023-12-21:
@sbwml `@@ -7,1 +7,1 @@` shows that line 7 and only line 7 is being patched and that the patch is also one line. In other words, the first pair of numbers is line number and number of lines being patched in _a_, and the second pair is the same for _b_.

stangri, 2023-12-21:
> * A lot of the patches can be deleted, especially regarding ones about unused hardware encryption engines, but turning on `NO_DEPRECATED` option may lead to a compilation error. I edited [...]

Do you mind publishing the patches you kept on GitHub?

> * The built in `toolchain` doesn't have all the libraries to complete without error. Make sure `gdb` is turned on in `menuconfig` before trying so that the config.log is created; it has a lot of useful information about exactly which libraries are missing.

Do you recall/can you compare configs to confirm which libraries solved the build problem for libcurl for you? I've successfully built HTTP/3 supporting curl before creating PRs/merging nghttp3/ngtcp2 libraries into the tree, but I've been unsuccessful since and would like to build it again to publish a recipe in the curl README.

abelian424, 2023-12-22:
```
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_AFALG_UPDATE_CTR_IV=y
CONFIG_AUTOREMOVE=y
CONFIG_BUILDBOT=y
CONFIG_BUILD_PATENTED=y
CONFIG_CLEAN_IPKG=y
CONFIG_COLLECT_KERNEL_DEBUG=y
CONFIG_DOWNLOAD_FOLDER="/workdir/dl"
# CONFIG_GDB is not set
CONFIG_INCLUDE_CONFIG=y
CONFIG_JSON_CYCLONEDX_SBOM=y
CONFIG_KERNEL_AUDIT=y
CONFIG_KERNEL_BUILD_DOMAIN="buildhost"
CONFIG_KERNEL_BUILD_USER="builder"
CONFIG_KERNEL_EXT4_FS_SECURITY=y
CONFIG_KERNEL_F2FS_FS_SECURITY=y
CONFIG_KERNEL_JFFS2_FS_SECURITY=y
# CONFIG_KERNEL_KALLSYMS is not set
CONFIG_KERNEL_SECURITY=y
CONFIG_KERNEL_SECURITY_NETWORK=y
CONFIG_KERNEL_SQUASHFS_XATTR=y
CONFIG_KERNEL_UBIFS_FS_SECURITY=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NGHTTP3=y
CONFIG_LIBCURL_NGTCP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_OPENSSL=y
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_THREADED_RESOLVER=y
CONFIG_LIBCURL_TLS_SRP=y
CONFIG_LIBCURL_UNIX_SOCKETS=y
CONFIG_LIGHTTPD_CRYPTOLIB_NETTLE=y
CONFIG_LIGHTTPD_PCRE2=y
CONFIG_LIGHTTPD_SSL=y
CONFIG_LIGHTTPD_SSL_SELECT=m
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_SMALL_FOOTPRINT=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_DTLS=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_PACKAGE_ath10k-board-qca988x=m
CONFIG_PACKAGE_block-mount=y
# CONFIG_PACKAGE_blockd is not set
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_ddns-scripts=y
CONFIG_PACKAGE_ddns-scripts-services=y
# CONFIG_PACKAGE_dnsmasq is not set
CONFIG_PACKAGE_dnsmasq-full=y
CONFIG_PACKAGE_dnsmasq_full_auth=y
CONFIG_PACKAGE_dnsmasq_full_conntrack=y
CONFIG_PACKAGE_dnsmasq_full_dhcp=y
CONFIG_PACKAGE_dnsmasq_full_dnssec=y
CONFIG_PACKAGE_dnsmasq_full_nftset=y
CONFIG_PACKAGE_dnsmasq_full_noid=y
CONFIG_PACKAGE_dnsmasq_full_tftp=y
[... 116 lines cut ...]
CONFIG_TARGET_INITRAMFS_COMPRESSION_LZ4=y
# CONFIG_TARGET_INITRAMFS_COMPRESSION_NONE is not set
CONFIG_TARGET_OPTIONS=y
CONFIG_WPA_MSG_MIN_PRIORITY=3
CONFIG_ZLIB_OPTIMIZE_SPEED=y
# CONFIG_BPF_TOOLCHAIN_BUILD_LLVM is not set
CONFIG_BPF_TOOLCHAIN_PREBUILT=y
# CONFIG_IB is not set
# CONFIG_MAKE_TOOLCHAIN is not set
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set
CONFIG_PACKAGE_kmod-crypto-kpp=m
CONFIG_PACKAGE_libelf=m
# CONFIG_PACKAGE_libncurses-dev is not set
CONFIG_PACKAGE_tc-tiny=m
CONFIG_PACKAGE_zlib=m
# CONFIG_PACKAGE_zlib-dev is not set
# CONFIG_PCRE2_JIT_ENABLED is not set
# CONFIG_SDK is not set
CONFIG_USE_LLVM_PREBUILT=y
CONFIG_ZLIB_OPTIMIZE_SPEED=y
```
There is the `diffconfig`, but I would wager the failure of `libcurl` was more to do with the build environment. Using the Debian container linked with only `expat` as an addition allows it to build for me. However, I was unsuccessful with an Arch Linux environment, and Alpine Linux needs fixes/workarounds for its low support of `perl`.

sbwml, 2023-12-22:
> Have you tried building it with quictls 3.1.4?

Since it is separate from the version used by [OpenWrt](https://github.com/openwrt/openwrt/blob/main/package/libs/openssl/Makefile#L11), why don't you just use openssl 3.2.0? This way you can support quic without making any changes to the source code.

abelian424, 2023-12-22:
> Since it is separate from the version used by [OpenWrt](https://github.com/openwrt/openwrt/blob/main/package/libs/openssl/Makefile#L11), why don't you just use [openssl [...]

That... is good to know. News to me that 3.2.0 has HTTP/3 support.

stangri, 2023-12-23:
@sbwml thanks for the tip about OpenSSL 3.2.0, after switching to it from quictls library and with the in-tree ngtcp2 and nghttp3 I get the following when trying to build curl with QUICK support enabled and OpenSSL selected as the library:
```
...
checking size of size_t... configure: error: Failed to find size of size_t
make[2]: *** [Makefile:192: /SDK_x86-64_23.05.2/build_dir/target-x86_64_musl/curl-8.5.0/.configured_ee316af4a147e2242e111d0ebb731019] Error 1
make[2]: Leaving directory '/SDK_x86-64_23.05.2/feeds/packages/net/curl'
time: package/feeds/packages/curl/compile#45.61#3.60#48.82
    ERROR: package/feeds/packages/curl failed to build.
make[1]: *** [package/Makefile:128: package/feeds/packages/curl/compile] Error 1
make[1]: Leaving directory '/SDK_x86-64_23.05.2'
make: *** [/SDK_x86-64_23.05.2/include/toplevel.mk:225: package/curl/compile] Error 2
```

Any ideas @sbwml @abelian424 ?

abelian424, 2023-12-23:
@abelian424 Oh, that's also a perl issue. Just install the whole perl package and it will build. I need to find some macro to narrow down the exact dependencies needed.

stangri, 2023-12-23:
> @stangri Oh, that's also a perl issue. Just install the whole perl language package and it will build. I need to find some macro to narrow down the exact dependencies needed, because I keep forgetting to clean files properly before rebuilding.

Thanks for your prompt reply, I'll try.

I remember one of the curl updates affected the `perl-www-curl` module, so that would be my primary suspect.

stangri, 2023-12-24:
> @stangri Oh, that's also a perl issue. Just install the whole perl language package and it will build. I need to find some macro to narrow down the exact dependencies needed, because I keep forgetting to clean files properly before rebuilding.

I've installed/enabled all perl packages that opkg lists:
```
	./scripts/feeds install -dy perl perl-ack perl-authen-sasl perl-authen-sasl-xs perl-cgi perl-compress-bzip2 perl-dbi perl-device-serialport perl-device-usb perl-encode-locale perl-file-listing perl-file-next perl-file-rsyncp perl-file-sharedir-install perl-html-form perl-html-parser perl-html-tagset perl-html-tree perl-http-cookies perl-http-daemon perl-http-date perl-http-message perl-http-negotiate perl-http-server-simple perl-inline perl-inline-c perl-io-html perl-lockfile-simple perl-lwp-mediatypes perl-net-cidr-lite perl-net-dns perl-net-http perl-net-telnet perl-netaddr-ip 
	./scripts/feeds install -dy perl-parse-recdescent perl-parse-yapp perl-sub-uplevel perl-test-harness perl-test-warn perl-text-csv_xs perl-try-tiny perl-uri perl-www perl-www-curl perl-www-mechanize perl-www-robotrules perl-xml-parser perlbase-anydbm-file perlbase-app perlbase-archive perlbase-arybase perlbase-attribute perlbase-attributes perlbase-autodie perlbase-autoloader perlbase-autosplit perlbase-autouse perlbase-b perlbase-base perlbase-benchmark perlbase-bigint perlbase-bignum perlbase-blib perlbase-bytes perlbase-charnames perlbase-class perlbase-compress perlbase-config perlbase-cpan perlbase-cwd perlbase-data perlbase-db perlbase-db-file perlbase-dbm-filter perlbase-devel perlbase-diagnostics perlbase-digest perlbase-dirhandle perlbase-dumpvalue perlbase-dumpvar perlbase-dynaloader perlbase-encode perlbase-encoding perlbase-english perlbase-env perlbase-errno perlbase-essential perlbase-experimental perlbase-extutils perlbase-fatal perlbase-fcntl perlbase-feature
	./scripts/feeds install -dy perlbase-fields perlbase-file perlbase-filecache perlbase-filehandle perlbase-filetest perlbase-filter perlbase-findbin perlbase-gdbm-file perlbase-getopt perlbase-hash perlbase-http-tiny perlbase-i18n perlbase-if perlbase-integer perlbase-io perlbase-ipc perlbase-json-pp perlbase-less perlbase-list perlbase-locale perlbase-math perlbase-memoize perlbase-meta-notation perlbase-mime perlbase-module perlbase-mro perlbase-net perlbase-next perlbase-o perlbase-opcode perlbase-open perlbase-ops perlbase-ostype perlbase-params perlbase-perl5db perlbase-perlio perlbase-pod perlbase-posix perlbase-re perlbase-safe perlbase-scalar perlbase-sdbm-file perlbase-search perlbase-selectsaver perlbase-selfloader perlbase-sigtrap perlbase-socket perlbase-sort perlbase-storable perlbase-symbol perlbase-sys perlbase-tap perlbase-term perlbase-test perlbase-text perlbase-thread perlbase-threads perlbase-tie perlbase-time perlbase-unicode perlbase-unicore perlbase-universal perlbase-user perlbase-utf8 perlbase-version perlbase-xsloader 
```

And I still get the `checking size of size_t... configure: error: Failed to find size of size_t` error when building curl if I enable either HTTP/3 or QUIC in menuconfig. Without either, it builds fine even with openssl 3.2.0.

I've tried googling it and didn't find anything relevant to my case, I'm wondering if it may be the build environment.

abelian424, 2023-12-24:
@stangri I've been trying to build other TLS libraries to look for something slimmer than OpenSSL - I saw that 'checking size of size_t' complete successfully when I was building cmake. What is your build environment? I've been using Alpine because OpenWrt on GCC 13 doesn't work in Debian-unstable. Here's the Dockerfile:

```
FROM alpine:latest

ENV GCC_COLORS='error=01;31:warning=01;35:note=01;36:caret=01;32:locus=01:quote=01'
WORKDIR /workdir

RUN apk add \
    'argp-standalone' \
    'asciidoc' \
    'bash' \
    'bc' \
    'binutils' \
    'bzip2' \
    'cdrkit' \
    'coreutils' \
    'diffutils' \
    'elfutils-dev' \
    'findutils' \
    'flex' \
    'g++' \
    'gawk' \
    'gettext' \
    'git' \
    'grep' \
    'gzip' \
    'intltool' \
    'libxslt' \
    'linux-headers' \
    'musl-fts-dev' \
    'musl-libintl' \
    'musl-obstack-dev' \
    'ncurses-dev' \
    'libressl' \
    'patch' \
    'perl-dev' \
    'python3-dev' \
    'rsync' \
    'unzip' \
    'util-linux' \
    'zlib-dev' \
    'curl' 'build-base' 'wget' 'gnupg' 'perl-module-cpanfile' 'perl-module-scandeps' 'tar' \
    'nano' 'expat' 'zsh' 'zsh-theme-powerlevel10k'

RUN curl -LO https://raw.githubusercontent.com/miyagawa/cpanminus/master/cpanm \
    && chmod +x cpanm \
    && ./cpanm App::cpanminus \
    && rm -fr ./cpanm /root/.cpanm

ENV PERL_CPANM_OPT --verbose --mirror https://cpan.metacpan.org --mirror-only
RUN cpanm Digest::SHA Module::Signature && rm -rf ~/.cpanm
ENV PERL_CPANM_OPT $PERL_CPANM_OPT --verify

RUN ln -s '/usr/lib/libncurses.so' '/usr/lib/libtinfo.so' && \
  addgroup 'buildbot' && \
  adduser -s '/bin/bash' -G 'buildbot' -D 'buildbot'

USER buildbot
```

The additions are to test perl functionality. Mind, I've gotten stuck on building ngtcp2 but that error is due to trying to build with libressl not being recognized as an openssl/quictls replacement. libcurl shows an error with ngtcp2 not being accepted due to lack of quic support. If you want you can wait until I verify that libcurl successully builds - I just need to figure out how to tell ngtcp2 to locate libressl in staging_dir/host/lib.

stangri, 2023-12-24:
I don't think the error I'm getting is perl-related. I've added/enabled all the perl modules in build and it still doesn't help.

abelian424, 2023-12-25:
> ```
> $ lsb_release -a
> No LSB modules are available.
> Distributor ID:	Ubuntu
> Description:	Ubuntu 22.04.3 LTS
> Release:	22.04
> Codename:	jammy
> ```
> 
> I don't think the error I'm getting is perl-related. I've added/enabled all the perl modules in build and it still doesn't help.

Sorry, it isn't perl-related. Do you have gcc-multilib installed? I've successfully built an image with quictls 3.1.2 with ngtcp2 and nghttp3, and the only issue was building perl which I fixed by copying missing files into staging_dir from the host and prefacing `make` with `PERL5LIB=/workdir/openwrt/staging_dir/hostpkg/usr/lib/perl5/5.38`.

stangri, 2023-12-25:
> Do you have gcc-multilib installed?

I didn't, but I have installed it now and I'm still getting the `checking size of size_t... configure: error: Failed to find size of size_t`.

Any other suggestions?

abelian424, 2023-12-25:
@stangri This is stock Ubuntu 22.04.3 - which is based on Debian Bookworm? I didn't have any libc-related problems (which this is) with Debian-stable but even so, why don't you try the dockerfile I posted. Without determining a clean chroot it's hard to say what package would fix this.

stangri, 2023-12-26:
> EDIT: I've made a [repo](https://github.com/abelian424/openwrt-quictls/tree/main) so that you can easily look through the files I used.

I've installed apline 3.19.0 in a VM and added all the packages from your docker file, however when I'm trying to build the toolchain (or curl), it fails with:
```
/home/stangri/SDK_x86-64_23.05.2/scripts/ipkg-build: line 150: 10817 Segmentation fault      mkdir "$tmp_dir"
```

abelian424, 2023-12-26:
@stangri That seems like it's crashing due to not having file/directory permissions.

stangri, 2023-12-27:
@abelian424 I've been using SDK before, but when I switched to building from source I was successful rebuilding all the necessary libraries and openssl-quictls-based libcurl in Alpine VM. Weirdly, the curl 8.5.0 building still fails, but I have install the binary from the OpenWrt repo, I just need the updated libraries.

[35 more replies]