openwrt/packages #19383: collectd: multiple issues (fails to compile)

PolynomialDivision, 2022-09-16:
Collectd has warnings while configure
```
configure: WARNING: unrecognized options: --disable-nls, --disable-ampq1, --disable-curl_jolokia, --disable-genericjmx, --disable-lvm, --disable-monitorus, --disable-openvz, --with-libnetlink
```

and fails in the end with the error:
```
src/nut.c:40:2: error: #error "Unable to determine the UPS connection type."
   40 | #error "Unable to determine the UPS connection type."
      |  ^~~~~
src/nut.c:46:3: error: unknown type name 'collectd_upsconn_t'
   46 |   collectd_upsconn_t *conn;
      |   ^~~~~~~~~~~~~~~~~~
```

See: https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/packages/collectd/compile.txt

PolynomialDivision, 2022-09-16:
Maybe it has something to do with this nut update that currently fails:
https://github.com/openwrt/packages/commit/f5c0f8297d8fadda2f4da81f09422985e997c67f

hnyman, 2022-09-16:
Probably yes. Looks like collectd module does not like the new nut.
cc @EricLuehrsen who did the nut update in #19340

hnyman, 2022-09-16:
possible recent fixes in upstream collectd:
Collectd PR https://github.com/collectd/collectd/pull/4043

hnyman, 2022-09-16:
Apparently collectd is incompatible with the new nut 2.8.0
I restored the compiling of the other psrts of collectd by disabling the nut plugin with #19385

closed 2022-09-16 by hnyman

PolynomialDivision, 2022-09-16:
Nice, thanks! :)

cotequeiroz, 2022-09-16:
I think https://github.com/openwrt/packages/pull/19373 fixes this. The data type detection fails because time_t is used without time.h in upsclient.h, failing to compile no regardless of the actual data type being checked.
The int type detection from https://github.com/collectd/collectd/commit/d409ffa2a64cac3fc2abe2bb86ec3a80cb34d0a6 might be needed, depending on the arch, perhaps.  I have only built it for aarch64.

hnyman, 2022-09-16:
Thanks, I will look into it after the weekend. 
Based on the upstream collectd and nut PR/issue discussions, both the time_t fix and the int detection fix are likely needed.

hnyman, 2022-09-18:
Thanks, I re-enabled the plugin without further changes. 
My own compilation succeeded, but the automatic CI failed due to some unrelated Lantiq stuff ;-(  
Let's see how it compiles in the actual buildbot.