openwrt/packages #19386: fish: Fish Shell don't work in OpenWRT 22.03.0 in mipsel_24kc architecture.

dainslef, 2022-09-16:
Maintainer: @
Description:

`fish` shell always crash after upgrading to `OpenWRT 22.03.0` (In `OpenWRT 21.02`, `fish` shell can work correctly).

```
$ fish
Assertion failed: (status_.load() & STATUS_NEEDS_WAKEUP) == 0 && "No thread should be waiting" (fish-3.3.1/src/topic_monitor.cpp: try_update_gens_maybe_becoming_reader: 208)
Aborted
```

This issue seems that happens in all `mipsel_24kc` architecture devices, I have `Redmi Router AC2100`(MT7621A), `GL.iNet MT1300`(MT7621A), `Xiaomi Router 4C`(MT7628AN) ...
All of these devices had this issue.

`fish` package info:

```
Package: fish
Version: 3.3.1-2
Depends: libc, libncurses6, libstdcpp6, libatomic1, librt, libpcre2-32
Status: install user installed
Section: utils
Architecture: mipsel_24kc
Size: 2050273
Filename: fish_3.3.1-2_mipsel_24kc.ipk
Description: Fish is a smart and user-friendly command line shell for OS X, Linux, and the
 rest of the family. Fish includes features like syntax highlighting,
 autosuggest-as-you-type, and fancy tab completions that just work, with no
 configuration required.
Installed-Time: 1663171114
```

Environment (Redmi Router AC2100): 

```
NAME="OpenWrt"
VERSION="22.03.0"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 22.03.0"
VERSION_ID="22.03.0"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r19685-512e76967f"
OPENWRT_BOARD="ramips/mt7621"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 22.03.0 r19685-512e76967f"
```

jqqqqqqqqqq, 2022-09-16:
I see in master branch, fish version is 3.5.0
Can you try to compile it from master or get it from snapshot repo and try again?

jqqqqqqqqqq, 2022-09-16:
Otherwise try to build 3.5.1 and see if it works. If not, it is probably an upstream bug
https://github.com/openwrt/packages/pull/19388/commits/dea8906ef45cde72bd42d7318f106679978b3874

dainslef, 2022-09-17:
> I see in master branch, fish version is 3.5.0 Can you try to compile it from master or get it from snapshot repo and try again?

I build the `fish 3.5.0`  from snapshot, but this issue still exists.
Package info:

```
Package: fish
Version: 3.3.1-2
Depends: libc, libncurses6, libstdcpp6, libatomic1, librt, libpcre2-32
Status: unknown ok not-installed
Section: utils
Architecture: mipsel_24kc
Size: 2050273
Filename: fish_3.3.1-2_mipsel_24kc.ipk
Description: Fish is a smart and user-friendly command line shell for OS X, Linux, and the
 rest of the family. Fish includes features like syntax highlighting,
 autosuggest-as-you-type, and fancy tab completions that just work, with no
 configuration required.

Package: fish
Version: 3.5.0-1
Depends: libc, libncurses6, libstdcpp6, libatomic1, libpcre2-32
Status: install user installed
Architecture: mipsel_24kc
Installed-Time: 1663396062
```

dainslef, 2022-09-17:
> Otherwise try to build 3.5.1 and see if it works. If not, it is probably an upstream bug [dea8906](https://github.com/openwrt/packages/commit/dea8906ef45cde72bd42d7318f106679978b3874)

I just upgrade to `fish 3.5.1`,  the issue still exists, package info:

```
root@OpenWrt:~# opkg info fish
Package: fish
Version: 3.3.1-2
Depends: libc, libncurses6, libstdcpp6, libatomic1, librt, libpcre2-32
Status: unknown ok not-installed
Section: utils
Architecture: mipsel_24kc
Size: 2050273
Filename: fish_3.3.1-2_mipsel_24kc.ipk
Description: Fish is a smart and user-friendly command line shell for OS X, Linux, and the
 rest of the family. Fish includes features like syntax highlighting,
 autosuggest-as-you-type, and fancy tab completions that just work, with no
 configuration required.

Package: fish
Version: 3.5.1-1
Depends: libc, libncurses6, libstdcpp6, libatomic1, libpcre2-32
Status: install user installed
Architecture: mipsel_24kc
Installed-Time: 1663398786

root@OpenWrt:~# fish
Welcome to fish, the friendly interactive shell
Type help for instructions on how to use fish
root@OpenWrt ~# 
root@OpenWrt ~# 
root@OpenWrt ~# ls
Assertion failed: (status_.load() & STATUS_NEEDS_WAKEUP) == 0 && "No thread should be waiting" (fish-3.5.1/src/topic_monitor.cpp: try_update_gens_maybe_becoming_reader: 208)
Aborted
```

dainslef, 2022-09-17:
I just found some already submit a similar issue in [fish project](https://github.com/fish-shell/fish-shell/issues/8653) .

neheb, 2022-09-17:
tested on QEMU mips and mipsel. Only an issue on mipsel. Couldn't test 64-bit as it doesn't even boot.

hdhoang, 2022-09-18:
thanks for reproducing! hopefully with qemu, fish people can get a handle on it more easily.

for other users, as a workaround i have been using the [3.1.2 ipk file for 21.02.3](https://downloads.openwrt.org/releases/21.02.3/packages/mipsel_24kc/packages/) for a few days now.

hdhoang, 2024-03-31:
Let's close this, and move on with the new fish.

There is some hiccups because this platform doesn't have `AtomicU64` support. Replacing that with `AtomicU32` at a guess, and converting callsites, I could build a binary that runs. Its commandline reader doesn't work, but argument-parsing does.

```shell
> run/fish --version
fish, version 3.1.2
> run/rfish --version
fish, version a29cc8f-dirty
> run/rfish --command 'date'
Sun Mar 31 16:28:23 +07 2024
```

We can hope for better support along with other 32-bit platforms.

nshopik, 2025-01-06:
So do we need anything on this one, or we could just bump fish package version?

neheb, 2025-01-07:
Doesn't the most recent fish use rust?

hdhoang, 2025-01-07:
IME fish 4.0b1 works, so this ticket is wontfix. Hopefully the packagers can build fish4 for this target