openwrt/packages #19389: ddns script for noip.com is not working,

alirz1, 2022-09-16:
setting up no-ip.com ddns script version: 2.8.2-12
However ddns update fails with following message:

 183245       : verbose mode  : 0 - run normal, NO console output
 183245  WARN : No update_url found/defined or no update_script found/defined! - TERMINATE
 183245  WARN : PID '7511' exit WITH ERROR '1' at 2022-09-16 18:32\n

brada4, 2022-09-19:
you need ddns-scripts-noip package (and at least tell openwrt version)

alirz1, 2022-09-19:
OpenWrt 21.02-SNAPSHOT r16399+157-c67509efd7

I had already installed the noip ddns script package. Only then does the noip is listed as a ddns provider. Sorry it it wasn't clear.

brada4, 2022-09-19:
You have later 21.02.3 r16554 release or even 22.03.0 which both have maintained and updated releases of those scripts. (worked for me on old one, still works on upgrade, did not pay attention to versions frankly)

rmilecki, 2022-09-27:
@brada4: please provide output of two following commands:
```
uci show ddns | grep service_name
```
```
cat /usr/share/ddns/default/no-ip.com.json
```

alirz1, 2022-10-04:
root@GL-AXT1800:~# uci show ddns | grep service_name
ddns.XXXXXXXX.service_name='changeip.com'
ddns.YYYYYYYY.service_name='no-ip.com'

root@GL-AXT1800:~# cat /usr/share/ddns/default/no-ip.com.json
{
        "name": "no-ip.com",
        "ipv4": {
                "url": "update_no-ip_com.sh"
        },
        "ipv6": {
                "url": "update_no-ip_com.sh"
        }
}

I have since updated the firmware on my router. Now im running the following:
**OpenWrt 21.02-SNAPSHOT r16399+157-c67509efd7 / LuCI openwrt-22.03 branch git-21.284.67084-e4d24f0**

But now Im running into additional issues where my first ddns1 service, which prior to the fw update works, now doesnt even auto start or fails to start sometimes. I get a XHR timeout error and i see this in the log:

 150750       : Local IP '206.XX.X.X' detected on web at 'https://api.ipify.org'
 150750       : Waiting 600 seconds (Check Interval)
 150824  note : PID '22062' terminated by 'SIGTERM' at 2022-10-04 15:08\n

mbranchick, 2023-01-27:
@alirz1 were you ever able to resolve this?  Im having the same issue, although Im getting the behavior of your initial problem with the no script or URL found.

mbranchick, 2023-01-27:
Im seeing the following error: 

 ```
065351  note : PID '9256' started at 2023-01-27 06:53
 065351       : ddns version  : 2.7.7-6
 065351       : uci configuration:\nddns.NoIP4.check_unit='minutes'
ddns.NoIP4.domain='dynupdate.no-ip.com'
ddns.NoIP4.enabled='1'
ddns.NoIP4.force_unit='minutes'
ddns.NoIP4.interface='wan'
ddns.NoIP4.ip_network='wan'
ddns.NoIP4.ip_source='network'
ddns.NoIP4.lookup_host='brume2ip.ddns.net'
ddns.NoIP4.password='*password*'
ddns.NoIP4.retry_unit='seconds'
ddns.NoIP4.service_name='no-ip.com'
ddns.NoIP4.use_ipv6='0'
ddns.NoIP4.use_syslog='2'
ddns.NoIP4.username='mbranchick'
ddns.NoIP4=service
 065351       : verbose mode  : 0 - run normal, NO console output
 065351  WARN : No update_url found/defined or no update_script found/defined! - TERMINATE
 065351  WARN : PID '9256' exit WITH ERROR '1' at 2023-01-27 06:53\n
```

alirz1, 2023-01-27:
> @alirz1 were you ever able to resolve this? Im having the same issue, although Im getting the behavior of your initial problem with the no script or URL found.

Yes it seems to be working with some workarounds i guess. I have the DDNS auto start set to disabled.
I also have the ddns scripts package installed as that is needed.
My ddns is configured however i have a command set in the openwrt's local startup tasks to start the ddns servie. By doing so it seems to work for me.
startup task: /usr/lib/ddns/dynamic_dns_updater.sh -- start

alirz1, 2023-02-04:
So im setting up a new router and trying to get the DDNS scripts to work. I have installed the "no-ip.com" scripts package etc and running into the same original issue that i opened this issue for. The fact that this script doesnt work out of the box, THERE IS SOMETHING BROKEN IN IT!!!!

openwrt running on router: [LuCI openwrt-22.03 branch (git-21.284.67084-e4d24f0)](https://github.com/openwrt/luci) / OpenWrt 21.02-SNAPSHOT r16399+157-c67509efd7

noip script installed: ddns-scripts-noip | 2.8.2-12

Log:

 221746       : verbose mode  : 0 - run normal, NO console output
 221746  WARN : No update_url found/defined or no update_script found/defined! - TERMINATE
 221746  WARN : PID '8913' exit WITH ERROR '1' at 2023-02-04 22:17\n
 222134       : ************ ************** ************** **************
 222134  note : PID '19506' started at 2023-02-04 22:21
 222134       : ddns version  : 2.7.7-6

brada4, 2023-02-04:
What makes you think pre-release snapshot will ever change?
Please upgrade to 22.02.**3** and install ddns-scripts-noip 2.8.2**-25**

mbranchick, 2023-03-15:
@brada4 Im unable to update because I use the device firmware which controls the underlying openWRT version which has not been upgraded to version 22.

mbranchick, 2023-03-15:
@alirz1 I was able to get this working using somewhat of a work around.  I'm used the luci UI to setup a provider using the custom service type.  From there I manually edited the config and custom script to resolve some issues.  

1. Install everything requested by the OpenWRT documentation to enable DDNS support
2. Create a provider using the "custom" service type
3. Edit the update URL in the custom no-ip script.  It was wrong in my version, it was using noip.com rather than no-ip.com
4. Remove the service name option from the config file entry for your custom provider.  It was configured by the UI as "-" which the luci documentation says should be removed.  
5. The user/pass can be entered in plain text rather than URL encoded in the config because I think the no-ip custom script is handling that conversion
6. Set the script path in the config to noip script manually.

brada4, 2023-03-16:
0. ask your vendor to adjust to supported OpenWRT package feed which contains fixed ddns script. snapshot packages will never change, they are development releases keeping eventual bugs of that moment forever.