openwrt/packages #19403: transmission: ujail missed incomplete dir

luizluca, 2022-09-19:
Maintainer: @dangowrt
OpenWrt 22.03.0

When incomplete dir is in use, transmission will fail to download. You can workaround that editing init.d script and adding that directory as a rw mount.

salim-b, 2023-02-10:
I can confirm this observation. As a workaround, setting `incomplete_dir` to a subdir of `download_dir` works.

---

Additionally, the **`$config_dir/dht.dat`** seems to be missing.

```sh
# cat /etc/init.d/transmission | grep procd_add_jail
procd_add_jail transmission log
procd_add_jail_mount "$config_file"
procd_add_jail_mount_rw "$config_dir/resume"
procd_add_jail_mount_rw "$config_dir/torrents"
procd_add_jail_mount_rw "$config_dir/blocklists"
procd_add_jail_mount_rw "$config_dir/stats.json"
procd_add_jail_mount_rw "$download_dir"
[ -d "$web_home" ] && procd_add_jail_mount "$web_home"
[ -f "$ca_bundle_file" ] && procd_add_jail_mount "$ca_bundle_file"
```

This fills the system log with errors like

```
Fri Feb 10 13:00:09 2023 daemon.err transmission-daemon[14411]: [2023-02-10 12:00:09.272] Couldn't save temporary file "/mnt/storage/transmission/config/dht.dat.tmp.XXXXXX": Permission denied (variant.c:1235)
```

Not being able to update the `dht.dat` file seems a bad idea.

---

Adding the following two additional lines to `/etc/init.d/transmission` fixes both issues:

```sh
        procd_add_jail_mount_rw "$incomplete_dir"
        procd_add_jail_mount_rw "$config_dir/dht.dat"
```

salim-b, 2023-02-12:
@dangowrt Addendum: Why not just granting transmission r/w access to the whole config dir, i.e. `procd_add_jail_mount_rw "$config_dir"`? The worst that could happen is a hijacked transmission daemon fiddling with its own config—which shouldn't get the hijacker too far since the daemon is isolated to the jail anyways, right?

Granting access to the whole config dir would allow transmission to write temporary files in the format `filename.ext.tmp.XXXXXX` there which it currently is disallowed to, resulting in failing to update `stats.json` etc.

vasvir, 2023-10-10:
I have been bitten by this (the incomplete dir missing mount).

The other issue with the temp files access denial in the settings dir is something that is beyond me to comprehend.

panaut0lordv, 2023-11-04:
In my case (23.05-rc4) I had to adjust /etc/config/transmission to be in line with what I had in "downstream" (which was earlier set to "do not overwrite" because it's an external data store), which was highly surprising.
I'm also getting lovely
```
Couldn't save '/var/transmission/settings.json': Permission denied (13) (variant.cc:1098)
Couldn't save '/var/transmission/bandwidth-groups.json': Permission denied (13) (variant.cc:1098)"
```
and
`ERR crypto-utils-mbedtls.cc:232 MbedTLS error: CTR_DRBG - The requested random buffer length is too big (-54) (crypto-utils-mbedtls.cc:232)"`
Now existing torrents were able to find their data and for new ones downloads would start. In case someone with external storage follows my steps: make sure you reinstall your package as well.