openwrt/packages #19407: adblock: fails to update lists and never retries

rgl, 2022-09-19:
# Issue template

Maintainer: @dibdot
Environment: x64, OpenWrt 22.03

Description:

When the device boots, it takes some time for its network to be ready (e.g. due to DHCP on the WAN interface) before adblock can reach the internet. When that happens, the `adblock` service fails (and never seems to retry) to create the block list with `Failed to send request: Operation not permitted` as can been see in the service logs:

```
Mon Sep 19 06:19:44 2022 user.info adblock-4.1.4[2238]: backup directory '/tmp/adblock-Backup' created
Mon Sep 19 06:19:44 2022 user.info adblock-4.1.4[2238]: adblock instance started ::: action: start, priority: 0, pid: 2238
Mon Sep 19 06:19:50 2022 user.info adblock-4.1.4[2238]: download of 'stevenblack' failed, url: https://raw.githubusercontent.com/StevenBlack/hosts/master/, rule: /^0\.0\.0\.0[[:space:]]+([[:alnum:]_-]{1,63}\.)+[[:alpha:]]+([[:space:]]|$)/{print tolower($2)}, categories:  hosts, rc: 4, log: Downloading 'https://raw.githubusercontent.com/StevenBlack/hosts/master/hosts' Failed to send request: Operation not permitted
Mon Sep 19 06:19:50 2022 user.info adblock-4.1.4[2238]: download of 'adaway' failed, url: https://raw.githubusercontent.com/AdAway/adaway.github.io/master/hosts.txt, rule: /^127\.0\.0\.1[[:space:]]+([[:alnum:]_-]{1,63}\.)+[[:alpha:]]+([[:space:]]|$)/{print tolower($2)}, categories: -, rc: 4, log: Downloading 'https://raw.githubusercontent.com/AdAway/adaway.github.io/master/hosts.txt' Failed to send request: Operation not permitted
Mon Sep 19 06:19:50 2022 user.info adblock-4.1.4[2238]: download of 'adguard' failed, url: https://adguardteam.github.io/AdGuardSDNSFilter/Filters/filter.txt, rule: BEGIN{FS="[/|^|\r]"}/^\|\|([[:alnum:]_-]{1,63}\.)+[[:alpha:]]+[\/\^\r]+$/{print tolower($3)}, categories: -, rc: 4, log: Downloading 'https://adguardteam.github.io/AdGuardSDNSFilter/Filters/filter.txt' Failed to send request: Operation not permitted
Mon Sep 19 06:19:50 2022 user.info adblock-4.1.4[2238]: download of 'disconnect' failed, url: https://s3.amazonaws.com/lists.disconnect.me/simple_malvertising.txt, rule: /^([[:alnum:]_-]{1,63}\.)+[[:alpha:]]+([[:space:]]|$)/{print tolower($1)}, categories: -, rc: 4, log: Downloading 'https://s3.amazonaws.com/lists.disconnect.me/simple_malvertising.txt' Failed to send request: Operation not permitted
Mon Sep 19 06:19:55 2022 user.info adblock-4.1.4[2238]: download of 'yoyo' failed, url: https://pgl.yoyo.org/adservers/serverlist.php?hostformat=nohtml&showintro=0&mimetype=plaintext, rule: /^([[:alnum:]_-]{1,63}\.)+[[:alpha:]]+([[:space:]]|$)/{print tolower($1)}, categories: -, rc: 4, log: Downloading 'https://pgl.yoyo.org/adservers/serverlist.php?hostformat=nohtml&showintro=0&mimetype=plaintext' Failed to send request: Operation not permitted
Mon Sep 19 06:19:58 2022 user.info adblock-4.1.4[2238]: blocklist with overall 0 blocked domains loaded successfully (Default string ODYSSEY-X86J4125, OpenWrt 22.03.0 r19685-512e76967f)
Mon Sep 19 06:36:29 2022 user.info adblock-4.1.4[7057]: Please enable the 'DNS Report' option to use the reporting feature
```

dibdot, 2022-09-19:
Just raise the "Trigger Delay" to your needs (in LuCI under Addional Settings tab).

closed 2022-09-19 by dibdot

rgl, 2022-09-19:
Isn't that a fragile workaround? Mainly because we do not known how long do we need, as that depends on a lot of external factors that we do not control.

The OpenWrt init system does not provide a way to restart a service until it works?

dibdot, 2022-09-19:
> Isn't that a fragile workaround

Nope, adblock will be triggerd by an interface up event (usually of your wan interface). With the above delay you can fine tune the time period between the event and the actual start of adblock . The default just works fine for most people, just pppoe interfaces tends to take more time...