openwrt/packages #19424: php8: xdebug and phpize extension is missing

barhom, 2022-09-22:
Maintainer: @mhei @flyn-org 
Environment: x86_64, 21.02

Description:
My PHP development on openwrt requires xdebug to run our test-coverage however however this mod cannot be installed.

php8-mod-dev or php8-mod-devel is also not available (we require phpize in order to compile xdebug)

Please add support for these extensions.

mhei, 2022-09-24:
I've prepared a PR in https://github.com/openwrt/packages/pull/19439 which adds Xdebug PECL extension. However, you stated that your environment is 21.02 which is feature complete. That means, that this new package will not make it into 21.02 "official" branch, but maybe you can backport itself.

php8-mod-dev or php8-mod-devel sounds like packages you know from PC Linux distros. While OpenWrt provides some infrastructure to compile natively on the target devices, I personally don't see this as main focus. That's why there are currently no development packages for PHP available. And while I don't dislike the idea per se, I don't have time or plan to work on this - so patches are welcome :smile:

mhei, 2022-09-25:
PR merged, but I did not reference this issue correctly, so closing manually.

closed 2022-09-25 by mhei

barhom, 2022-09-25:
Thank you for this. I have successfully compiled php8-pecl-xdebug (backported to 21.02).