openwrt/packages #19464: tailscale: does not work as it can not find iptables in $PATH

kaikikokuke, 2022-09-26:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @kesslerm @BKPepe  (find it by checking history of the package Makefile)
Environment: 

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.0'
DISTRIB_REVISION='r19685-512e76967f'
DISTRIB_TARGET='ipq40xx/generic'
DISTRIB_ARCH='arm_cortex-a7_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt 22.03.0 r19685-512e76967f'
DISTRIB_TAINTS=''

Description:

I updated the OpenWRT to version 22.03.0 and install tailscale. After rebooting I tried running tailscale up and got the error that iptables can not be found in path. I assume this has something to do with the change to nftables, howevert I am not sure. I am also willing to help to solve the issue.

```
root@OpenWrt:~# tailscaled
logtail started
Program starting: v1.24.2-dev-t, Go 1.18.4: []string{"tailscaled"}
LogID: a5cd4a2f71dc9...
logpolicy: using system state directory "/var/lib/tailscale"
wgengine.NewUserspaceEngine(tun "tailscale0") ...
wgengine.NewUserspaceEngine(tun "tailscale0") error: creating router: exec: "iptables": executable file not found in $PATH
flushing log.
logger closing down
createEngine: creating router: exec: "iptables": executable file not found in $PATH
root@OpenWrt:~# 
```

obsy, 2022-09-26:
Install iptabless-nft

champtar, 2022-09-26:
Would be nice to have them support nft also (please someone open a feature request on tails ale tracker)

williamla, 2023-01-16:
Please see https://github.com/tailscale/tailscale/issues/4638 and https://github.com/tailscale/tailscale/issues/391 .

`opkg install iptables-nft` is the interim solution/requirement https://openwrt.org/docs/guide-user/services/vpn/tailscale/start

boretom, 2024-01-07:
@kaikikokuke did you follow the [OpenWrt Tailscale Guide](https://openwrt.org/docs/guide-user/services/vpn/tailscale/start). Specifically did you start tailscale with '--netfilter-mode=off'? 'nft' should already be installed on 22.03.