openwrt/packages #19472: mwan3: Legacy rules detected

thecode, 2022-09-27:
Maintainer: @feckert
Environment: ARMv7, Linksys WRT32X, OpenWrt 22.03.0
Package version: 2.11.1-1

**Description:**
According to https://github.com/openwrt/packages/issues/16818 and https://github.com/openwrt/packages/pull/17940 `mwan3` is migrated to `nftables`, I have only a single basic port forward rule (tried to disable it) and I get a warning about legacy rules:
[image]

**Content of `/etc/config.mwan3`:**
```
config globals 'globals'
	option mmx_mask '0x3F00'

config interface 'wan'
	option enabled '1'
	list track_ip '1.0.0.1'
	list track_ip '1.1.1.1'
	list track_ip '208.67.222.222'
	list track_ip '208.67.220.220'
	option family 'ipv4'
	option down '3'
	option up '6'
	option initial_state 'online'
	option track_method 'ping'
	option reliability '1'
	option count '1'
	option size '56'
	option max_ttl '60'
	option check_quality '0'
	option timeout '4'
	option interval '10'
	option failure_interval '5'
	option recovery_interval '5'

config interface 'wan6'
	option family 'ipv6'
	option down '3'
	option up '6'
	option initial_state 'online'
	option track_method 'ping'
	option reliability '1'
	option enabled '1'
	option count '1'
	option size '56'
	option max_ttl '60'
	option timeout '4'
	option interval '10'
	option failure_interval '5'
	option recovery_interval '5'
	list track_ip '2606:4700:4700::1001'
	list track_ip '2606:4700:4700::1111'
	list track_ip '2620:0:ccd::2'
	list track_ip '2620:0:ccc::2'

config interface 'lte'
	option enabled '1'
	option family 'ipv4'
	option down '5'
	option up '10'
	option initial_state 'online'
	list track_ip '1.0.0.1'
	list track_ip '1.1.1.1'
	list track_ip '208.67.222.222'
	list track_ip '208.67.220.220'
	option track_method 'ping'
	option reliability '1'
	option size '56'
	option max_ttl '60'
	option timeout '4'
	option interval '10'
	option failure_interval '5'
	option recovery_interval '5'
	option count '3'

config member 'wan_m1_w1'
	option interface 'wan'
	option metric '1'
	option weight '1'

config member 'wan6_m1_w1'
	option interface 'wan6'
	option metric '1'
	option weight '1'

config member 'lte_m2_w1'
	option interface 'lte'
	option metric '2'
	option weight '1'

config policy 'wan_lte'
	list use_member 'wan_m1_w1'
	list use_member 'lte_m2_w1'
	option last_resort 'unreachable'

config policy 'wan6_only'
	list use_member 'wan6_m1_w1'
	option last_resort 'unreachable'

config rule 'default_rule_v4'
	option dest_ip '0.0.0.0/0'
	option use_policy 'wan_lte'
	option family 'ipv4'

config rule 'default_rule_v6'
	option dest_ip '::/0'
	option use_policy 'wan6_only'
	option family 'ipv6'

```

Note: I have also tried to remove anything related to IPv6, just to rule out if it is related to IPv6

When checking the `iptables` rules (clicking on the link to `/cgi-bin/luci/admin/status/nftables/iptables`) the `Mangle` table shows `mwan3` related rules, all other tables are empty.

**Output of `iptables -vL -t mangle`:**
```
Chain PREROUTING (policy ACCEPT 3143K packets, 2699M bytes)
 pkts bytes target     prot opt in     out     source               destination
3144K 2699M mwan3_hook  all  --  any    any     anywhere             anywhere

Chain INPUT (policy ACCEPT 188K packets, 38M bytes)
 pkts bytes target     prot opt in     out     source               destination

Chain FORWARD (policy ACCEPT 2901K packets, 2638M bytes)
 pkts bytes target     prot opt in     out     source               destination

Chain OUTPUT (policy ACCEPT 177K packets, 25M bytes)
 pkts bytes target     prot opt in     out     source               destination
 178K   25M mwan3_hook  all  --  any    any     anywhere             anywhere

Chain POSTROUTING (policy ACCEPT 3076K packets, 2663M bytes)
 pkts bytes target     prot opt in     out     source               destination

Chain mwan3_connected_ipv4 (2 references)
 pkts bytes target     prot opt in     out     source               destination
56364   24M MARK       all  --  any    any     anywhere             anywhere             match-set mwan3_connected_ipv4 dst MARK or 0x3f00

Chain mwan3_custom_ipv4 (2 references)
 pkts bytes target     prot opt in     out     source               destination
    0     0 MARK       all  --  any    any     anywhere             anywhere             match-set mwan3_custom_ipv4 dst MARK or 0x3f00

Chain mwan3_dynamic_ipv4 (2 references)
 pkts bytes target     prot opt in     out     source               destination
    0     0 MARK       all  --  any    any     anywhere             anywhere             match-set mwan3_dynamic_ipv4 dst MARK or 0x3f00

Chain mwan3_hook (2 references)
 pkts bytes target     prot opt in     out     source               destination
3290K 2722M CONNMARK   all  --  any    any     anywhere             anywhere             mark match 0x0/0x3f00 CONNMARK restore mask 0x3f00
 111K   31M mwan3_ifaces_in  all  --  any    any     anywhere             anywhere             mark match 0x0/0x3f00
 104K   30M mwan3_custom_ipv4  all  --  any    any     anywhere             anywhere             mark match 0x0/0x3f00
 104K   30M mwan3_connected_ipv4  all  --  any    any     anywhere             anywhere             mark match 0x0/0x3f00
47911 6114K mwan3_dynamic_ipv4  all  --  any    any     anywhere             anywhere             mark match 0x0/0x3f00
47911 6114K mwan3_rules  all  --  any    any     anywhere             anywhere             mark match 0x0/0x3f00
3321K 2724M CONNMARK   all  --  any    any     anywhere             anywhere             CONNMARK save mask 0x3f00
3222K 2694M mwan3_custom_ipv4  all  --  any    any     anywhere             anywhere             mark match ! 0x3f00/0x3f00
3222K 2694M mwan3_connected_ipv4  all  --  any    any     anywhere             anywhere             mark match ! 0x3f00/0x3f00
3222K 2694M mwan3_dynamic_ipv4  all  --  any    any     anywhere             anywhere             mark match ! 0x3f00/0x3f00

Chain mwan3_iface_in_lte (1 references)
 pkts bytes target     prot opt in     out     source               destination
    0     0 MARK       all  --  3g-lte any     anywhere             anywhere             match-set mwan3_custom_ipv4 src mark match 0x0/0x3f00 /* default */ MARK or 0x3f00
    0     0 MARK       all  --  3g-lte any     anywhere             anywhere             match-set mwan3_connected_ipv4 src mark match 0x0/0x3f00 /* default */ MARK or 0x3f00
    0     0 MARK       all  --  3g-lte any     anywhere             anywhere             match-set mwan3_dynamic_ipv4 src mark match 0x0/0x3f00 /* default */ MARK or 0x3f00
  396 54698 MARK       all  --  3g-lte any     anywhere             anywhere             mark match 0x0/0x3f00 /* lte */ MARK xset 0x300/0x3f00

Chain mwan3_iface_in_wan (1 references)
 pkts bytes target     prot opt in     out     source               destination
    0     0 MARK       all  --  pppoe-wan any     anywhere             anywhere             match-set mwan3_custom_ipv4 src mark match 0x0/0x3f00 /* default */ MARK or 0x3f00
    0     0 MARK       all  --  pppoe-wan any     anywhere             anywhere             match-set mwan3_connected_ipv4 src mark match 0x0/0x3f00 /* default */ MARK or 0x3f00
    0     0 MARK       all  --  pppoe-wan any     anywhere             anywhere             match-set mwan3_dynamic_ipv4 src mark match 0x0/0x3f00 /* default */ MARK or 0x3f00
 5862  264K MARK       all  --  pppoe-wan any     anywhere             anywhere             mark match 0x0/0x3f00 /* wan */ MARK xset 0x100/0x3f00

Chain mwan3_ifaces_in (1 references)
 pkts bytes target     prot opt in     out     source               destination
 110K   31M mwan3_iface_in_wan  all  --  any    any     anywhere             anywhere             mark match 0x0/0x3f00
 105K   30M mwan3_iface_in_lte  all  --  any    any     anywhere             anywhere             mark match 0x0/0x3f00

Chain mwan3_policy_wan6_only (0 references)
 pkts bytes target     prot opt in     out     source               destination
    0     0 MARK       all  --  any    any     anywhere             anywhere             mark match 0x0/0x3f00 /* unreachable */ MARK xset 0x3e00/0x3f00

Chain mwan3_policy_wan_lte (1 references)
 pkts bytes target     prot opt in     out     source               destination
47785 6103K MARK       all  --  any    any     anywhere             anywhere             mark match 0x0/0x3f00 /* wan 1 1 */ MARK xset 0x100/0x3f00

Chain mwan3_rules (1 references)
 pkts bytes target     prot opt in     out     source               destination
47911 6114K mwan3_policy_wan_lte  all  --  any    any     anywhere             anywhere             mark match 0x0/0x3f00
```

Any recommendations how to debug this issue?

feckert, 2022-09-28:
Yes, unfortunately, that is correct.
The mwan3 has not yet been converted to native nftables, but it works via iptables-nft.
Since this is not a matter of packages filtering but of packages marking.
The mwan3 can be used in parallel with nftables via iptables-nft backend, which generates nft rules.
Have a look at 'nft list ruleset'.

Unfortunately, due to time constraints, I am not able to switch the mwan3 to native nft.
Pullrequest are coming.

jamesmacwhite, 2023-05-01:
Someone updated the Wiki to mention `iptables-nft` and `ip6tables-nft` after a few forum posts about issues, but I've made it clearer in the installation steps.

https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#installation

Hopefully in the future mwan3 can support nftables natively, but I've made it clearer for now.

feckert, 2023-05-03:
Thank you for correcting that on the wiki.

Megachip, 2024-09-25:
Issue is closed, but I still get the notification on latest snapshot version :/

mangkoran, 2025-09-30:
I'm also still getting "Legacy rules detected" notice on 24.10.3.

[image]

Could we reopen this issue while `mwan3` still not updated to use native nftables?