openwrt/packages #19485: miniupnpd-nftables: multiple issues in 22.03.0 r19685-512e76967f

birdie-github, 2022-09-29:
Maintainer: @stintel @ldir-EDB0 @neheb
Environment: 22.03.0 r19685-512e76967f , TP-Link TL-WDR3600 v1.

```
opkg install miniupnpd-nftables
Installing miniupnpd-nftables (2022-08-31-68c8ec50-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/packages/miniupnpd-nftables_2022-08-31-68c8ec50-1_mips_24kc.ipk
Installing libcap-ng (0.8.2-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/packages/libcap-ng_0.8.2-1_mips_24kc.ipk
Installing librt (1.2.3-4) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0/targets/ath79/generic/packages/librt_1.2.3-4_mips_24kc.ipk
Installing libuuid1 (2.37.3-3) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/base/libuuid1_2.37.3-3_mips_24kc.ipk
Configuring libcap-ng.
Configuring librt.
Configuring libuuid1.
Configuring miniupnpd-nftables.
uci: Invalid argument
uci: Invalid argument

Section @redirect[0] (FTP) is disabled, ignoring section
Section @redirect[1] (HTTP) is disabled, ignoring section
Section @redirect[2] (HTTPS) is disabled, ignoring section
Section @redirect[6] (iperf) is disabled, ignoring section
Section @redirect[7] (RDP 1) is disabled, ignoring section
Section @redirect[8] (RDP 2) is disabled, ignoring section
Section @include[0] is not marked as compatible with fw4, ignoring section
Section @include[0] requires 'option fw4_compatible 1' to be considered compatible
Automatically including '/usr/share/nftables.d/table-post/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/dstnat/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/forward/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/srcnat/20-miniupnpd.nft'
```
Followed by a two minutes pause, then
```
Include '/usr/share/miniupnpd/firewall.include' failed with exit code -9
Collected errors:
 * resolve_conffiles: Existing conffile /etc/config/upnpd is different from the conffile in the new package. The new conffile will be placed at /etc/config/upnpd-opkg.
```

I've got questions:

 *  What does work now?  What doesn't work now?  How to check it?
 *  What are these errors? How to deal with them?

dropwhile, 2022-10-22:
Same errors here. "OpenWrt 22.03.2"

dropwhile, 2022-10-22:
@birdie-github I followed the instructions in the release notes [here](https://openwrt.org/releases/22.03/notes-22.03.2#known_issues) and upgraded the `firewall4` package, and it seemed to help.

birdie-github, 2022-10-22:
> I followed the instructions in the release notes [here](https://openwrt.org/releases/22.03/notes-22.03.2#known_issues) and upgraded the `firewall4` package, and it seemed to help.

I'm waiting for the next .point release which should be released shortly. There are glaring major [errors](https://www.openwall.com/lists/oss-security/2022/10/13/2) in Wi-Fi (DoS/remote root with no interaction) which has been fixed recently.

birdie-github, 2022-10-22:
> I followed the instructions in the release notes [here](https://openwrt.org/releases/22.03/notes-22.03.2#known_issues) and upgraded the `firewall4` package, and it seemed to help.

I'm waiting for the next .point release which should be released shortly. There are glaring major [errors](https://www.openwall.com/lists/oss-security/2022/10/13/2) in Wi-Fi (DoS/remote root with no interaction) which have been fixed recently.