openwrt/packages #19488: mwan3: routes not being added

dyarkovoy, 2022-09-29:
Maintainer: @feckert
Environment: RTL8380, RTL8380, OpenWrt OpenWrt SNAPSHOT r20775-165b66d910
Package version: 2.11.1-3

Description:

I get the following errors upon mwan3 startup:
```
Thu Sep 29 16:47:21 2022 user.warn mwan3rtmon[5761]: failed to add 192.168.1.0/24 dev switch.1 proto kernel scope link src 192.168.1.1 offload to table 1 - error: Error: either "to" is duplicate, or "offload" is a garbage.
Thu Sep 29 16:47:21 2022 user.warn mwan3rtmon[5761]: failed to add 192.168.1.0/24 dev switch.1 proto kernel scope link src 192.168.1.1 offload to table 2 - error: Error: either "to" is duplicate, or "offload" is a garbage.
Thu Sep 29 16:47:21 2022 user.warn mwan3rtmon[5761]: failed to add 192.168.1.0/24 dev switch.1 proto kernel scope link src 192.168.1.1 offload to table 3 - error: Error: either "to" is duplicate, or "offload" is a garbage.
```
and traffic is not routed from lan to wan* interfaces.
When I manully add missing routes, everything works fine:

```
ip route add 192.168.1.0/24 dev switch.1 proto kernel scope link src 192.168.1.1 table 1
ip route add 192.168.1.0/24 dev switch.1 proto kernel scope link src 192.168.1.1 table 2
ip route add 192.168.1.0/24 dev switch.1 proto kernel scope link src 192.168.1.1 table 3
```

dyarkovoy, 2022-09-30:
I think I found the problem.
[This line](https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/common.sh#L143) does not account for a new modifier "offload" that probably has to do with hardware offloading supported by this switch.
I suggest PR #19492 to fix it

raintonr, 2022-12-11:
OpenWrt 22.03.2 on Netgear GS308T v1 here.

PR #19520 does indeed fix the problem. Thanks :smiley:

closed 2022-12-12 by feckert