openwrt/packages #19491: https-dns-proxy: do NOT enable option force_dns 1 by default

birdie-github, 2022-09-30:
Please for the love of God do not enable this option by default.

I've just spent three hours trying to understand why name resolution has stopped working for my connected devices.

Turns out this benign option
```
/etc/config/https-dns-proxy
config main 'config'
        option force_dns '1'
```

Breaks everything because it actually **adds firewall rules** to disable access to port 853!

Nothing in OpenWRT should add any firewall rules by default.

birdie-github, 2022-09-30:
You could at least add this text:

```
config main 'config'
        option force_dns '1' # this option will add firewall rules which block clients from accessing outside DNS servers
```

birdie-github, 2022-09-30:
Maintainer: @stangri

stangri, 2022-09-30:
Between the config file and [README](https://docs.openwrt.melmac.net/https-dns-proxy/) it's pretty obvious what the service does and how to change the default behavior.

closed 2022-09-30 by stangri

birdie-github, 2022-09-30:
> Between the config file and [README](https://docs.openwrt.melmac.net/https-dns-proxy/) it's pretty obvious what the service does and how to change the default behavior.

Packages installed by default must **not** change firewall rules, let along **block** traffic.

There are other DNS over TLS servers for OpenWRT and none of them do this crap.

Why `https-dns-proxy` suddenly has its own mind, I've no idea.

birdie-github, 2022-09-30:
And adding explanatory text to the config costs you nothing.

If you're so occupied I can submit a merge request.

birdie-github, 2022-09-30:
https://github.com/openwrt/packages/pull/19494

stangri, 2022-09-30:
> Packages installed by default must **not** change firewall rules, let along **block** traffic.

I disagree. Packages installed must most efficiently cover the needs of the large majority of users and provide options for less often used/fringe cases. OpenWrt firewall itself is pretty restrictive, to most efficiently cover majority of uses.

> There are other DNS over TLS servers for OpenWRT and none of them do this crap.

I'm sure their users are happy that despite installing a package to encrypt DNS requests, they may still be leaking unencrypted DNS requests which are then likely resold by their ISPs.

birdie-github, 2022-09-30:
> I'm sure their users are happy that despite installing a package to encrypt DNS requests, they may still be leaking unencrypted DNS requests which are then likely resold by their ISPs.

You install a package with a feature, instead you're shoved a whole new unintended feature. Speaking of the README you're so eager to push in my face.

The first two pages of it do **not** mention any firewalling taking place **by default**. It's only close to the bottom where you see this option and its explanation.

God, it's all so dishonest, insincere and "I decide what's best for everyone". OK, never mind. Sorry for wasting your time.

> they may still be leaking unencrypted DNS requests which are then likely resold by their ISPs

If the user is smart enough to find this package and install it, I'm sure as hell they can figure out how to enable this feature for their devices.

I have a desktop PC, laptop and smartphone all of which have DoT enabled and used exclusively. I don't send a single unencrypted DNS request. It's only for my friends who need WiFi I've installed this package.

I used `stubby` for years but the latest firmware is so large, I've got basically no free space after installing the package, that's why I opted for `https-dns-proxy`. Didn't expect my DNS resolution to break spectacularly but whatever.

birdie-github, 2022-10-02:
Lastly, imagine https-dns-proxy stops working for some reasons, a segfault, configured servers are inaccessible, etc.

This means your local users can no longer access the internet at all.

Meister1593, 2024-05-04:
Just spent multiple hours debugging, checking every single openwrt forums, finding how to dump tcp packets and finding out my DoT traffic from android got implicitly blocked by this.
This is not said anywhere except for this issue when i found it out...
IMO this is very bad to not warn users about port blocking

birdie-github, 2024-05-04:
> Just spent multiple hours debugging, checking every single openwrt forum, finding how to dump tcp packets and finding out my DoT traffic from android got implicitly blocked by this. This is not said anywhere except for this issue when i found it out... IMO this is very bad to not warn users about port blocking

The respective maintainer thinks otherwise and doesn't accept any counter-arguments.

stangri, 2024-05-05:
> Just spent multiple hours debugging, checking every single openwrt forum, finding how to dump tcp packets and finding out my DoT traffic from android got implicitly blocked by this. This is not said [...]

That is incorrect, explicitly mentioned in the README and is part of the config-file: https://docs.openwrt.melmac.net/https-dns-proxy/#force_dns

The WebUI also has numerous links to the relevant README sections.

stangri, 2024-05-05:
> The respective maintainer thinks otherwise and doesn't accept any counter-arguments.

Your counter-arguments have been heard and you can disable the default behaviour.

sinancetinkaya, 2025-01-08:
also removing `https-dns-proxy` breaks the router's internet access

stangri, 2025-01-08:
> also removing `https-dns-proxy` breaks the router's internet access. 

Can't reproduce. With the default settings it doesn't happen, it only happens if user manually adjust DNS settings and then remove the proxy package.

> Shouldn't `https-dns-proxy` revert all the changes it made when removed?

It reverts all the changes it made when it's stopped (prior to removal).

sinancetinkaya, 2025-01-08:
> Can't reproduce. With the default settings it doesn't happen, it only happens if user manually adjust DNS settings and then remove the proxy package.
>
>
> It reverts all the changes it made when it's [...]

In case someone removes it before stopping it, can you add the revert steps for uninstall to the readme?

sinancetinkaya, 2025-01-08:
> Can't reproduce. With the default settings it doesn't happen, it only happens if user manually adjust DNS settings and then remove the proxy package.
>
>
> It reverts all the changes it made when it's [...]

Sorry, it turned out that the problem was PBR as mentioned here #25616