openwrt/packages #19500: python pyproject.toml

lucize, 2022-10-01:
what is needed so we can support pyproject.toml packages ?
I would like to update homeassistant but it needs orjson that needs rust and it seems that using a generated setup.py is not working.
I know rust is not in yet but I have it built and the hostfiles are there

karlp, 2023-01-12:
I'm running into this as well, for a new package: https://pypi.org/project/asyncio-mqtt
@commodo  and @BKPepe  You're the python maintainers?

karlp, 2023-01-12:
I've worked around it by using the "recommended stub" setup.py, but is that something that we might like to try and create internally if it's not available in the project itself?    @lucize here's an example of patching a package directly, that might be sufficient? https://github.com/etactica/owrt_pub_feeds/blob/wip/egm400-202211/lang/python/python-asyncio-mqtt/patches/0001-support-legacy-setup.py-installation.patch

lucize, 2023-01-12:
I know that workaround, but I guess a modern approach would be desired, but seems that rust is still kept in PR status

commodo, 2023-01-12:
> I'm running into this as well, for a new package: https://pypi.org/project/asyncio-mqtt @commodo and @BKPepe You're the python maintainers?

so, i "think" proposed a "workaround" (a while back) with the build infra-structure creating that `setup.py` stub;
now, there can be a debate about whether that's a good idea (and i think we've had it; but i am not so sure about it);

(personally, i think the `setup.py` stub-thingi is not a bad idea for a while, given that the inner-workings of `pyproject.toml` still elude me, and i didn't/don't have time to really dig into how that works)

officially, @jefferyto is the current maintainer of Python;
i've pinged him via email (a few days ago), but no reply yet;
he's probably busy with other stuff (which is totally fine)
i'm running into the pyproject.toml with updating `chardet`

i resigned from being an official maintainer of Python (due to some disagreement with Jeffery; which is not nice to say in public, but ¯\\\_(ツ)_/¯);

now, in the past, Jeffery has had some absences with some package stuff (and that's totally fine);
but usually, he would show up after a while (or me doing some things), and then disagree with me on some stuff;
(now that may be really overly simplified and me being an asshole)

but the point of the above is that i was thinking of maintaining Python again, seeing that it's still at 3.10.7 in the feeds, and 3.11.x is out already, and this `pyproject.toml ` thingi
and it may also cause Jeffery to re-appear :P   (and i don't know of other ways to do that)

commodo, 2023-01-12:
Oh.
So, I found this PR https://github.com/openwrt/packages/pull/17130
I still have that branch around.

We can re-open it, if needed

BKPepe, 2023-01-17:
I am afraid that I am not the Python maintainer, but @jefferyto with @commodo are. :)

commodo, 2023-01-22:
Right.
I opened PR https://github.com/openwrt/packages/pull/20366 for this.

jefferyto, 2023-05-19:
This was done in #20801, so I'm going to close this.

> and it may also cause Jeffery to re-appear :P (and i don't know of other ways to do that)

This was... correct :sweat_smile:

closed 2023-05-19 by jefferyto