openwrt/packages #19508: prometheus-node-exporter-lua: netclass.lua may produce duplicated '# TYPE' lines

PichetGoulu, 2022-10-02:
Maintainer: @champtar
Environment: Ubiquiti EdgeRouter X, OpenWrt 22.03.0 r19685-512e76967f
Package version: 2022.06.12-1

Description:

The netclass.lua collector may produce duplicated `# TYPE node_network_*` lines.
This is invalid per the [Prometheus documentation on the exposition format](https://prometheus.io/docs/instrumenting/exposition_formats/#comments-help-text-and-type-information).
> If the token is TYPE, exactly two more tokens are expected. The first is the metric name, and the second is either counter, gauge, histogram, summary, or untyped, defining the type for the metric of that name. *Only one TYPE line may exist for a given metric name.* The TYPE line for a metric name must appear before the first sample is reported for that metric name. If there is no TYPE line for a metric name, the type is set to untyped.

On my router with `prometheus-node-exporter-lua` and `prometheus-node-exporter-lua-nat_traffic` installed:

```
% wget -q http://192.168.0.254:9100/metrics -O - | grep "# TYPE" | sort | uniq -cd
      8 # TYPE node_network_address_assign_type gauge
      8 # TYPE node_network_carrier_changes_total counter
      8 # TYPE node_network_carrier_down_changes_total counter
      8 # TYPE node_network_carrier gauge
      8 # TYPE node_network_carrier_up_changes_total counter
      8 # TYPE node_network_device_id gauge
      8 # TYPE node_network_dormant gauge
      8 # TYPE node_network_flags gauge
      8 # TYPE node_network_iface_id gauge
      8 # TYPE node_network_iface_link gauge
      8 # TYPE node_network_iface_link_mode gauge
      8 # TYPE node_network_info gauge
      8 # TYPE node_network_mtu_bytes gauge
      8 # TYPE node_network_net_dev_group gauge
      8 # TYPE node_network_protocol_type gauge
      5 # TYPE node_network_speed_bytes gauge
      8 # TYPE node_network_transmit_queue_length gauge
```

Using the `inputs.prometheus' plugin of Telegraf to scrape the metrics produces an error.

```
E! [inputs.prometheus] Error in plugin: error reading metrics for http://192.168.0.254:9100/metrics: reading text format failed: text format parsing error in line 268: second TYPE line for metric name "node_network_info", or TYPE reported after samples
```

The netclass.lua collector has been developed by @rtreffer and merged in PR #18170

rtreffer, 2022-10-03:
I am traveling at the moment, I think this complaint is valid but only under strict validation. Prometheus itself has no notion of types and thus accepts multiple `# TYPE` statements (IIRC they are just ignored) so this will impact Telegraf and potentially openmetrics based consumers.

I took a quick look at the code and [it looks like each outputted metric is prefixed by `# TYPE` in the metric function](https://github.com/openwrt/packages/blob/master/utils/prometheus-node-exporter-lua/files/usr/bin/prometheus-node-exporter-lua#L45).

I don't think this is strictly related to the net collector, it should impact anything that outputs multiple metrics that are only distinct by labels.

@champtar yeah, I think it would be great if you could take a look, I think the complaint is valid (though not impacting prometheus as such) and potentially tied to the overall lua prometheus node exporter. I would be curious if the same issue happens with the wifi exporter.

PichetGoulu, 2022-10-03:
Looking at the `scrape()` function in wifi.lua and wifi_stations.lua: all metrics are defined first with `metric()` then used to add labels and values.

rtreffer, 2022-10-03:
Oh right, if called without value then it returns a function to return it. Ok, so it just needs to be moved up.

rtreffer, 2022-10-03:
Ok, I have a replacement, but have some other stuff to do now till tomorrow evening.... @champtar I can turn this into a PR tomorrow. Or feel free to put it in.
```lua
local ubus = require "ubus"

-- reference/further reading:
-- - node_exporter netclass_linux (upstream metrics): https://github.com/prometheus/node_exporter/blob/master/collector/netclass_linux.go
-- - relevant sysfs files: https://github.com/prometheus/procfs/blob/5f46783c017ef6a934fc8cfa6d1a2206db21401b/sysfs/net_class.go#L121
-- - get devices / read files: https://github.com/openwrt/packages/blob/openwrt-21.02/utils/prometheus-node-exporter-lua/files/usr/lib/lua/prometheus-collectors/snmp6.lua

local function get_devices() -- based on hostapd_stations.lua
 	local u = ubus.connect()
	local status = u:call("network.device", "status", {})
	local devices = {}

	for dev, dev_table in pairs(status) do
		table.insert(devices, dev)
	end
	return devices
end

local function load(device, file) -- load a single sysfs file, trim trailing newline, return "" on error
	local success, data = pcall(function () return string.gsub(get_contents("/sys/class/net/" .. device .. "/" .. file), "\n$", "") end)
	if success then
		return data
	else
		return ""
	end
end

local function get_metric(device, metric_node_network)
	local address = load(device, "address")
	local broadcast = load(device, "broadcast")
	local duplex = load(device, "duplex")
	local operstate = load(device, "operstate")
	local ifalias = load(device, "ifalias")
	metric_node_network.info({device = device, address = address, broadcast = broadcast, duplex = duplex, operstate = operstate, ifalias = ifalias}, 1)
	local speed = load(device, "speed")
	if not tonumber(speed) == nil then
		metric_node_network_speed_bytes({device = device}, tonumber(speed)*1000*1000/8)
	end
	local file_to_metric = {
		addr_assign_type   = "address_assign_type",
		carrier            = "carrier",
		carrier_changes    = "carrier_changes_total",
		carrier_up_count   = "carrier_up_changes_total",
		carrier_down_count = "carrier_down_changes_total",
		dev_id             = "device_id",
		dormant            = "dormant",
		flags              = "flags",
		ifindex            = "iface_id",
		iflink             = "iface_link",
		link_mode          = "iface_link_mode",
		mtu                = "mtu_bytes",
		name_assign_type   = "name_assign_type",
		netdev_group       = "net_dev_group",
		tx_queue_len       = "transmit_queue_length",
		type               = "protocol_type",
	}
	for file, metric in pairs(file_to_metric) do
		local value = load(device, file)
		if not tonumber(value) == nil then
			metric_node_network[metric](tonumber(value))
		end
	end
end

local function scrape()
	local metric_node_network = {
		info                       = metric("node_network_info", "gauge"),
		address_assign_type        = metric("node_network_address_assign_type", "gauge"),
		carrier                    = metric("node_network_carrier", "gauge"),
		carrier_changes_total      = metric("node_network_carrier_changes_total", "counter"),
		carrier_up_changes_total   = metric("node_network_carrier_up_changes_total", "counter"),
		carrier_down_changes_total = metric("node_network_carrier_down_changes_total", "counter"),
		device_id                  = metric("node_network_device_id", "gauge"),
		dormant                    = metric("node_network_dormant", "gauge"),
		flags                      = metric("node_network_flags", "gauge"),
		iface_id                   = metric("node_network_iface_id", "gauge"),
		iface_link                 = metric("node_network_iface_link", "gauge"),
		iface_link_mode            = metric("node_network_iface_link_mode", "gauge"),
		mtu_bytes                  = metric("node_network_mtu_bytes", "gauge"),
		name_assign_type           = metric("node_network_name_assign_type", "gauge"),
		net_dev_group              = metric("node_network_net_dev_group", "gauge"),
		transmit_queue_length      = metric("node_network_transmit_queue_length", "gauge"),
		protocol_type              = metric("node_network_protocol_type", "gauge"),
		speed_bytes                = metric("node_network_speed_bytes", "gauge"),
	}
	for _, devicename in ipairs(get_devices()) do
		get_metric(devicename, metric_node_network)
	end
end

return { scrape = scrape }
```
@PichetGoulu could you perhaps give this a try?

frizziedad, 2022-11-25:
I had exactly this problem with reporting Prometheus stats on my OpenWRT Netgear router and using an Influxdb2 scraper. I can also confirm that replacing the netclass.lua collector with this version fixed the issue. Took me quite a while and a lot of googling to eventually find this post so thank you for providing a fix.

wrenix, 2024-05-20:
i found multiple values with same metrics and labels in my output, so i have to uninstall prometheus-node-exporter-lua-nat_traffic.

curl -q "http://192.168.1.11:9100/metrics" | grep -v "^#" | sed -E 's/\}? .*//' | sort | uniq -c

```
    ....
      7 node_nat_traffic{dest="192.168.1.10",src="192.168.1.2"
     12 node_nat_traffic{dest="192.168.1.255",src="192.168.1.50"
     ....
```

PR #24319 (2024-06-05): prometheus-node-exporter-lua: fix netclass duplicate TYPE lines
Fixes a315c40b7232bbc83582685c98e41466d84d7a35

[initial fix]

[fixup René version]

[actual commit]

Maintainer: me

Fixes #19508
closed 2024-06-05 by champtar