openwrt/packages #19511: glib2: undefined reference to `__fn_local_printf_frexpl' for mipsel_24kc_24kf

MoetaYuko, 2022-10-03:
Maintainer: @neheb
Environment: mipsel_24kc_24kf snapshot sdk provided by `openwrt/sdk:mipsel_24kc_24kf-snapshot` docker image

Description:

```
2022-10-03T03:54:11.8678551Z FAILED: glib/libglib-2.0.so.0.7400.0 
2022-10-03T03:54:11.8689537Z mipsel-openwrt-linux-musl-gcc  -o glib/libglib-2.0.so.0.7400.0 glib/libglib-2.0.so.0.7400.0.p/deprecated_gallocator.c.o glib/libglib-2.0.so.0.7400.0.p/deprecated_gcache.c.o glib/libglib-2.0.so.0.7400.0.p/deprecated_gcompletion.c.o glib/libglib-2.0.so.0.7400.0.p/deprecated_grel.c.o glib/libglib-2.0.so.0.7400.0.p/deprecated_gthread-deprecated.c.o glib/libglib-2.0.so.0.7400.0.p/garcbox.c.o glib/libglib-2.0.so.0.7400.0.p/garray.c.o glib/libglib-2.0.so.0.7400.0.p/gasyncqueue.c.o glib/libglib-2.0.so.0.7400.0.p/gatomic.c.o glib/libglib-2.0.so.0.7400.0.p/gbacktrace.c.o glib/libglib-2.0.so.0.7400.0.p/gbase64.c.o glib/libglib-2.0.so.0.7400.0.p/gbitlock.c.o glib/libglib-2.0.so.0.7400.0.p/gbookmarkfile.c.o glib/libglib-2.0.so.0.7400.0.p/gbytes.c.o glib/libglib-2.0.so.0.7400.0.p/gcharset.c.o glib/libglib-2.0.so.0.7400.0.p/gchecksum.c.o glib/libglib-2.0.so.0.7400.0.p/gconvert.c.o glib/libglib-2.0.so.0.7400.0.p/gdataset.c.o glib/libglib-2.0.so.0.7400.0.p/gdate.c.o glib/libglib-2.0.so.0.7400.0.p/gdatetime.c.o glib/libglib-2.0.so.0.7400.0.p/gdir.c.o glib/libglib-2.0.so.0.7400.0.p/genviron.c.o glib/libglib-2.0.so.0.7400.0.p/gerror.c.o glib/libglib-2.0.so.0.7400.0.p/gfileutils.c.o glib/libglib-2.0.so.0.7400.0.p/ggettext.c.o glib/libglib-2.0.so.0.7400.0.p/ghash.c.o glib/libglib-2.0.so.0.7400.0.p/ghmac.c.o glib/libglib-2.0.so.0.7400.0.p/ghook.c.o glib/libglib-2.0.so.0.7400.0.p/ghostutils.c.o glib/libglib-2.0.so.0.7400.0.p/giochannel.c.o glib/libglib-2.0.so.0.7400.0.p/gkeyfile.c.o glib/libglib-2.0.so.0.7400.0.p/glib-init.c.o glib/libglib-2.0.so.0.7400.0.p/glib-private.c.o glib/libglib-2.0.so.0.7400.0.p/glist.c.o glib/libglib-2.0.so.0.7400.0.p/gmain.c.o glib/libglib-2.0.so.0.7400.0.p/gmappedfile.c.o glib/libglib-2.0.so.0.7400.0.p/gmarkup.c.o glib/libglib-2.0.so.0.7400.0.p/gmem.c.o glib/libglib-2.0.so.0.7400.0.p/gmessages.c.o glib/libglib-2.0.so.0.7400.0.p/gnode.c.o glib/libglib-2.0.so.0.7400.0.p/goption.c.o glib/libglib-2.0.so.0.7400.0.p/gpattern.c.o glib/libglib-2.0.so.0.7400.0.p/gpoll.c.o glib/libglib-2.0.so.0.7400.0.p/gprimes.c.o glib/libglib-2.0.so.0.7400.0.p/gqsort.c.o glib/libglib-2.0.so.0.7400.0.p/gquark.c.o glib/libglib-2.0.so.0.7400.0.p/gqueue.c.o glib/libglib-2.0.so.0.7400.0.p/grand.c.o glib/libglib-2.0.so.0.7400.0.p/grcbox.c.o glib/libglib-2.0.so.0.7400.0.p/grefcount.c.o glib/libglib-2.0.so.0.7400.0.p/grefstring.c.o glib/libglib-2.0.so.0.7400.0.p/gregex.c.o glib/libglib-2.0.so.0.7400.0.p/gscanner.c.o glib/libglib-2.0.so.0.7400.0.p/gsequence.c.o glib/libglib-2.0.so.0.7400.0.p/gshell.c.o glib/libglib-2.0.so.0.7400.0.p/gslice.c.o glib/libglib-2.0.so.0.7400.0.p/gslist.c.o glib/libglib-2.0.so.0.7400.0.p/gstdio.c.o glib/libglib-2.0.so.0.7400.0.p/gstrfuncs.c.o glib/libglib-2.0.so.0.7400.0.p/gstring.c.o glib/libglib-2.0.so.0.7400.0.p/gstringchunk.c.o glib/libglib-2.0.so.0.7400.0.p/gstrvbuilder.c.o glib/libglib-2.0.so.0.7400.0.p/gtestutils.c.o glib/libglib-2.0.so.0.7400.0.p/gthread.c.o glib/libglib-2.0.so.0.7400.0.p/gthreadpool.c.o glib/libglib-2.0.so.0.7400.0.p/gtimer.c.o glib/libglib-2.0.so.0.7400.0.p/gtimezone.c.o glib/libglib-2.0.so.0.7400.0.p/gtrace.c.o glib/libglib-2.0.so.0.7400.0.p/gtranslit.c.o glib/libglib-2.0.so.0.7400.0.p/gtrashstack.c.o glib/libglib-2.0.so.0.7400.0.p/gtree.c.o glib/libglib-2.0.so.0.7400.0.p/guniprop.c.o glib/libglib-2.0.so.0.7400.0.p/gutf8.c.o glib/libglib-2.0.so.0.7400.0.p/gunibreak.c.o glib/libglib-2.0.so.0.7400.0.p/gunicollate.c.o glib/libglib-2.0.so.0.7400.0.p/gunidecomp.c.o glib/libglib-2.0.so.0.7400.0.p/guri.c.o glib/libglib-2.0.so.0.7400.0.p/gutils.c.o glib/libglib-2.0.so.0.7400.0.p/guuid.c.o glib/libglib-2.0.so.0.7400.0.p/gvariant.c.o glib/libglib-2.0.so.0.7400.0.p/gvariant-core.c.o glib/libglib-2.0.so.0.7400.0.p/gvariant-parser.c.o glib/libglib-2.0.so.0.7400.0.p/gvariant-serialiser.c.o glib/libglib-2.0.so.0.7400.0.p/gvarianttypeinfo.c.o glib/libglib-2.0.so.0.7400.0.p/gvarianttype.c.o glib/libglib-2.0.so.0.7400.0.p/gversion.c.o glib/libglib-2.0.so.0.7400.0.p/gwakeup.c.o glib/libglib-2.0.so.0.7400.0.p/gprintf.c.o glib/libglib-2.0.so.0.7400.0.p/glib-unix.c.o glib/libglib-2.0.so.0.7400.0.p/gspawn.c.o glib/libglib-2.0.so.0.7400.0.p/giounix.c.o glib/libglib-2.0.so.0.7400.0.p/gjournal-private.c.o glib/libglib-2.0.so.0.7400.0.p/gthread-posix.c.o -L/home/build/openwrt/staging_dir/toolchain-mipsel_24kc+24kf_gcc-11.3.0_musl/usr/lib -L/home/build/openwrt/staging_dir/toolchain-mipsel_24kc+24kf_gcc-11.3.0_musl/lib -flto -Wl,--as-needed -Wl,--no-undefined -shared -fPIC -Wl,--start-group -Wl,-soname,libglib-2.0.so.0 -znow -zrelro -Wl,--gc-sections -Wl,-rpath,/home/build/openwrt/staging_dir/target-mipsel_24kc+24kf_musl/usr/lib -Wl,-rpath-link,/home/build/openwrt/staging_dir/target-mipsel_24kc+24kf_musl/usr/lib glib/libcharset/libcharset.a glib/gnulib/libgnulib.a -Wl,-z,nodelete -Wl,-Bsymbolic-functions /home/build/openwrt/staging_dir/target-mipsel_24kc+24kf_musl/usr/lib/libpcre2-8.so -pthread -lm -Wl,--end-group
2022-10-03T03:54:11.8696368Z /home/build/openwrt/staging_dir/toolchain-mipsel_24kc+24kf_gcc-11.3.0_musl/bin/../lib/gcc/mipsel-openwrt-linux-musl/11.3.0/../../../../mipsel-openwrt-linux-musl/bin/ld: /home/build/openwrt/tmp/ccrpKuEw.ltrans11.ltrans.o: in function `$L1074':
2022-10-03T03:54:11.8696995Z <artificial>:(.text+0x4284): undefined reference to `__fn_local_printf_frexpl'
2022-10-03T03:54:11.8697851Z /home/build/openwrt/staging_dir/toolchain-mipsel_24kc+24kf_gcc-11.3.0_musl/bin/../lib/gcc/mipsel-openwrt-linux-musl/11.3.0/../../../../mipsel-openwrt-linux-musl/bin/ld: <artificial>:(.text+0x428a): undefined reference to `__fn_local_printf_frexpl'
2022-10-03T03:54:11.8698405Z collect2: error: ld returned 1 exit status
2022-10-03T03:54:11.8698697Z ninja: build stopped: subcommand failed.
2022-10-03T03:54:11.8699224Z make[3]: *** [Makefile:122: /home/build/openwrt/build_dir/target-mipsel_24kc+24kf_musl/glib-2.74.0/.built] Error 1
2022-10-03T03:54:11.8699743Z make[3]: Leaving directory '/home/build/openwrt/feeds/packages/libs/glib2'
2022-10-03T03:54:11.8708916Z time: package/feeds/packages/glib2/compile#36.42#8.70#46.40
2022-10-03T03:54:11.8720107Z     ERROR: package/feeds/packages/glib2 failed to build.
2022-10-03T03:54:11.8738961Z make[2]: *** [package/Makefile:116: package/feeds/packages/glib2/compile] Error 1
2022-10-03T03:54:11.9061647Z make[2]: *** Waiting for unfinished jobs....
```
https://downloads.openwrt.org/snapshots/faillogs/mipsel_24kc_24kf/packages/glib2/compile.txt

neheb, 2022-10-03:
yep. I have absolutely no idea how to fix.

The issue is somehow related to gnulib.

predators46, 2022-10-04:
i get this error build mysql. So far haven't been able to fix it. it's still mysterious

/home/ioio/mt7621/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/bin/../lib/gcc/mipsel-openwrt-linux-musl/11.2.0/../../../../mipsel-openwrt-linux-musl/bin/ld: my_atomic-t.o: in function `test_atomic_add64':                                                    
my_atomic-t.c:(.text+0x244): undefined reference to `__sync_fetch_and_add_8'           
/home/ioio/mt7621/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/bin/../lib/gcc/mipsel-openwrt-linux-musl/11.2.0/../../../../mipsel-openwrt-linux-musl/bin/ld: my_atomic-t.c:(.text+0x268): undefined reference to `__sync_fetch_and_add_8'                       
/home/ioio/mt7621/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/bin/../lib/gcc/mipsel-openwrt-linux-musl/11.2.0/../../../../mipsel-openwrt-linux-musl/bin/ld: my_atomic-t.c:(.text+0x27c): undefined reference to `__sync_fetch_and_add_8'                       
/home/ioio/mt7621/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/bin/../lib/gcc/mipsel-openwrt-linux-musl/11.2.0/../../../../mipsel-openwrt-linux-musl/bin/ld: my_atomic-t.c:(.text+0x288): undefined reference to `__sync_fetch_and_add_8'                       
/home/ioio/mt7621/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/bin/../lib/gcc/mipsel-openwrt-linux-musl/11.2.0/../../../../mipsel-openwrt-linux-musl/bin/ld: my_atomic-t.o: in function `do_tests':                                                             
my_atomic-t.c:(.text+0x7ac): undefined reference to `__sync_fetch_and_add_8'           
/home/ioio/mt7621/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/bin/../lib/gcc/mipsel-openwrt-linux-musl/11.2.0/../../../../mipsel-openwrt-linux-musl/bin/ld: my_atomic-t.o:my_atomic-t.c:(.text+0x7c0): more undefined references to `__sync_fetch_and_add_8' follow                                                                                   
collect2: error: ld returned 1 exit status                                             
Makefile:642: recipe for target 'my_atomic-t' failed                                   
make[5]: *** [my_atomic-t] Error 1

neheb, 2022-10-04:
That's a missing -latomic and unrelated to this issue.

predators46, 2022-10-04:
> That's a missing -latomic and unrelated to this issue.

how do i add that in the makefile?

neheb, 2022-10-04:
TARGET_LDFLAGS += -latomic .

predators46, 2022-10-04:
> TARGET_LDFLAGS += -latomic .

i have tried but get error. i will show the makefile maybe i'm wrong

#
# Copyright (C) 2006-2015 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk
include $(INCLUDE_DIR)/uclibc++.mk

PKG_NAME:=mysql
PKG_VERSION:=5.5.7-rc
PKG_RELEASE:=2

PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
PKG_SOURCE_URL:=\
	https://downloads.mysql.com/archives/get/p/23/file/

PKG_MD5SUM:=9b14f1c8136457ff29890d571e196f53
PKG_MAINTAINER:=Jo-Philipp Wich <jo@mein.io>
PKG_LICENSE:=GPL-2.0

PKG_BUILD_DEPENDS:=libncurses libreadline
PKG_BUILD_PARALLEL:=1
PKG_USE_MIPS16:=0

PKG_FIXUP:=libtool

include $(INCLUDE_DIR)/package.mk

define Package/libmysqlclient/Default
  SECTION:=libs
  CATEGORY:=Libraries
  DEPENDS:=$(CXX_DEPENDS) +zlib
  TITLE:=MySQL client library
  URL:=http://dev.mysql.com/
endef

define Package/mysql-server
  SECTION:=utils
  CATEGORY:=Utilities
  DEPENDS:=+libmysqlclient +libpthread +libncursesw +libreadline
  TITLE:=MySQL Server
  URL:=http://dev.mysql.com/
  SUBMENU:=database
endef

define Package/libmysqlclient
  $(call Package/libmysqlclient/Default)
endef

define Package/libmysqlclient-r
  $(call Package/libmysqlclient/Default)
  TITLE += threadsafe
  DEPENDS+= +libpthread
endef

ifneq ($(CONFIG_USE_UCLIBCXX),)
  TARGET_CXX=g++-uc
endif

TARGET_CFLAGS += $(FPIC)
TARGET_LDFLAGS += -latomic

CONFIGURE_ARGS += \
	--enable-shared \
	--enable-static \
	--enable-thread-safe-client \
	--disable-assembler \
	--with-pthread \
	--with-server \
	--without-mysqlmanager \
	--with-mysqld-user=root \
	--with-unix-socket-path=/tmp \
	--without-libwrap \
	--with-low-memory \
	--without-embedded-server \
	--without-query-cache \
	--without-ssl \
	--without-docs \
	--without-man \
	--without-readline \
	--without-debug \
	--with-named-thread-libs=-lpthread \
	CXXFLAGS="-Wno-narrowing -fpermissive"

CONFIGURE_VARS += \
	mysql_cv_compress=yes \
	mysql_cv_gethostname_style=glibc2 \
	mysql_cv_gcc_atomic_builtins=yes \
	mysql_cv_gcc_atomic_builtins_pthread_t=yes \
	ac_cv_c_stack_direction=-1

define Build/Compile
	+$(MAKE) $(PKG_JOBS) -C "$(PKG_BUILD_DIR)" \
		SUBDIRS="include" \
		DESTDIR="$(PKG_INSTALL_DIR)" \
		all
	$(MAKE) -C "$(PKG_BUILD_DIR)" \
		SUBDIRS="include" \
		DESTDIR="$(PKG_INSTALL_DIR)" \
		install
	g++ -I $(PKG_BUILD_DIR)/include \
		-o $(PKG_BUILD_DIR)/sql/gen_lex_hash \
		$(PKG_BUILD_DIR)/sql/gen_lex_hash.cc
	+$(MAKE) $(PKG_JOBS) -C "$(PKG_BUILD_DIR)/libmysql" \
		CC="g++" \
		CFLAGS="$(HOST_CFLAGS)" \
		CPPFLAGS="$(HOST_CFLAGS)" \
		LDFLAGS="$(HOST_LDFLAGS)" \
		conf_to_src
	+$(MAKE) $(PKG_JOBS) -C "$(PKG_BUILD_DIR)" \
		SUBDIRS="libmysql" \
		DESTDIR="$(PKG_INSTALL_DIR)" \
		all
	$(MAKE) -C "$(PKG_BUILD_DIR)" \
		SUBDIRS="libmysql" \
		DESTDIR="$(PKG_INSTALL_DIR)" \
		install
	+$(MAKE) $(PKG_JOBS) -C "$(PKG_BUILD_DIR)" \
		SUBDIRS="libmysql_r" \
		DESTDIR="$(PKG_INSTALL_DIR)" \
		all
	$(MAKE) -C "$(PKG_BUILD_DIR)" \
		SUBDIRS="libmysql_r" \
		DESTDIR="$(PKG_INSTALL_DIR)" \
		install
	$(MAKE) -C "$(PKG_BUILD_DIR)" \
		SUBDIRS="scripts" \
		DESTDIR="$(PKG_INSTALL_DIR)" \
		bin_SCRIPTS="mysql_config" \
		install
	+$(MAKE) $(PKG_JOBS) -C "$(PKG_BUILD_DIR)" \
		DESTDIR="$(PKG_INSTALL_DIR)" \
		all
	$(MAKE) -C "$(PKG_BUILD_DIR)" \
		DESTDIR="$(PKG_INSTALL_DIR)" \
		install
endef

define Build/InstallDev
	$(INSTALL_DIR) $(2)/bin $(1)/usr/bin $(1)/usr/include $(1)/usr/lib
	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/bin/mysql_config $(1)/usr/bin/
	ln -sf $(STAGING_DIR)/usr/bin/mysql_config $(2)/bin/
	$(CP) $(PKG_INSTALL_DIR)/usr/include/mysql $(1)/usr/include/
	# NOTE: needed for MySQL-Python
	$(CP) $(PKG_BUILD_DIR)/include/mysqld_error.h $(1)/usr/include/mysql/
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/mysql $(1)/usr/lib/
	rm -f $(1)/usr/lib/mysql/libmysqlclient.la
endef

define Package/libmysqlclient/install
	$(INSTALL_DIR) $(1)/usr/lib
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/mysql/libmysqlclient.so.* $(1)/usr/lib/
endef
define Package/libmysqlclient-r/install
	$(INSTALL_DIR) $(1)/usr/lib
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/mysql/libmysqlclient_r.so.* $(1)/usr/lib/
endef

define Package/mysql-server/install
	$(INSTALL_DIR) $(1)/usr/bin
	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/bin/mysql $(1)/usr/bin/
	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/lib/mysqld $(1)/usr/bin/
	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/bin/myisamchk $(1)/usr/bin/
	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/bin/mysqladmin $(1)/usr/bin/
	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/bin/mysqldump $(1)/usr/bin/
	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/bin/mysql_install_db $(1)/usr/bin/
	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/bin/my_print_defaults $(1)/usr/bin/
	$(INSTALL_DIR) $(1)/etc/init.d/
	$(INSTALL_BIN) files/mysqld.init $(1)/etc/init.d/mysqld
	$(INSTALL_CONF) conf/my.cnf $(1)/etc/
	$(INSTALL_DIR) $(1)/usr/share/mysql
	$(INSTALL_DIR) $(1)/usr/share/mysql/english
	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/share/mysql/english/errmsg.sys $(1)/usr/share/mysql/english
	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/share/mysql/fill_help_tables.sql $(1)/usr/share/mysql/
	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/share/mysql/mysql_system_tables.sql $(1)/usr/share/mysql/
	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/share/mysql/mysql_system_tables_data.sql $(1)/usr/share/mysql/
endef

define Package/mysql-server/conffiles
/etc/my.cnf
endef

$(eval $(call BuildPackage,mysql-server))
$(eval $(call BuildPackage,libmysqlclient))
$(eval $(call BuildPackage,libmysqlclient-r))

neheb, 2022-10-04:
That is...ancient.

predators46, 2022-10-10:
> That is...ancient.

Hm...

MoetaYuko, 2023-06-10:
@neheb issue can be fixed by adding `no-mips16` to `PKG_BUILD_FLAGS`, is this sane?

neheb, 2023-06-10:
Probably not.

MoetaYuko, 2023-06-11:
I found that locally disabling mips16 by adding `__attribute__((nomips16))` to `printf_frexpl` also fixes the issue.

The `__fn_local_printf_frexpl` symbol is a stub of `printf_frexpl` and is supposed to be automatically generated by the compiler for mips16 hard-float targets. Maybe for some reason (or compiler bug?) the stub is not generated so linker fails to linking?

The most interesting thing is that `printf_frexp` compiles fine while their implementations are nearly identical.

Ref:
https://github.com/gcc-mirror/gcc/blob/20c85207abd543c76818032632a5f1b3aea5495b/gcc/config/mips/mips.cc#L7307
https://github.com/llvm/llvm-project/blob/5217e7756c123972bd8340163ba20d05284ec24e/llvm/lib/Target/Mips/Mips16HardFloat.cpp#L453

neheb, 2023-06-11:
A patch like that makes sense.

Should be reported upstream in any case.

MoetaYuko, 2023-06-11:
@neheb Apart from the root cause of the issue, I figured out why it compiled fine previously.

Prior to https://github.com/openwrt/packages/commit/5d27631d9fbca076eb6377fc8c10f474ce9f4fea we had `006-c99.patch` which unconditionally disabled gnulib build. What about adding it back for now?

neheb, 2023-06-11:
Oh that is…interesting.

Adding it back sounds good.

PR #21337 (2023-06-11): glib2: disable gnulib printf
Maintainer: @neheb
Compile tested: mipsel_24kc_24kf master
Run tested: n/a

Description:
A compiler bug (suspiciously) blocks gnulib compilation for mipsel_24kc_24kf. While we had this patch to disable gnulib, it was accidentally removed by 5d27631d9fbca076eb6377fc8c10f474ce9f4fea. Add it back to fix the long broken build for mipsel_24kc_24kf.

Fixes: #19511
commit 68f23a9c (2023-06-11): glib2: disable gnulib printf

A compiler bug (suspiciously) blocks gnulib compilation for
mipsel_24kc_24kf. While we had this patch to disable gnulib, it was
accidentally removed by 5d27631d9fbca076eb6377fc8c10f474ce9f4fea. Add it
back to fix the long broken build for mipsel_24kc_24kf.

Fixes: #19511
PR #21338 (2023-06-11): [23.05] glib2: disable gnulib printf
Maintainer: @neheb
Compile tested: mipsel_24kc_24kf master
Run tested: n/a

Description:
A compiler bug (suspiciously) blocks gnulib compilation for mipsel_24kc_24kf. While we had this patch to disable gnulib, it was accidentally removed by 5d27631d9fbca076eb6377fc8c10f474ce9f4fea. Add it back to fix the long broken build for mipsel_24kc_24kf.

Fixes: #19511
commit 4ed85ea2 (2023-06-11): glib2: disable gnulib printf

A compiler bug (suspiciously) blocks gnulib compilation for
mipsel_24kc_24kf. While we had this patch to disable gnulib, it was
accidentally removed by 5d27631d9fbca076eb6377fc8c10f474ce9f4fea. Add it
back to fix the long broken build for mipsel_24kc_24kf.

Fixes: #19511
closed 2023-06-11 by neheb