openwrt/packages #19514: libqmi: ccache is not used during compile

bam80, 2022-10-04:
Maintainer: Nicholas Smith <nicholas@nbembedded.com> @aleksander0m @mips171 @neheb @Linaro1985 @kuanyili @feckert @micmac1
Environment: [docker pull openwrtorg/sdk:ramips-mt76x8-snapshot](https://hub.docker.com/layers/openwrtorg/sdk/ramips-mt76x8-snapshot/images/sha256-8203494b7cb195daa49a151bc81b353b01e60ccb38c798a638ef8bb759b26cd2?context=explore)
https://hub.docker.com/r/openwrtorg/sdk

Description:
With `CONFIG_CCACHE=y`, the package compilation doesn't use `ccache` which can be monitored with `CCACHE_DIR=.ccache ccache -s` during the compile.
`ccache` is used properly for other packages.

It's a pity because libqmi compilation is a quite time consuming.

neheb, 2022-10-04:
should be using it. check build_dir/target_x/libqmi-1.30.8/openwrt-build/compile_commands.json. ccache_cc is used on my end.

bam80, 2022-10-04:
Hmm, I don't have such file:
```
$ find build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/ 
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.pkgdir
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.pkgdir/qmi-utils.installed
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.pkgdir/libqmi.installed
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.pkgdir/qmi-utils
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.pkgdir/qmi-utils/usr
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.pkgdir/qmi-utils/usr/bin
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.pkgdir/qmi-utils/usr/bin/qmicli
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.pkgdir/qmi-utils/usr/bin/qmi-firmware-update
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.pkgdir/qmi-utils/usr/bin/qmi-network
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.pkgdir/libqmi
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.pkgdir/libqmi/usr
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.pkgdir/libqmi/usr/libexec
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.pkgdir/libqmi/usr/libexec/qmi-proxy
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.pkgdir/libqmi/usr/lib
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.pkgdir/libqmi/usr/lib/libqmi-glib.so.5.8.0
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.pkgdir/libqmi/usr/lib/libqmi-glib.so.5
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.prepared_2108283e5c880f3ea95064c41cb0cdea_18f1e190c5d53547fed41a3eaa76e9e9
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.prepared_2108283e5c880f3ea95064c41cb0cdea_18f1e190c5d53547fed41a3eaa76e9e9_check
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.autoremove
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.built_check
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.configured_a17fb5ef857664f03cd0ce37cc5ea591
build_dir/target-mipsel_24kc_musl/libqmi-1.30.8/.built
```
Are you building in the SDK?

neheb, 2022-10-04:
no. The SDK by default wipes the build directory. Need to disable AUTOREMOVE.

bam80, 2022-10-04:
You are right, for some reason I didn't see the ccache stats bumping (it probably happens in the very beginning). Thanks.

closed 2022-10-04 by bam80