openwrt/packages #19547: libwolfssl/libcurl: CVE-Fix (PR#19513) will crash https-dns-proxy

realizelol, 2022-10-07:
Maintainer: @tapper82 , @stangri , @ynezz , @neheb
Environment: OpenWrt 21.02.3 r16554-1d4dea6d4f - x86_64 - Generic v0 (ImageBuilder)

Description:
------------

I think since PR https://github.com/openwrt/packages/pull/19513 the actual packages @ repo [downloads.openwrt.org](https://downloads.openwrt.org/releases/21.02.3/packages/x86_64/packages/) will stop SSL support for curl.

Maybe in case of the missing commits in branch `openwrt-21.02` ?:
https://github.com/openwrt/packages/commits/master/net/curl/Makefile
-> https://github.com/openwrt/packages/commit/312ab68be498f0d895d8d4dc5b2f0fc857e45eb9
-> https://github.com/openwrt/packages/commit/b814eec3b59beec0e5610c3f0eda4429ba23132c
-> https://github.com/openwrt/packages/commit/03a32717bc15d8dd0c99e200dd64ae0bbd558c35
-> https://github.com/openwrt/packages/commit/c812153f8d4f73b3f82cb19e3b98c84ca680eecb
https://github.com/openwrt/packages/commits/openwrt-21.02/net/curl/Makefile

<br/>

<br/>

```
curl -v https://openwrt.org
* Protocol "https" not supported or disabled in libcurl
curl: (1) Protocol "https" not supported or disabled in libcurl
```

#### Syslog:
```
root@OpenWrt:~# grep -i 'https-dns-proxy' /var/log/syslog
Fri Oct  7 12:37:22 2022 user.notice https-dns-proxy: Starting service ✓
Fri Oct  7 12:37:22 2022 daemon.info https-dns-proxy[4494]: [W] 1665139042.858021 main.c:236 HTTP/3 is not supported by current libcurl
Fri Oct  7 12:37:23 2022 daemon.info https-dns-proxy[4494]: [F] 1665139043.083623 https-dns-proxy-2021-11-22/src/https_client.c:317 08C8: CURLOPT_SSLVERSION error 4: Error
Fri Oct  7 12:37:28 2022 daemon.info https-dns-proxy[6972]: [W] 1665139048.086884 main.c:236 HTTP/3 is not supported by current libcurl
Fri Oct  7 12:37:28 2022 daemon.info https-dns-proxy[6972]: [F] 1665139048.244363 https-dns-proxy-2021-11-22/src/https_client.c:317 EA6E: CURLOPT_SSLVERSION error 4: Error
Fri Oct  7 12:37:33 2022 daemon.info https-dns-proxy[7388]: [W] 1665139053.246573 main.c:236 HTTP/3 is not supported by current libcurl
Fri Oct  7 12:37:33 2022 daemon.info https-dns-proxy[7388]: [W] 1665139053.249886 main.c:119 EA6E: Query received before bootstrapping is completed, discarding.
Fri Oct  7 12:37:35 2022 daemon.info https-dns-proxy[7388]: [F] 1665139055.593302 https-dns-proxy-2021-11-22/src/https_client.c:317 A465: CURLOPT_SSLVERSION error 4: Error
Fri Oct  7 12:37:40 2022 daemon.info https-dns-proxy[7430]: [W] 1665139060.596336 main.c:236 HTTP/3 is not supported by current libcurl
Fri Oct  7 12:37:41 2022 daemon.info https-dns-proxy[7430]: [F] 1665139061.270342 https-dns-proxy-2021-11-22/src/https_client.c:317 89C1: CURLOPT_SSLVERSION error 4: Error
Fri Oct  7 12:37:46 2022 daemon.info https-dns-proxy[7451]: [W] 1665139066.271223 main.c:236 HTTP/3 is not supported by current libcurl
Fri Oct  7 12:37:46 2022 daemon.info https-dns-proxy[7451]: [W] 1665139066.273961 main.c:119 89C1: Query received before bootstrapping is completed, discarding.
Fri Oct  7 12:37:46 2022 daemon.info https-dns-proxy[7451]: [F] 1665139066.601962 https-dns-proxy-2021-11-22/src/https_client.c:317 B7CC: CURLOPT_SSLVERSION error 4: Error
Fri Oct  7 12:37:51 2022 daemon.info https-dns-proxy[7495]: [W] 1665139071.604249 main.c:236 HTTP/3 is not supported by current libcurl
Fri Oct  7 12:37:51 2022 daemon.info https-dns-proxy[7495]: [W] 1665139071.604720 main.c:119 A465: Query received before bootstrapping is completed, discarding.
Fri Oct  7 12:37:51 2022 daemon.info https-dns-proxy[7495]: [W] 1665139071.604844 main.c:119 7CC2: Query received before bootstrapping is completed, discarding.
Fri Oct  7 12:37:53 2022 daemon.info https-dns-proxy[7495]: [F] 1665139073.683444 https-dns-proxy-2021-11-22/src/https_client.c:317 A807: CURLOPT_SSLVERSION error 4: Error
Fri Oct  7 12:37:53 2022 daemon.info procd: Instance https-dns-proxy::instance1 s in a crash loop 6 crashes, 2 seconds since last crash
```

#### Installed packages (ERROR):
```
opkg list-installed | grep -i 'curl\|ssl\|https-dns-proxy'
curl - 7.83.1-2.1
https-dns-proxy - 2021-11-22-3
libcurl4 - 7.83.1-2.1
libopenssl-conf - 1.1.1q-1
libopenssl1.1 - 1.1.1q-1
libustream-wolfssl20201210 - 2022-01-16-868fd881-2
libwolfssl5.5.1.99a5b54a - 5.5.1-stable-2
luci-ssl - git-20.244.36115-e10f954
openssl-util - 1.1.1q-1
php8-mod-curl - 8.0.24-1
px5g-wolfssl - 4.1
wget-ssl - 1.21.1-1
```

#### Installed packages (WORKING):
```
opkg list-installed | grep -i 'curl\|ssl\|https-dns-proxy'
curl - 7.83.1-1
https-dns-proxy - 2021-11-22-3
libcurl4 - 7.83.1-1
libopenssl-conf - 1.1.1q-1
libopenssl1.1 - 1.1.1q-1
libustream-wolfssl20201210 - 2022-01-16-868fd881-1
libwolfssl5.2.0.99a5b54a - 5.2.0-stable-2
luci-ssl - git-20.244.36115-e10f954
openssl-util - 1.1.1q-1
php8-mod-curl - 8.0.20-1
px5g-wolfssl - 3
wget-ssl - 1.21.1-1
```

best regards
realizelol

ynezz, 2022-10-07:
Sorry for the breakage, should be fixed in #19548.

wentuq, 2022-10-07:
Same issue, can I download somewhere fixed package, or have to wait until it appears in official repo?

realizelol, 2022-10-07:
> Same issue, can I download somewhere fixed package, or have to wait until it appears in official repo?

<br/>

Until it's fixed you could use the archive repo:
https://archive.openwrt.org/releases/
https://archive.openwrt.org/releases/21.02.3/packages/x86_64/packages/ e.g.
and download these packages which aren't affected:
```
curl - 7.83.1-1 (packages)
libcurl4 - 7.83.1-1 (packages)
libwolfssl5.2.0.99a5b54a - 5.2.0-stable-2 (base)
```

e.g. for 21.02 x86_64 - step by step:
```dash
wget -q https://archive.openwrt.org/releases/21.02.3/packages/x86_64/packages/curl_7.83.1-2.1_x86_64.ipk
wget -q https://archive.openwrt.org/releases/21.02.3/packages/x86_64/packages/libcurl4_7.83.1-2.1_x86_64.ipk
wget -q https://archive.openwrt.org/releases/21.02.3/packages/x86_64/base/libwolfssl5.2.0.99a5b54a_5.2.0-stable-2_x86_64.ipk
opkg install libwolfssl5.2.0.99a5b54a_5.2.0-stable-2_x86_64.ipk
opkg install libcurl4_7.83.1-2.1_x86_64.ipk
opkg install curl_7.83.1-2.1_x86_64.ipk 
```
Maybe you have to swap the order of the install. But opkg will inform you about the dependency error if it occurs.
There is no reboot needed. But some services dependent on libcurl so it would be recommended to do so.

<br/> 

best regards
realizelol

closed 2022-10-08 by ynezz

ray-308, 2022-10-08:
Sorry for my noob question, but what packages to use for an raspberry pi 4b?

ntninja, 2022-10-09:
When is this going to be fixed for `mips_24kc`? The current `7.83.1-2.1` version of `libcurl4` appears to still be affected by this issue?

ynezz, 2022-10-10:
> When is this going to be fixed for `mips_24kc`? The current `7.83.1-2.1` version of `libcurl4` appears to still be affected by this issue?

When the build workers finishes the package rebuilds for those affected archs https://buildbot.openwrt.org/openwrt-21.02/packages/#/waterfall

mietzen, 2022-10-15:
> > Same issue, can I download somewhere fixed package, or have to wait until it appears in official repo?
> 
> 
> Until it's fixed you could use the archive repo: https://archive.openwrt.org/releases/ https://archive.openwrt.org/releases/21.02.3/packages/x86_64/packages/ e.g. and download these packages which aren't affected:
> 
> ```
> curl - 7.83.1-1 (packages)
> libcurl4 - 7.83.1-1 (packages)
> libwolfssl5.2.0.99a5b54a - 5.2.0-stable-2 (base)
> ```
> 
> e.g. for 21.02 x86_64 - step by step:
> 
> ```
> wget -q https://archive.openwrt.org/releases/21.02.3/packages/x86_64/packages/curl_7.83.1-2.1_x86_64.ipk
> wget -q https://archive.openwrt.org/releases/21.02.3/packages/x86_64/packages/libcurl4_7.83.1-2.1_x86_64.ipk
> wget -q https://archive.openwrt.org/releases/21.02.3/packages/x86_64/base/libwolfssl5.2.0.99a5b54a_5.2.0-stable-2_x86_64.ipk
> opkg install libwolfssl5.2.0.99a5b54a_5.2.0-stable-2_x86_64.ipk
> opkg install libcurl4_7.83.1-2.1_x86_64.ipk
> opkg install curl_7.83.1-2.1_x86_64.ipk 
> ```
> 
> Maybe you have to swap the order of the install. But opkg will inform you about the dependency error if it occurs. There is no reboot needed. But some services dependent on libcurl so it would be recommended to do so.
> 
> 
> best regards realizelol

The versions are incorrect!
https://forum.openwrt.org/t/curl-stopped-working-for-https-after-latest-woflssl-patch-21-02/139000/46?u=mietz

It must be:

```
curl - 7.83.1-1 (packages)
libcurl4 - 7.83.1-1 (packages)
libwolfssl5.2.0.99a5b54a - 5.2.0-stable-2 (base)
```

```
wget -q https://archive.openwrt.org/releases/21.02.3/packages/x86_64/packages/curl_7.83.1-1_x86_64.ipk
wget -q https://archive.openwrt.org/releases/21.02.3/packages/x86_64/packages/libcurl4_7.83.1-1_x86_64.ipk
wget -q https://archive.openwrt.org/releases/21.02.3/packages/x86_64/base/libwolfssl5.2.0.99a5b54a_5.2.0-stable-2_x86_64.ipk
opkg install --force-reinstall libwolfssl5.2.0.99a5b54a_5.2.0-stable-2_x86_64.ipk
opkg install --force-reinstall libcurl4_7.83.1-1_x86_64.ipk
opkg install --force-reinstall curl_7.83.1-1_x86_64.ipk
```

realizelol, 2022-10-16:
> > > Same issue, can I download somewhere fixed package, or have to wait until it appears in official repo?
> 
> > 
> 
> > 
> 
> > Until it's fixed you could use the archive repo: https://archive.openwrt.org/releases/ https://archive.openwrt.org/releases/21.02.3/packages/x86_64/packages/ e.g. and download these packages which aren't affected:
> 
> > 
> 
> > ```
> 
> > curl - 7.83.1-1 (packages)
> 
> > libcurl4 - 7.83.1-1 (packages)
> 
> > libwolfssl5.2.0.99a5b54a - 5.2.0-stable-2 (base)
> 
> > ```
> 
> > 
> 
> > e.g. for 21.02 x86_64 - step by step:
> 
> > 
> 
> > ```
> 
> > wget -q https://archive.openwrt.org/releases/21.02.3/packages/x86_64/packages/curl_7.83.1-2.1_x86_64.ipk
> 
> > wget -q https://archive.openwrt.org/releases/21.02.3/packages/x86_64/packages/libcurl4_7.83.1-2.1_x86_64.ipk
> 
> > wget -q https://archive.openwrt.org/releases/21.02.3/packages/x86_64/base/libwolfssl5.2.0.99a5b54a_5.2.0-stable-2_x86_64.ipk
> 
> > opkg install libwolfssl5.2.0.99a5b54a_5.2.0-stable-2_x86_64.ipk
> 
> > opkg install libcurl4_7.83.1-2.1_x86_64.ipk
> 
> > opkg install curl_7.83.1-2.1_x86_64.ipk 
> 
> > ```
> 
> > 
> 
> > Maybe you have to swap the order of the install. But opkg will inform you about the dependency error if it occurs. There is no reboot needed. But some services dependent on libcurl so it would be recommended to do so.
> 
> > 
> 
> > 
> 
> > best regards realizelol
> 
> 
> 
> The versions are incorrect!
> 
> https://forum.openwrt.org/t/curl-stopped-working-for-https-after-latest-woflssl-patch-21-02/139000/46?u=mietz
> 
> 
> 
> It must be:
> 
> 
> 
> ```
> 
> wget -q https://archive.openwrt.org/releases/21.02.3/packages/x86_64/packages/curl_7.83.1-1_x86_64.ipk
> 
> wget -q https://archive.openwrt.org/releases/21.02.3/packages/x86_64/packages/libcurl4_7.83.1-1_x86_64.ipk
> 
> wget -q https://archive.openwrt.org/releases/21.02.3/packages/x86_64/base/libwolfssl5.2.0.99a5b54a_5.2.0-stable-2_x86_64.ipk
> 
> opkg install --force-reinstall libwolfssl5.2.0.99a5b54a_5.2.0-stable-2_x86_64.ipk
> 
> opkg install --force-reinstall libcurl4_7.83.1-1_x86_64.ipk
> 
> opkg install --force-reinstall curl_7.83.1-1_x86_64.ipk
> 
> ```

Sorry, yes you're right. My mentioned packages versions are  correct but my links are not.

Yesterday there was an update to the actual repo (downloads.openwrt.org), so hopefully this bug has been fixed now.

<br>

best regards

realizelol

commit f4cc6adc (2022-10-08): curl: fix compilation with wolfSSL

options.h header is needed after bump of libwolfssl to version 5.5.1,
otherwise libcurl autodetection for libwolfssl availability fails and
libcurl is then compiled without https support.

Fixes: #19547

(cherry picked from commit 317575755a83fe21888439f1fd9adccca9e0f36e)
(cherry picked from commit ef545e0317e06cb388eec33c9dc4292f09473fdb)