openwrt/packages #19573: samba4: Hidden in menuconfig utill perl is selected

DazzyWalkman, 2022-10-10:
Recent update to samba4 https://github.com/openwrt/packages/commit/402f4ba4eff65b80a9deaa6085256112bec4d67b brought in perlbase-json-pp dependency.

It's kind of surprise to see samba4 related packages all hidden from menuconfig unless perl is selected.

I have to press "/"  to search for the samba4 definitions to find out why.

Samba4 should always be visible.

trippleflux, 2022-10-10:
It's already being [discuss ](https://forum.openwrt.org/t/samba-4-x-package-support-thread/10841/475)in the forum. Please try the recent changes from that link.

DazzyWalkman, 2022-10-10:
Thanks for the head-up. Now I see the related PR. Will give it a try.

DazzyWalkman, 2022-10-17:
Closed with 31ec2def9b5dfa35fa9e7bc40695a1713101e2a4

closed 2022-10-17 by DazzyWalkman