openwrt/packages #19590: libqmi: missing NAS option --nas-get-tx-rx-info

drathir, 2022-10-11:
Maintainer: @aleksander0m
Environment: (ramips, mt7620 , 22.03.1)

Description:

```
Seems like there is now missing option --nas-get-tx-rx-info from --help-nas at -V "qmicli 1.30.6" if not wrong and good remember works just fine at 21.02.3 version... Greetings...
```

aleksander0m, 2022-10-28:
What's the full output of `--help-nas`?

TuxPowered42, 2023-11-08:
`root@zyxel-lte:~# qmicli --version
qmicli 1.30.8
Copyright (C) 2012-2021 Aleksander Morgado
License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl-2.0.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

root@zyxel-lte:~# qmicli -d /dev/cdc-wdm0 -p --help-nas
Usage:
  qmicli [OPTION…] - Control QMI devices

NAS options:
  --nas-get-signal-strength                                                                                         Get signal strength
  --nas-get-signal-info                                                                                             Get signal info
  --nas-get-home-network                                                                                            Get home network
  --nas-get-serving-system                                                                                          Get serving system
  --nas-get-system-info                                                                                             Get system info
  --nas-get-technology-preference                                                                                   Get technology preference
  --nas-get-preferred-networks                                                                                      Get preferred networks
  --nas-set-preferred-networks=[[MCCMNC,access_tech],...]                                                           Set preferred networks list
  --nas-get-system-selection-preference                                                                             Get system selection preference
  --nas-set-system-selection-preference=[cdma-1x|cdma-1xevdo|gsm|umts|lte|td-scdma][,[automatic|manual=MCCMNC]]     Set system selection preference
  --nas-network-scan                                                                                                Scan networks
  --nas-get-plmn-name=[mccmnc]                                                                                      Get plmn name data
  --nas-get-rf-band-info                                                                                            Get RF Band Info
  --nas-noop                                                                                                        Just allocate or release a NAS client. Use with `--client-no-release-cid' and/or `--client-cid'`

aleksander0m, 2023-11-08:
libqmi has different "collections" to enable, and looks like this build was done with LIBQMI_COLLECTION_BASIC.
In order to support all QMI commands, you need to build with LIBQMI_COLLECTION_FULL. This should be part of the available libqmi package configuration in the openwrt buid.

feckert, 2023-12-21:
I'm closing the issue because the question has been answered from my point of view.

closed 2023-12-21 by feckert