openwrt/packages #19631: <https-dns-proxy>:  dnsmasq catching SIGTERM repeatedly on OpenWRT 22.03.X

clvgt12, 2022-10-18:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

Please see related github ticket 11006:  [https://github.com/openwrt/openwrt/issues/11006](https://github.com/openwrt/openwrt/issues/11006)

Problem statement:  https-dns-proxy package forces dnsmasq to capture SIGTERM, and repeatedly be restarted.
Workaround:  removal https-dns-proxy package stops the SIGTERM signal from being raised and caught by dnsmasq.
```
root@tomato:~/scripts# opkg remove https-dns-proxy luci-app-https-dns-proxy
Removing package luci-app-https-dns-proxy from root...
Collected errors:
 * print_dependents_warning: Package https-dns-proxy is depended upon by packages:
 * print_dependents_warning: 	luci-app-https-dns-proxy
 * print_dependents_warning: These might cease to work if package https-dns-proxy is removed.

 * print_dependents_warning: Force removal of this package with --force-depends.
 * print_dependents_warning: Force removal of this package and its dependents
 * print_dependents_warning: with --force-removal-of-dependent-packages.
root@tomato:~/scripts# opkg remove luci-app-https-dns-proxy https-dns-proxy
Removing package https-dns-proxy from root...
Stopping https-dns-proxy 2022-10-15-1 ✓
Not deleting modified conffile /etc/config/https-dns-proxy.
root@tomato:~/scripts# logread|grep SIGTERM
Tue Oct 18 13:26:05 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:26:11 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:26:19 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:26:25 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:26:34 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:26:39 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:26:48 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:26:53 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:27:01 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:27:07 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:27:16 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:27:21 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:27:32 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:27:38 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:27:46 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:27:51 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:28:19 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:28:24 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:28:34 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:28:39 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:28:48 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:28:53 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Tue Oct 18 13:30:17 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
root@tomato:~/scripts# /etc/init.d/log restart;/etc/init.d/dnsmasq restart;
udhcpc: started, v1.35.0
udhcpc: broadcasting discover
udhcpc: no lease, failing
root@tomato:~/scripts# /etc/init.d/log restart;/etc/init.d/dnsmasq restart;
udhcpc: started, v1.35.0
udhcpc: broadcasting discover
udhcpc: no lease, failing
root@tomato:~/scripts# logread|grep SIGTERM
root@tomato:~/scripts# logread|grep SIGTERM
root@tomato:~/scripts# logread|grep SIGTERM
root@tomato:~/scripts# logread|grep SIGTERM
root@tomato:~/scripts# logread|grep SIGTERM
root@tomato:~/scripts# logread|grep SIGTERM
root@tomato:~/scripts# logread|grep SIGTERM
root@tomato:~/scripts# logread|grep SIGTERM
root@tomato:~/scripts# logread|grep SIGTERM
root@tomato:~/scripts# logread|grep SIGTERM
root@tomato:~/scripts# logread|grep SIGTERM
root@tomato:~/scripts# logread|grep SIGTERM
root@tomato:~/scripts# logread|grep SIGTERM
root@tomato:~/scripts# logread|grep SIGTERM
root@tomato:~/scripts# logread|grep SIGTERM
root@tomato:~/scripts# logread|grep SIGTERM
root@tomato:~/scripts# logread|grep SIGTERM
root@tomato:~/scripts# logread|grep SIGTERM
root@tomato:~/scripts# logread|grep SIGTERM
root@tomato:~/scripts# do_sysupgrade.sh 
Usage: /root/scripts/do_sysupgrade.sh [RELEASE_LABEL]
Release 22.03.0 is currently running
A current stable release available for download is 22.03.2
A current stable release available for download is 21.02.5
OpenWRT runtime environment: openwrt ath79 generic
Hardware environment:
	Model ID=tplink_archer-a7-v5
	Model Name=tp-link_archer_a7_v5
	Board=tplink_archer-a7-v5

```

stangri, 2022-10-18:
Relevant configs?

clvgt12, 2022-10-18:
[git.zip](https://github.com/openwrt/packages/files/9814893/git.zip)

stangri, 2022-10-18:
Thanks. I believe SIGTERM is dnsmasq's reaction to `/etc/init.d/dnsmasq restart`. The `https-dns-proxy` restarts dnsmasq to ensure it uses the new (proxy) resolvers. I don't understand why it happens so often, do you have a flakey wan/wan6 interface? The service should get restarted (and causes dnsmasq to restart in turn) when wan/wan6 get updated.

If you're comfortable editing the init script on your router, can you try changing [lines 259 and 261](https://github.com/openwrt/packages/blob/master/net/https-dns-proxy/files/https-dns-proxy.init#L259-L261) from `"/etc/init.d/${packageName}" restart` to `"/etc/init.d/${packageName}" start`?

dibdot, 2022-10-19:
@stangri it might be not the best idea to put wan6 interfaces in your reload trigger logic, see an old, still unresolved ticket here: https://github.com/openwrt/openwrt/issues/5723#issuecomment-1040233241

stangri, 2022-10-19:
@dibdot thank you for pointing this out. Do you have an idea how can I determine if wan6 will be used when the service starts on boot and the wan6 may not even be up yet?

closed 2022-10-19 by stangri