openwrt/packages #19637: miniupnpd-nftables: produces multiple duplicate rules in upnp_prerouting and upnp_forward chains

bvasiliev, 2022-10-20:
Maintainer: @stintel @hnyman @tiagogaspar8 @jow- 
Environment: ath79/generic, OpenWrt 22.03.2 r19803-9a599fee93, kernel 5.10.146, package version 2022-08-31-68c8ec50-1

Description:
miniupnpd with nftables and fw4 creates multiple duplicated rules in firewall chains for same pairs of hosts and ports. The number of duplicate rules increases over time.

For example:
```
        chain upnp_forward {
                iif "eth0.2" th dport 45738 @nh,128,32 0xa0a0a02 @nh,72,8 0x6 accept
                iif "eth0.2" th dport 45738 @nh,128,32 0xa0a0a02 @nh,72,8 0x11 accept
                iif "eth0.2" th dport 45738 @nh,128,32 0xa0a0a02 @nh,72,8 0x6 accept
                iif "eth0.2" th dport 45738 @nh,128,32 0xa0a0a02 @nh,72,8 0x11 accept
                iif "eth0.2" th dport 45738 @nh,128,32 0xa0a0a02 @nh,72,8 0x6 accept
				... skip about 200 lines
                iif "eth0.2" th dport 45738 @nh,128,32 0xa0a0a02 @nh,72,8 0x11 accept
                iif "eth0.2" th dport 12002 @nh,128,32 0xa0a0a02 @nh,72,8 0x6 accept
                iif "eth0.2" th dport 12002 @nh,128,32 0xa0a0a02 @nh,72,8 0x6 accept
        }

        chain upnp_prerouting {
                iif "eth0.2" @nh,72,8 0x6 th dport 45738 dnat ip to 192.168.0.2:45738
                iif "eth0.2" @nh,72,8 0x11 th dport 45738 dnat ip to 192.168.0.2:45738
                iif "eth0.2" @nh,72,8 0x6 th dport 45738 dnat ip to 192.168.0.2:45738
                iif "eth0.2" @nh,72,8 0x11 th dport 45738 dnat ip to 192.168.0.2:45738
                iif "eth0.2" @nh,72,8 0x6 th dport 45738 dnat ip to 192.168.0.2:45738
				... skip about 200 lines
                iif "eth0.2" @nh,72,8 0x11 th dport 45738 dnat ip to 192.168.0.2:45738
                iif "eth0.2" @nh,72,8 0x6 th dport 20373 dnat ip to 192.168.0.2:12002
                iif "eth0.2" @nh,72,8 0x6 th dport 53634 dnat ip to 192.168.0.2:12002
        }
```

nms42, 2023-01-25:
Recently released version 2.3.2 fix the problem.

stintel, 2023-03-10:
This issue was reported for 22.03 so the Fixes: tag in master is not entirely correct. I will do a PR cherry-picking the bump to 22.03, but as I don't run 22.03 anywhere, someone else shall have to runtime test this before it can be merged.

closed 2023-03-10 by stintel

yesbox, 2025-09-13:
miniupnpd-nftables 2.3.7-r1 on OpenWrt 24.10.2 leaks entries in the upnp_forward chain. Rules are not removed from it, but they are from the upnp_prerouting chain. This happens both with PMP and UPnP.

commit f7c0e413 (2023-03-10): miniupnpd: bump to 2.3.3

Fixes: #19637
commit b1f684f0 (2023-03-16): miniupnpd: bump to 2.3.3

Fixes: #19637

(cherry picked from commit f7c0e413f4fac6bfc0c2d17b0bb87da359d9479a)