openwrt/packages #19641: MariaDB: Using mysql.sock causes permission denied error (with a fix)

brackenhill-mob, 2022-10-20:
Maintainer: @miska 
Environment: x86 running on Intel i3 with OpenWrt v22.03.01

Description:

Using any PHP connect call (ie WordPress or a simple test script), connections to the database fail with mysqli or pdo with the message "Permission denied"

Solution is to chmod 0777 ```/var/run/mysql``` (ie the folder holding the socket file)

I'm using nginx with uci disabled (I can't think this would make a difference but...). I've had this issue under both the release and nightly versions of OpenWrt and with different x86 hardware. My "fix" solves it every time.

Also just noticed that /init.d/mysqld sets ```rundir=/var/run/mysqld``` but the actual socket ends up in ```/var/run/mysql``` making ```/var/run/mysqld``` redundant.

miska, 2022-10-21:
Other possible solution is to make a user you are using to access the MariaDB database part of `mariadb` group. Thank you for your report, will look into that and the directories.

miska, 2022-10-21:
Hmmm, seems like run dir is never used and comment suggests that is was about socketdir, so yes, you are right, not 777, but 755, but still. Will fix that.

brackenhill-mob, 2022-10-22:
Changed my system to 755 and it works. Thanks for the quick fix in #19666

closed 2022-10-22 by brackenhill-mob