openwrt/packages #19650: softflowd: not able to capture traffic from huge downloads

HolgerHees, 2022-10-20:
Please make sure that the issue subject starts with `<package-name>: `

Maintainer: @jesferman 
Environment: OpenWrt 22.03.0 r19685-512e76967f / LuCI openwrt-22.03 branch git-22.245.77528-487e58a
MediaTek MT7621 ver:1 eco:3

Description:

I'm almost able to track all my traffic. including IPv6 traffic. But for some reason I miss some traffic.

e.g. If I download a 3.6 GB file, I can't see this in my captured flows.

my configuration looks like

```
config softflowd
  option enabled        '1'
  option interface      'br-lan2'
  option pcap_file      ''
  option timeout        'maxlife=60'
  option max_flows      '8192'
  option host_port      '192.168.0.50:2055'
  option pid_file       '/var/run/softflowd.pid'
  option control_socket '/var/run/softflowd.ctl'
  option export_version '9'
  option hoplimit       ''
  option tracking_level 'full'
  option track_ipv6     ''
  option sampling_rate  '1'
```

and the download finished inside this maxlife timeframe.

If I chow my interface statistics it shows me ~80GB just from my testing.

any ideas how can I proceed to debug?

normally I would expect that the summarized IN_BYTES are equal to the traffic shown in my interface statistics itself

for testing I'm using this url

wget https://ftp.gwdg.de/pub/opensuse/distribution/leap/15.4/iso/openSUSE-Leap-15.4-DVD-x86_64-Build243.2-Media.iso

rvandegrift, 2022-10-20:
softflowd has a max of 2GiB in either direction for all flows - see the "Timeouts" section of [1].  Since the download is still running, I'd expect a new flow to be created right after that one is expired.  But you then might need to aggregate those on your collector.

[1] - https://manpages.debian.org/bullseye/softflowd/softflowd.8.en.html

HolgerHees, 2022-10-20:
I already aggregate them together. But still there is traffic missing.

e.g. I started my download right now

wget https://ftp.gwdg.de/pub/opensuse/distribution/leap/15.4/iso/openSUSE-Leap-15.4-DVD-x86_64-Build243.2-Media.iso

after 700MB downloaded filesize I stopped this download.

during the whole time only this single flow was visible with "softflowctl  dump-flows | grep "134.76.12.6"

ACTIVE seq:61323 [134.76.12.6]:443 <> [192.168.0.124]:60226 proto:6 octets>:46360 packets>:8 octets<:112 packets<:2 start:2022-10-20T17:44:06.852 finish:(null).2140967628 tcp>:7f9c925c tcp<:36 flowlabel>:00000000 flowlabel<:0000001a vlan>:18 vlan<:0 ether:(null) <> (null)

I double checked also the IPv6 variant

----
root@gateway:~# nslookup ftp.gwdg.de
Server:         127.0.0.1
Address:        127.0.0.1:53

Non-authoritative answer:
ftp.gwdg.de     canonical name = ftp6.gwdg.de
Name:   ftp6.gwdg.de
Address: 134.76.12.6

Non-authoritative answer:
ftp.gwdg.de     canonical name = ftp6.gwdg.de
Name:   ftp6.gwdg.de
Address: 2001:638:60f:110::1:2

HolgerHees, 2022-10-20:
strange thing is that I don't see much more if I'm using tcpdump like

"tcpdump -n -i br-lan2  | grep 134.76.12.6"

```
18:14:22.417827 IP 192.168.0.124.37912 > 134.76.12.6.443: Flags [S], seq 1428727213, win 64240, options [mss 1460,sackOK,TS val 1936079179 ecr 0,nop,wscale 7], length 0
18:14:22.432837 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [S.], seq 125200151, ack 1428727214, win 28960, options [mss 1460,sackOK,TS val 676623708 ecr 1936079179,nop,wscale 10], length 0
18:14:22.433362 IP 192.168.0.124.37912 > 134.76.12.6.443: Flags [.], ack 1, win 502, options [nop,nop,TS val 1936079195 ecr 676623708], length 0
18:14:22.448856 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], ack 402, win 30, options [nop,nop,TS val 676623724 ecr 1936079195], length 0
18:14:22.453940 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [P.], seq 1:5342, ack 402, win 30, options [nop,nop,TS val 676623729 ecr 1936079195], length 5341
18:14:22.475521 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [P.], seq 5342:5393, ack 560, win 31, options [nop,nop,TS val 676623751 ecr 1936079220], length 51
18:14:22.494495 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 5393:19873, ack 817, win 32, options [nop,nop,TS val 676623770 ecr 1936079238], length 14480
18:14:22.510881 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 19873:21321, ack 817, win 32, options [nop,nop,TS val 676623786 ecr 1936079257], length 1448
18:14:22.510928 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 21321:22769, ack 817, win 32, options [nop,nop,TS val 676623786 ecr 1936079257], length 1448
18:14:22.510959 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 22769:24217, ack 817, win 32, options [nop,nop,TS val 676623786 ecr 1936079257], length 1448
18:14:22.510987 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 24217:25665, ack 817, win 32, options [nop,nop,TS val 676623786 ecr 1936079257], length 1448
18:14:22.511017 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 25665:27113, ack 817, win 32, options [nop,nop,TS val 676623786 ecr 1936079257], length 1448
18:14:22.511046 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 27113:28561, ack 817, win 32, options [nop,nop,TS val 676623786 ecr 1936079257], length 1448
18:14:22.511075 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 28561:30009, ack 817, win 32, options [nop,nop,TS val 676623786 ecr 1936079257], length 1448
18:14:22.511104 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 30009:31457, ack 817, win 32, options [nop,nop,TS val 676623786 ecr 1936079257], length 1448
18:14:22.511135 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 31457:32905, ack 817, win 32, options [nop,nop,TS val 676623786 ecr 1936079257], length 1448
18:14:22.511165 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 32905:34353, ack 817, win 32, options [nop,nop,TS val 676623786 ecr 1936079257], length 1448
18:14:22.511194 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 34353:35801, ack 817, win 32, options [nop,nop,TS val 676623786 ecr 1936079257], length 1448
18:14:22.511226 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 35801:37249, ack 817, win 32, options [nop,nop,TS val 676623786 ecr 1936079257], length 1448
18:14:22.511256 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 37249:38697, ack 817, win 32, options [nop,nop,TS val 676623786 ecr 1936079257], length 1448
18:14:22.511284 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 38697:40145, ack 817, win 32, options [nop,nop,TS val 676623786 ecr 1936079257], length 1448
18:14:22.511314 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 40145:41593, ack 817, win 32, options [nop,nop,TS val 676623786 ecr 1936079257], length 1448
18:14:22.511342 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 41593:43041, ack 817, win 32, options [nop,nop,TS val 676623786 ecr 1936079257], length 1448
18:14:22.511373 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 43041:44489, ack 817, win 32, options [nop,nop,TS val 676623786 ecr 1936079257], length 1448
18:14:22.511401 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 44489:45937, ack 817, win 32, options [nop,nop,TS val 676623786 ecr 1936079257], length 1448
18:14:22.511430 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 45937:47385, ack 817, win 32, options [nop,nop,TS val 676623786 ecr 1936079257], length 1448
18:14:22.511455 IP 134.76.12.6.443 > 192.168.0.124.37912: Flags [.], seq 47385:48833, ack 817, win 32, options [nop,nop,TS val 676623786 ecr 1936079257], length 1448
```

this is all what I get and if I cancle the dowload after while I get a lot of

```
18:14:35.036720 IP 192.168.0.124.37912 > 134.76.12.6.443: Flags [R.], seq 817, ack 1037501489, win 24566, options [nop,nop,TS val 1936091798 ecr 676636311], length 0
18:14:35.036962 IP 192.168.0.124.37912 > 134.76.12.6.443: Flags [R], seq 1428728030, win 0, length 0
18:14:35.036962 IP 192.168.0.124.37912 > 134.76.12.6.443: Flags [R], seq 1428728030, win 0, length 0
18:14:35.037243 IP 192.168.0.124.37912 > 134.76.12.6.443: Flags [R], seq 1428728030, win 0, length 0
18:14:35.037514 IP 192.168.0.124.37912 > 134.76.12.6.443: Flags [R], seq 1428728030, win 0, length 0
18:14:35.037514 IP 192.168.0.124.37912 > 134.76.12.6.443: Flags [R], seq 1428728030, win 0, length 0
18:14:35.037772 IP 192.168.0.124.37912 > 134.76.12.6.443: Flags [R], seq 1428728030, win 0, length 0
[... cut ...]
18:14:35.051135 IP 192.168.0.124.37912 > 134.76.12.6.443: Flags [R], seq 1428728030, win 0, length 0
18:14:35.051394 IP 192.168.0.124.37912 > 134.76.12.6.443: Flags [R], seq 1428728030, win 0, length 0
18:14:35.051651 IP 192.168.0.124.37912 > 134.76.12.6.443: Flags [R], seq 1428728030, win 0, length 0
18:14:35.051936 IP 192.168.0.124.37912 > 134.76.12.6.443: Flags [R], seq 1428728030, win 0, length 0
18:14:35.052132 IP 192.168.0.124.37912 > 134.76.12.6.443: Flags [R], seq 1428728030, win 0, length 0
```

rvandegrift, 2022-10-20:
I just tried to replicate with your config, and it's working okay for me.  After ~500MiB downloaded:
```
# cat /etc/config/softflowd
config softflowd
	option enabled        '1'
	option interface      'eth0.2'
	option pcap_file      ''
	option timeout        'maxlife=60'
	option max_flows      '8192'
	option host_port      ''
	option pid_file       '/var/run/softflowd.pid'
	option control_socket '/var/run/softflowd.ctl'
	option export_version '9'
	option hoplimit       ''
	option tracking_level 'full'
	option track_ipv6     ''
	option sampling_rate  '1'
# softflowctl dump-flows | grep 134.76.12.6
ACTIVE seq:169 [10.23.5.162]:48140 <> [134.76.12.6]:443 proto:6 octets>:7047144 packets>:134549 octets<:559843243 packets<:374971 start:2022-10-20T18:21:54.769 finish:2022-10-20T18:22:38.740 tcp>:1f tcp<:1a flowlabel>:00000000 flowlabel<:00000000 vlan>:0 vlan<:0 ether:00:00:00:00:00:00 <> 00:00:00:00:00:00
```

softflowd has a debug mode:
```
# /etc/init.d/softflowd stop
# softflowd -i eth0.2 -t maxlife=60 -m 8192 -v 9 -T full -d -s 1 -D
```
I don't know if it's output will be useful.

Is your device CPU constrained?  In the past, I've tried to run softflow on lower-powered routers and it was unable to keep up with the traffic.  For reference, the above test was run on a raspberry pi 4.

HolgerHees, 2022-10-20:
I I start the same tcpdump on my laptop where the download is running I get the expected logs the whole time

tcpdump -i eth0 -n | grep 134.76.12.6 => this works on my laptop

so it looks like it is related to the capturing... maybe a network driver issues?

btw. thanks for spending time on reproducing

HolgerHees, 2022-10-20:
I will now focus on getting tcpdump output working before.

btw. the cpuload is still low during this download

and I'm using https://mikrotik.com/product/hex_s

HolgerHees, 2022-10-21:
the problem seams to be related to my device. because tcpdump is also missing these traffic.

closed 2022-10-21 by HolgerHees

HolgerHees, 2022-10-21:
I found the issues.

I enabled "Software flow offloading" support and this bypassed the cpu and thats the reason why softflowd and tcpdump where not able to get these traffic.