openwrt/packages #19655: softflowd: do_page_fault(): sending SIGSEGV to softflowd for invalid read access from .....

HolgerHees, 2022-10-21:
Maintainer: @jesferman , @rvandegrift
Environment: OpenWrt 22.03.0, TP-Link Archer C7 v2, Qualcomm Atheros QCA9558 ver 1 rev 0

Description:

I'm running softflowd on a router where I get no flows when I call "softflowctl dump-flows"

When I call "softflowctl statistics" I see

```
root@OpenWrt:~# softflowctl statistics
softflowd[6688]: Accumulated statistics since 2022-10-21T07:12:06 UTC:
Number of active flows: 112
Packets processed: 20685
Fragments: 42
Ignored packets: 2 (2 non-IP, 0 too short)
Flows expired: 46 (0 forced)
Flows exported: 46 (79 records) in 4 packets (0 failures)
Packets received by libpcap: 21226
Packets dropped by libpcap: 0
Packets dropped by interface: 0

Expired flow statistics:  minimum       average       maximum
  Flow bytes:                  60        120147       1738147
  Flow packets:                 1           344          6350
  Duration:                  0.00s        47.31s        63.06s

Expired flow reasons:
       tcp =        14   tcp.rst =         1   tcp.fin =         0
       udp =         6      icmp =         0   general =         0
   maxlife =        25
over 2 GiB =         0
  maxflows =         0
   flushed =         0

Per-protocol statistics:     Octets      Packets   Avg Life    Max Life
           icmp (1):          10920          130      60.01s      60.02s
            tcp (6):         715552         2075      43.78s      63.06s
           udp (17):        4800272        13600      43.77s      63.01s
```

but when I call "softflowctl dump-flows" I get

```
root@OpenWrt:~# softflowctl dump-flows
softflowd[6688]: Dumping flow data:
```

an a kernel log message like 

```
Fri Oct 21 07:13:47 2022 kern.info kernel: [745743.417947] do_page_fault(): sending SIGSEGV to softflowd for invalid read access from 00000326
Fri Oct 21 07:13:47 2022 kern.info kernel: [745743.427175] epc = 77de0dc0 in libc.so[77d62000+a9000]
Fri Oct 21 07:13:47 2022 kern.info kernel: [745743.432682] ra  = 77de2b04 in libc.so[77d62000+a9000]
Fri Oct 21 07:13:47 2022 kern.info kernel: [745743.443552] device br-lan.1 left promiscuous mode
Fri Oct 21 07:13:47 2022 kern.info kernel: [745743.448485] device br-lan left promiscuous mode
Fri Oct 21 07:13:52 2022 kern.info kernel: [745748.464316] device br-lan.1 entered promiscuous mode
Fri Oct 21 07:13:52 2022 kern.info kernel: [745748.469774] device br-lan entered promiscuous mode
```

micmac1, 2022-10-21:
I can reproduce this on same target platform (mips24kc). Build log shows some time_t-related warnings.

```
softflowd.c: In function 'format_flow':
softflowd.c:307:13: warning: format '%ld' expects argument of type 'long int', but argument 15 has type 'suseconds_t' {aka 'long long int'} [-Wformat=]
  307 |             "seq:%" PRIu64 " [%s]:%hu <> [%s]:%hu proto:%u "
      |             ^~~~~~~
......
  315 |             (flow->flow_start.tv_usec + 500) / 1000, fin_time,
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                              |
      |                                              suseconds_t {aka long long int}
softflowd.c:309:27: note: format string is defined here
  309 |             "start:%s.%03ld finish:%s.%03ld tcp>:%02x tcp<:%02x "
      |                       ~~~~^
      |                           |
      |                           long int
      |                       %03lld
softflowd.c:307:13: warning: format '%ld' expects argument of type 'long int', but argument 17 has type 'suseconds_t' {aka 'long long int'} [-Wformat=]
  307 |             "seq:%" PRIu64 " [%s]:%hu <> [%s]:%hu proto:%u "
      |             ^~~~~~~
......
  316 |             (flow->flow_last.tv_usec + 500) / 1000, flow->tcp_flags[0],
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                             |
      |                                             suseconds_t {aka long long int}
softflowd.c:309:43: note: format string is defined here
  309 |             "start:%s.%03ld finish:%s.%03ld tcp>:%02x tcp<:%02x "
      |                                       ~~~~^
      |                                           |
      |                                           long int
      |                                       %03lld
softflowd.c: In function 'dump_flows':
softflowd.c:1173:16: warning: format '%ld' expects argument of type 'long int', but argument 4 has type 'time_t' {aka 'long long int'} [-Wformat=]
 1173 |                "EXPIRY EVENT for flow %" PRIu64 " in %ld seconds\n",
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~
 1174 |                expiry->flow->flow_seq, (long int) expiry->expires_at - now);
      |                                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                      |
      |                                                                      time_t {aka long long int}
softflowd.c:1173:56: note: format string is defined here
 1173 |                "EXPIRY EVENT for flow %" PRIu64 " in %ld seconds\n",
      |                                                      ~~^
      |                                                        |
      |                                                        long int
      |                                                      %lld
```

So source code is assuming time_t is 32 bit because target is 32 bit platform, but because of musl in 22.03 always using 64 bit for time_t this isn't correct.

I added a patch for this and recompiled. Then no more segfault.
[030-time_t.txt](https://github.com/openwrt/packages/files/9841006/030-time_t.txt)

Can you try and confirm from your end?

Regards,
Seb

Ansuel, 2022-10-21:
btw i found a very similar bug for the ripe atlas probe... 

the fix is to just cast everything related to time_t to %lld and cast them to (long long) to handle both 32 and 64 bit

HolgerHees, 2022-10-21:
@micmac1 the patch works for me. Thanks a lot. Hope it will be merged soon.

HolgerHees, 2022-10-23:
as this bug will be fixed in another ticket, I will close this one.

closed 2022-10-23 by HolgerHees