openwrt/packages #19671: [libmbim][22.03] package (and its dependants) removed from repo and manifest

syp1975, 2022-10-22:
Maintainer: Nicholas Smith
Environment: 22.03 x86_64

Description:
it seems one of the latests commits has broken libmbim package and its dependants as commented by @ynezz in [5b87373](https://github.com/openwrt/packages/commit/5b87373a39a2a5df25f86ee4563cdd8752312079#commitcomment-87566404)

diff of the snapshot packages manifest (https://downloads.openwrt.org/snapshots/packages/x86_64/packages/Packages.manifest) vs the manifest of the 22.02.3 release (https://downloads.openwrt.org/releases/22.03.2/packages/x86_64/packages/Packages.manifest), you can see that libmbim and all the packages that depend on it are gone:
```
*** "22.03.2 packages.manifest" 2022-10-22 08:12:14.432305511 +0200
--- "snapshot packages.manifest"        2022-10-22 08:11:52.200207926 +0200

...

***************
*** 25649,25668 ****
--- 25928,25965 ----
  Section: libs
  SourceDateEpoch: 1627557662
  Maintainer: Jan Pavlinec <jan.pavlinec1@gmail.com>
  Architecture: x86_64
  Installed-Size: 13440
  Filename: libmaxminddb_1.6.0-1_x86_64.ipk
  Size: 14193
  SHA256sum: 9c42448d4a67f3ce3a4aaab85cc3ba5af5b730aaf7efe3bd70d8361c4db54927
  Description:  The libmaxminddb library provides functions for working MaxMind DB files.
  
+ Package: libmbim
+ Version: 1.26.4-2
+ Depends: libc, glib2
+ Source: feeds/packages/libs/libmbim
+ SourceName: libmbim
+ License: LGPL-2.0-or-later
+ LicenseFiles: COPYING.LIB
+ Section: libs
+ SourceDateEpoch: 1655341666
+ Maintainer: Nicholas Smith <nicholas@nbembedded.com>
+ Architecture: x86_64
+ Installed-Size: 134828
+ Filename: libmbim_1.26.4-2_x86_64.ipk
+ Size: 135415
+ SHA256sum: 1f09c87c850b73fe72e5bc745795f2af2c1f9d370f111110880195338536b363
+ Description:  Helper library to talk to MBIM enabled modems.
+  Add mbim-utils for extra utilities.
+ 
  Package: libmcrypt
  Version: 2.5.8-2
  Depends: libc
  Source: feeds/packages/libs/libmcrypt
  SourceName: libmcrypt
  License: LGPL-2.1-or-later
  LicenseFiles: COPYING.LIB
  Section: libs
  SourceDateEpoch: 1607400704
  CPE-ID: cpe:/a:mcrypt:libmcrypt

...

```

BKPepe, 2022-10-27:
I can see that libmbim packages are compiled now. Builds are getting rebuilt quite often. It's not good. It is not bad, either. This seems to be related to https://github.com/openwrt/packages/issues/19241

kaoet, 2022-12-26:
The `libmbim` package is failing [again](https://downloads.openwrt.org/releases/faillogs-22.03/mips_24kc/packages/libmbim/compile.txt):
```
[10/92] Generating src/libmbim-glib/generated/mbim-atds with a custom command
FAILED: src/libmbim-glib/generated/mbim-atds.c src/libmbim-glib/generated/mbim-atds.h src/libmbim-glib/generated/mbim-atds.sections 
/builder/shared-workdir/build/sdk/build_dir/target-mips_24kc_musl/libmbim-1.26.4/build-aux/mbim-codegen/mbim-codegen --input /builder/shared-workdir/build/sdk/build_dir/target-mips_24kc_musl/libmbim-1.26.4/data/mbim-service-atds.json --output src/libmbim-glib/generated/mbim-atds
Python path configuration:
  PYTHONHOME = (not set)
  PYTHONPATH = (not set)
  program name = 'python3'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = '/builder/shared-workdir/build/sdk/staging_dir/hostpkg/bin/python3'
  sys.base_prefix = '/builder/shared-workdir/build/sdk/staging_dir/hostpkg'
  sys.base_exec_prefix = '/builder/shared-workdir/build/sdk/staging_dir/hostpkg'
  sys.platlibdir = 'lib'
  sys.executable = '/builder/shared-workdir/build/sdk/staging_dir/hostpkg/bin/python3'
  sys.prefix = '/builder/shared-workdir/build/sdk/staging_dir/hostpkg'
  sys.exec_prefix = '/builder/shared-workdir/build/sdk/staging_dir/hostpkg'
  sys.path = [
    '/builder/shared-workdir/build/sdk/staging_dir/hostpkg/lib/python310.zip',
    '/builder/shared-workdir/build/sdk/staging_dir/hostpkg/lib/python3.10',
    '/builder/shared-workdir/build/sdk/staging_dir/hostpkg/lib/python3.10/lib-dynload',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
LookupError: no codec search functions registered: can't find encoding

Current thread 0x00007fec4500f740 (most recent call first):
  <no Python frame>
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:98: /builder/shared-workdir/build/sdk/build_dir/target-mips_24kc_musl/libmbim-1.26.4/.built] Error 1
```

jefferyto, 2023-05-28:
This should be fixed by https://github.com/openwrt/packages/commit/9c5eb22fde6ec8173949c5bc147f61361d0a5d82.

closed 2023-05-28 by jefferyto