openwrt/packages #19687: netdata: please add dbengine option

bob-u, 2022-10-23:
Currently netdata only holds stats for a short period of time.  I tried to compile it with dbengine enabled, however current netdata package source does not include libJudy to be compiled with (--with-bundled-libJudy).

Could you add ability to compile dbengine in netdata package please?  Graphical interface and speed of netdata is much nicer than anything else.  This option would be a welcome addition for openwrt installations on modern hardware, with fast CPUs and USB ports for storage.

David-218, 2023-01-22:
that is an excellent suggestion, i too want netdata w/ dbengine.

WarHawk8080, 2023-10-02:
Same...why isn't that an option...or an addon for install?