openwrt/packages #19703: Adblocks DNS Report needs a reboot to work.

tapper82, 2022-10-25:
Maintainer: @dibdot
Environment:Netgear Nighthawk X4S R7800
Architecture
ARMv7 Processor rev 0 (v7l)
Target Platform
ipq806x/generic
Firmware Version
OpenWrt SNAPSHOT r21100-ffe91ca478 / LuCI Master git-22.297.83017-94dd7fd
Kernel Version
5.15.74 

Description: 
After boot up wate for about 5 10 mins then, log on to luci and click on adblock, DNS Repor there will be know data. If you restart the adblock Process then go back to DNS Report you will get data. This is the same no matter how long you wate. Adblock needs a restart for you to get any output from DNS Report. I have tcpbump installed and all so tryed it out with tcpdump-mini and same behaviour.

config adblock 'global'
	option adb_enabled '1'
	option adb_debug '0'
	option adb_safesearch '0'
	option adb_dnsfilereset '0'
	option adb_mail '0'
	option adb_backup '1'
	option adb_dns 'dnsmasq'
	option adb_maxqueue '16'
	option adb_forcedns '1'
	option adb_report '1'
	option adb_dnsflush '1'
	option adb_repiface 'br-lan'
	option adb_fetchutil 'curl'
	list adb_sources 'adaway'
	list adb_sources 'adguard'
	list adb_sources 'adguard_tracking'
	list adb_sources 'anti_ad'
	list adb_sources 'bitcoin'
	list adb_sources 'disconnect'
	list adb_sources 'easylist'
	list adb_sources 'phishing_army'
	list adb_sources 'spam404'
	list adb_sources 'stopforumspam'
	list adb_sources 'whocares'
	list adb_sources 'yoyo'

dibdot, 2022-10-25:
Not reproducible - just hit the "Refresh"-Button on the DNS report page.

tapper82, 2022-10-25:
Hi sorry I should of said that hitting the "Refresh"-Button does not work until I restart the adblock service.

dibdot, 2022-10-25:
Still not reproducible. I see that you don't have a trigger interface defined, the report engine needs at öeast one initial run after boot. Define a trigger interface and check that works...

closed 2025-06-01 by dibdot