openwrt/packages #19718: php8-mod-iconv - iconv_strlen() throws converter error

brackenhill-mob, 2022-10-26:
Maintainer: @mhei
Environment: x86, x86_64, OpenWrt 22.03.1

Description:
Calling iconv_strlen() throws this error
```Warning: iconv_strlen(): Cannot open converter in <source code file>```

I can create a one line php file which does this every time.

I've tried installing the iconv package and libraries and it made no difference.

I'm trying to move Invoice Ninja, a brilliant invoicing program, from our VPS to OpenWrt and it makes calls to this function when creating the PDF invoice to send to a client so it's kind of important for us ;)

brackenhill-mob, 2022-10-26:
```
<?php
    echo "This should return 4 -> " . iconv_strlen("test");
?>
```

mhei, 2022-10-28:
You use the precompiled binaries or do you build/compile on your own?

brackenhill-mob, 2022-10-28:
Pre-compiled, either by installing through LUCI or image builder. Result is the same.

To help debug this, I'm happy to try and build from source if you point me in the right direction of where to find the right source etc.

mhei, 2022-11-04:
Sorry for the late reply, unfortunately I don't have much free time at the moment.
Could you try latest snapshots instead of 22.03? I guess that this will only work in 22.03 when you compile yourself with `CONFIG_BUILD_NLS=y`, i.e. "Compile with full language support" enabled.

brackenhill-mob, 2022-11-04:
I was trying the latest snapshot earlier this afternoon before seeing your message.

It appears to be working ;) Something must have changed in the last week as the snapshot build I last tried failed. I tried again because I built the complete image from source (>4 hours!) and that worked, so hoped a new snapshot might work also.

This can be closed I think @mhei

mhei, 2022-11-04:
Yes, indeed something changed. In OpenWrt base system, the
following changes were applied:
92f0b7d4d25c8493270051298a4073f5b630bda3
b37ef4b8abf2769732078ba510bf75abc83f007e
c0ba4201f8372eee579584b7f5900e073b207b0c

And for the PHP package, we have a24a76783ea83225ebe242306fd3475bff34698c.

If it is ok for you to use snapshots, then closing is fine with me.

brackenhill-mob, 2022-11-05:
@mhei I'm comfortable using snapshots so I'll close this

closed 2022-11-05 by brackenhill-mob