openwrt/packages #19720: softethervpn5-client: general protection fault

ahasbini, 2022-10-26:
Maintainer: @Andy2244 
Environment: VirtualBox x86_64, OpenWrt 21.02.0 r16279-5cc0535800

Description:
After successfully configuring `softethervpn5-client` using `vpncmd`, I try to connect it using `AccountConnect <name>`, and let OpenWrt to get IP using DHCP, I achieve network connectivity  (`ping -I vpn_wand 8.8.8.8` works), but after a minute the `general protection fault` appears in `logread` and connectivity is lost suddenly.
All installed packages are up-to-date.
I followed most of the steps mentioned here: [SoftEther VPN Client](https://openwrt.org/docs/guide-user/services/vpn/softethervpn/client)

`vpncmd` commands (<param> are filled parameters):
```
NicCreate wand
AccountCreate test /SERVER:<hostname:port> /HUB:<hubname> /USERNAME:<username> /NICNAME:wand
AccountAnonymousSet test
AccountConnect test
```
Snippet of `/etc/config/dhcp`:
```
config dhcp 'wand'
        option interface 'wand'
        option ignore '1'
        list ra_flags 'none'
```
Snippet of `/etc/config/network`:
```
config interface 'wand'
        option proto 'dhcp'
        option device 'vpn_wand'
        option hostname '*'
        option metric '30'
```
Snippet of `/etc/config/firewall`:
```
config zone
        option name 'wan'
        option input 'REJECT'
        option output 'ACCEPT'
        option forward 'REJECT'
        option masq '1'
        option mtu_fix '1'
        list network 'wan'
        list network 'wan6'
        list network 'wand'
```
Packages:
```
root@OpenWrt:~# opkg list-installed | grep soft
luci-app-softether - git-21.011.72214-ec5ce98
softethervpn5-client - 5.02.5180-1
softethervpn5-libs - 5.02.5180-1
```
Logs:
```
Wed Oct 26 19:03:13 2022 daemon.notice netifd: Interface 'wand' is now up
Wed Oct 26 19:03:13 2022 daemon.info dnsmasq[7790]: reading /tmp/resolv.conf.d/resolv.conf.auto
Wed Oct 26 19:03:13 2022 daemon.info dnsmasq[7790]: using only locally-known addresses for domain test
Wed Oct 26 19:03:13 2022 daemon.info dnsmasq[7790]: using only locally-known addresses for domain onion
Wed Oct 26 19:03:13 2022 daemon.info dnsmasq[7790]: using only locally-known addresses for domain localhost
Wed Oct 26 19:03:13 2022 daemon.info dnsmasq[7790]: using only locally-known addresses for domain local
Wed Oct 26 19:03:13 2022 daemon.info dnsmasq[7790]: using only locally-known addresses for domain invalid
Wed Oct 26 19:03:13 2022 daemon.info dnsmasq[7790]: using only locally-known addresses for domain bind
Wed Oct 26 19:03:13 2022 daemon.info dnsmasq[7790]: using only locally-known addresses for domain lan
Wed Oct 26 19:03:13 2022 daemon.info dnsmasq[7790]: using nameserver 8.8.8.8#53
Wed Oct 26 19:03:13 2022 daemon.info dnsmasq[7790]: using nameserver 10.211.254.254#53
Wed Oct 26 19:03:13 2022 daemon.info dnsmasq[7790]: using nameserver 8.8.8.8#53
Wed Oct 26 19:03:13 2022 user.notice firewall: Reloading firewall due to ifup of wand (vpn_wand)
Wed Oct 26 19:04:13 2022 kern.info kernel: [  125.445977] traps: vpnclient[7966] general protection fault ip:7f6903967c0c sp:7f6903093700 error:0 in libc.so[7f6903965000+49000]
```

ahasbini, 2022-10-26:
In case this helps, I have uninstalled `softethervpn5*` packages and installed `softethervpn*` (4.34-9745-1), have applied the same steps to setup vpn connection and connected, so far did not face the same issue as in first post.

tew42, 2023-05-04:
I initially thought you were on 22.x, where I'm also having issues, but in 21.x it has always worked well for me.

1) I assume you also have a network entry for `config device option name 'vpn_wand'`?

2) I only ever got mine working with an interface entry like
```
	option proto 'dhcp'
	option device 'vpn_xxx'
	option defaultroute '0'
	option delegate '0'
```
plus static routes for some IP ranges