openwrt/packages #19726: kmod-ipt-nathelper-rtsp

deejayexe, 2022-10-28:
Maintainer: @Jo-Philipp Wich
Environment: Turris Omnia Based on the last OpenWrt 21.02.5 release
Description:
This package is only enabled for the version  OpenWrt-21.02.0, we would need in the community of Turris omnia that the package is enabled for the version OpenWrt 21.02.5 release that is the current stable version on these devices, to be able to use the control in the iptv

vcunat, 2022-10-28:
@jow- would be the way to mention, I believe.

jow-, 2022-10-28:
The package is there: https://downloads.openwrt.org/releases/21.02.5/targets/mvebu/cortexa9/kmods/5.4.215-1-e24fc183201cdcd9c5405121e9082e61/kmod-ipt-nathelper-rtsp_5.4.215+3.13-4_arm_cortex-a9_vfpv3-d16.ipk

closed 2022-10-28 by jow-