openwrt/packages #19738: atlas-sw-probe : ssh -o "ServerAliveInterval 60" causes an immediate "Connection to PEER port 22 timed out"

KsmoinO, 2022-10-30:
Maintainer: Jan Pavlinec
Environment: ath79, dlink_dir-842-c3, 22.03.2

Description:

In functions ssh and ssh_exec from /usr/libexec/atlas-probe-scripts/bin/arch/linux/linux-functions.sh options `-o "ServerAliveInterval 60"` cause an immediate `Connection to "PEER" port 22 timed out` with ssh from openssh-client (8.9p1-1)

Workaround (or solution) should be to remove this options or implement it on openssh-client (8.9p1-1)

Ansuel, 2022-11-02:
@KsmoinO can you suggest how to repro this? i'm testing this on openssh-client 9.0 but afaik this option is not new.

If it's not present in 8.9, we can consider bumping openssh-client on 22.03

KsmoinO, 2022-11-02:
I tested it like that:
```
root@OpenWrt:~# ssh -V
OpenSSH_8.9p1, OpenSSL 1.1.1q  5 Jul 2022
root@OpenWrt:~# time /usr/bin/ssh  -i /usr/libexec/atlas-probe-scripts/etc/probe_key -o "ServerAliveInterval 60" -o "UserKnownHostsFile /usr/libexec/atlas-probe-scripts/.ssh/known_hosts" -p 443 atlas@ctr-fsn01.atlas.ripe.net
Connection to 2a01:4f8:110:40b2::2 port 443 timed out
Command exited with non-zero status 255
real    0m 0.08s
user    0m 0.03s
sys     0m 0.00s
root@OpenWrt:~# time /usr/bin/ssh  -i /usr/libexec/atlas-probe-scripts/etc/probe_key -o "UserKnownHostsFile /usr/libexec/atlas-probe-scripts/.ssh/known_hosts" -p 443 atlas@ctr-fsn01.atlas.ripe.net
PTY allocation request failed on channel 0
OK

Connection to ctr-fsn01.atlas.ripe.net closed.
real    0m 1.92s
user    0m 0.44s
sys     0m 0.06s
```
I don't know how to install openssh-client 9.0 to test with it

BKPepe, 2022-11-03:
OpenWrt 21.02.5 here, but I don't get timed out, but I do get ``atlas@ctr-fsn01.atlas.ripe.net: Permission denied (publickey)`` with ``OpenSSH_8.4p1, OpenSSL 1.1.1q  5 Jul 2022``, but I do have only IPv4 connection. I will need to retest this with IPv6.

KsmoinO, 2022-11-03:
I also have the issue using IPv4:
```
root@OpenWrt:~# time /usr/bin/ssh -4 -i /usr/libexec/atlas-probe-scripts/etc/probe_key -o "ServerAliveInterval 60" -o "UserKnownHostsFile /usr/libexec/atlas-probe-scripts/.ssh/known_hosts" -p 443 atlas@ctr-fsn01.atlas.ripe.net
Connection to 178.63.8.31 port 443 timed out
Command exited with non-zero status 255
real    0m 0.25s
user    0m 0.02s
sys     0m 0.01s
```

Mohesity2, 2022-11-29:
I've encountered the same issue.

I upgraded my openwrt box from 21.02 to 22.03.2

``` 
root@openwrt:~# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.2'
DISTRIB_REVISION='r19803-9a599fee93'
DISTRIB_TARGET='ath79/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt 22.03.2 r19803-9a599fee93'
DISTRIB_TAINTS=''
root@openwrt:~# ssh -V
OpenSSH_8.9p1, OpenSSL 1.1.1s  1 Nov 2022
root@openwrt:~# /usr/bin/ssh -4 -i /usr/libexec/atlas-probe-scripts/etc/probe_key -o "ServerAliveInterval 60" -o "UserKnownHostsFile /usr/libexec/atlas-probe-scripts/.ssh/known_hosts" -p 443 atlas@ctr-fsn01.atlas.ripe.net
Warning: Identity file /usr/libexec/atlas-probe-scripts/etc/probe_key not accessible: No such file or directory.
Connection to 178.63.8.31 port 443 timed out

``` 

After removing "ServerAliveInterval 60", it works well.

``` 
root@openwrt:~# /usr/bin/ssh -4 -i /usr/libexec/atlas-probe-scripts/etc/probe_key  -o "UserKnownHostsFile /usr/
libexec/atlas-probe-scripts/.ssh/known_hosts" -p 443 atlas@ctr-fsn01.atlas.ripe.net
Warning: Identity file /usr/libexec/atlas-probe-scripts/etc/probe_key not accessible: No such file or directory.
The authenticity of host '[ctr-fsn01.atlas.ripe.net]:443 ([178.63.8.31]:443)' can't be established.
RSA key fingerprint is SHA256:eAAZuOzEsuxYzPVY7zUSeRdSGWxJHCXnq5LzaUGWiUY.
This key is not known by any other names
Are you sure you want to continue connecting (yes/no/[fingerprint])? 
``` 

My Ubuntu box also works well.

``` 
zbox[15] ssh -V
OpenSSH_8.9p1 Ubuntu-3, OpenSSL 3.0.2 15 Mar 2022
zbox[16] /usr/bin/ssh -4 -i /usr/libexec/atlas-probe-scripts/etc/probe_key -o "ServerAliveInterval 60" -o "UserKnownHostsFile /usr/libexec/atlas-probe-scripts/.ssh/known_hosts" -p 443 atlas@ctr-fsn01.atlas.ripe.net
Warning: Identity file /usr/libexec/atlas-probe-scripts/etc/probe_key not accessible: No such file or directory.
The authenticity of host '[ctr-fsn01.atlas.ripe.net]:443 ([178.63.8.31]:443)' can't be established.
RSA key fingerprint is SHA256:eAAZuOzEsuxYzPVY7zUSeRdSGWxJHCXnq5LzaUGWiUY.
This key is not known by any other names
Are you sure you want to continue connecting (yes/no/[fingerprint])? 
``` 

Is there any difference between v 22.03 and 21.02?  v21.02 has not shown such a "timed out" message even with "ServerAliveInterval 60" .

KsmoinO, 2022-12-02:
Should I open a related issue to openssh-client package if it works in 9.0 to ask for bumping it on 22.03 ?

Ansuel, 2022-12-02:
We should check if 9.0 on 22 fix it... It's probably a stupid bug with not
correctly checking the timeout.

SibrenVasse, 2023-02-08:
I have built openssh9.2p1 for 22.03.3, you can try installing the packages from here:
https://openwrt.sibrenvasse.nl/openssh-22.03/mips_24kc/

Be sure to upgrade all packages!
```
# opkg list-installed | grep openssh
openssh-client - 9.2p1-1
openssh-keygen - 9.2p1-1
openssh-server - 9.2p1-1
openssh-sftp-server - 9.2p1-1
```

KsmoinO, 2023-07-20:
it seems fixed in OpenWrt 22.03.5
```
root@OpenWrt:~# ssh -V
OpenSSH_8.9p1, OpenSSL 1.1.1u  30 May 2023
root@OpenWrt:~# time /usr/bin/ssh -4 -i /usr/libexec/atlas-probe-scripts/etc/probe_key -o "ServerAliveInterval 60" -o "UserKnownHostsFile /usr/libexec/atlas-probe-scripts/.ssh/known_hosts" -p 443 atlas@ctr-fsn01.atlas.ripe.net
PTY allocation request failed on channel 0
OK

Connection to ctr-fsn01.atlas.ripe.net closed.
real    0m 2.07s
user    0m 0.44s
sys     0m 0.05s
```

closed 2023-07-20 by KsmoinO