openwrt/packages #19739: atlas-probe: "do_page_fault(): sending SIGSEGV to eperd for invalid read access from" occurs evrey 45 minutes

KsmoinO, 2022-10-30:
Package version: 5040-1
Maintainer: @ja-pa 
Environment: ath79, dlink_dir-842-c3, 22.03.2

Description:
About every 45 minutes, around the 40th minute of each hour, eperd is killed with following kernel message `do_page_fault(): sending SIGSEGV to eperd for invalid read access from` followed by hexadecimal number (may be a memory address)
Restarting atlas service causes eperd to be killed again in less than 5 minutes twice then runs for 45 before be killed again

One a day, perd is also killed with the same message

Ansuel, 2022-10-30:
this has already been fixed but we need to backport it

BKPepe, 2022-10-30:
I already backported it to the stable branches when I merged your PR. 

Reference: https://github.com/openwrt/packages/pull/19704

closed 2022-10-30 by BKPepe

KsmoinO, 2022-10-31:
Fixed in 2.4.1-2