openwrt/packages #19740: atlas-probe 2.4.1-1/atlas-sw-probe 5040-1: path /home/atlas used in busybox and scripts but does not exists

KsmoinO, 2022-10-30:
Maintainer: @ja-pa 
Environment: ath79, dlink_dir-842-c3, 22.03.2

Description:

/usr/libexec/atlas-probe/bin/busybox, /usr/libexec/atlas-probe-scripts/bin/ATLAS and /usr/libexec/atlas-probe-scripts/bin/common-pre.sh use /home/atlas but it does not exists

A workaround could be to create a symlink (/home/atlas -> /var/ripe_atlas_probe/)

Remark : Is the atlas user (and group) are necessary?

Ansuel, 2022-10-30:
internally that path is replaced with the correct one....

KsmoinO, 2022-10-30:
not in /usr/libexec/atlas-probe-scripts/bin/ATLAS lines 138, 142 (and maybe 38)

Ansuel, 2022-10-30:
again from what i notice they modified cron and their custom eperd to replace that path with the base dir and stuff is correctly passed and sent with no error

KsmoinO, 2022-10-31:
Oh, sorry. You're right, removing the symlink does not seems to impact stuff

closed 2022-10-31 by KsmoinO