openwrt/packages #19741: v2ray-core: config.json loaded twice

hnyyghk, 2022-10-30:
https://github.com/openwrt/packages/blob/646fbeafdc6ad4b39217f767216c8fac27776151/net/v2ray-core/files/v2ray.init#L29

config v2ray 'config'
	option confdir '/etc/v2ray'
	list conffiles '/etc/v2ray/config.json'

config.json will be loaded twice

we should add a check, in case of conffiles under confdir

1715173329, 2022-10-31:
It's just a sample config file anyway, so you need adjust it based on your own requirements.

hnyyghk, 2022-10-31:
https://github.com/hnyyghk/packages/commit/e5dd4e0f2011aecedec308d8ad69ef8dbd2aada6

i add some check to avoid this case
and tested with following config:

/etc/config/v2ray
`
config v2ray 'enabled'
	option enabled '1'

config v2ray 'config'
	option confdir '/etc/v2ray'
	list conffiles '/etc/v2ray/config.json'
	option datadir '/usr/share/v2ray'
	option format 'json'
	option memconservative '1'

`

can i pr it?

1715173329, 2022-10-31:
tbh it looks really bad...

hnyyghk, 2022-10-31:
hah, at the beginning, i want to realize like this
```
if [[ -z "$confdir" || ! "${FILE_ARR[@]}" =~ $i ]]; then
	procd_append_param command -config "$i"
fi
```

but ${FILE_ARR[@]} and =~ can't work in ash
that is the only way i can found

do you have any ideas?

1715173329, 2022-10-31:
If you're really interested in this sample config, just leave `confdir` empty by default, or make it conflict with `conffiles`.
users can fill it with whatever strange stuff and it's really hard to validate, just keep it simple and stupid

hnyyghk, 2022-10-31:
ok, i got it
thank you for your prompt reply

closed 2022-10-31 by hnyyghk