openwrt/packages #19755: unbound: add UCI option to serve expired cache entries

ahuse, 2022-10-31:
Maintainer: @EricLuehrsen

Hi,

i have done some reading on how to optimize unbound for small environments (only a few clients, 2-3 people issuing DNS-requests). If option `recursion` in UCI is set to `aggressive`, prefetching is enabled for unbound. In my opinion enabling just prefetching is not feasible for small environments. A lot of entries are just not prefetched, because they are not requested in the last 10 % of the TLL (especially for very small TLLs).

Adding the following option to serve expired entries could be a good solution to solve this "problem" for small environments. [Here](https://unbound.docs.nlnetlabs.nl/en/latest/topics/core/serve-stale.html) is the corresponding article in the unbound-docs.

```
server:
	serve-expired: yes
	serve-expired-ttl: 86400
```

Maybe adding this to the UCI option `aggressive` or creating a new UCI option would be a good idea. I know this could be counterproductive in case of heavy DNS load balancing on the site, but i had no problems regarding this so far. The option `serve-expired-ttl` can be set to up to three days.

For those how want resilience instead of performance the additional option `serve-expired-client-timeout` is interesting as well.

What do you think?

EricLuehrsen, 2022-11-12:
Serving expired is not a good idea. When multisite hosts use DNS as a "poor mans" load balance, then clients can be left with dead ends.

ahuse, 2022-11-13:
Do you mean multihost sites? Is this really a problem? As far as my research goes, DNS Load Balancing is only done by rotating the entries and not by removing entries. So the expired record is maybe pointing to a server which the DNS Load Balancing is not preferring at the moment. But as soon as the expired record is served (with a TLL of 30), unbound updates the cache (if prefetching is enabled).

As i read through multiple forums (pfsense, opnsense, pi-hole, etc.), this topic seems to be a question of faith. I used this setting for multiple weeks now without problems and increased cache hits and prefetching by a lot. Some people reduce `serve-expired-ttl` to lower values (like `7200`).

closed 2023-04-26 by ahuse