openwrt/packages #19794: ModemManager: broken in 22.03 and netifd reconnect issue

lynxthecat, 2022-11-03:
**Firstly**, ModemManager is broken in 22.03 because modem disconnect does not reach through to netifd absent this patch from @aleksander0m: https://github.com/openwrt/packages/commit/bc754f31cfdb004eefa43038f8f0827922107fc6

In short, ModemManager sees the disconnect, but netifd does not and the output of e.g. 'ifstatus' or 'ip addr show dev' is wrong (reflecting the old connected state and not updated to reflect disconnection). See e.g. here: https://github.com/openwrt/openwrt/issues/8368#issuecomment-1299719052.

So ModemManager in 22.03 needs to be updated to fix this. 

**Secondly**, even with the above patch in the ModemManager version on OpenWrt master, upon disconnection we see:

```
Wed Nov  2 20:38:54 2022 daemon.info [2716]: <info>  [modem0] state changed (connected -> disconnecting)
Wed Nov  2 20:38:54 2022 daemon.info [2716]: <info>  [modem0] state changed (disconnecting -> registered)
Wed Nov  2 20:38:54 2022 daemon.info [2716]: <info>  [modem0/bearer3] connection #1 finished: duration 121s, tx: 258333 bytes, rx: 371928 bytes
Wed Nov  2 20:38:54 2022 user.notice modemmanager: interface wan (network device wwan0) disconnected
Wed Nov  2 20:38:54 2022 daemon.notice netifd: Interface 'wan' has lost the connection
Wed Nov  2 20:38:54 2022 daemon.notice netifd: Network device 'wwan0' link is down
Wed Nov  2 20:38:54 2022 daemon.warn dnsmasq[1]: no servers found in /tmp/resolv.conf.d/resolv.conf.auto, will retry
```
And so although now with the patched version of ModemManager netifd sees the disconnect, netifd does not actually reconnect. 

This is using /etc/config/network:

```
config interface 'wan'
        option proto 'modemmanager'
        option device '/sys/devices/platform/1e1c0000.xhci/usb2/2-1'
        option auth 'pap'
        option iptype 'ipv4v6'
        option apn 'xx
        option username 'yy'
        option password 'zz'
```

desolated40, 2022-11-04:
I am also facing this issue. Currently using ModemManager 1.18.12-5 in OpenWRT 21.02.

ModemManager netifd notifies that the interface is down but netifd does not reconnect ModemManager by itself. 

I am able to reproduce this problem consistently as I have a private LTE network in my lab where my device with OpenWRT installed is connected to.

lynxthecat, 2022-11-04:
I just had a big chat with @jow- about this on #openwrt-devel and it turns out that the assumption is wrong that netifd should reconnect. Actually ModemManager should reconnect - see this post here:

https://forum.openwrt.org/t/rfc-wwan-uqmi-revamp/141506/16?u=lynx

desolated40, 2022-11-04:
@lynxthecat Thanks for the info! I'm currently using basic serial (PPP) but looking to implement ModemManager as this is the only protocol right now supported by OpenWisp.

aleksander0m, 2022-11-04:
> We need fix to ModemManager to address this, namely ModemManager should:
> 
>     inform netifd as it presently does in patched version using the report-down script
>     reconnect
>     inform netifd with new report-up script of the new connection details

MM is not in charge of any reconnection logic, MM does not need to be fixed here. MM only receives requests to connect/disconnect from upper layers (netifd) and monitors the ongoing connection and notifies to upper layers (netifd) of network-initiated disconnections. It should be netifd the one managing the connection, so the autoreconnection logic should be handled by netifd.

The way to solve this, after briefly talking with @jow- about this in IRC would be to have the netifd protocol handler launch a "watcher" process which brings up the connection, and is kept alive and running for as long as the network interface is assumed connected. If MM detects a network-initiated disconnection, the dispatcher script called by MM should kill that watcher process. At that point, netifd (if configured to autoconnect) will then kill the netdev, restart that process and await proto updates.

Anyone up to the task of writing this logic in the netfid modemmanager protocol handler?

lynxthecat, 2022-11-04:
I thought the whole point of ModemManager was daemon to manage connections and keep it alive. If not what's the point?

aleksander0m, 2022-11-04:
> I thought the whole point of ModemManager was that it's a daemon to manage connections and keep them alive. If not what's the point

MM provides a unified API to control modems, so that the user does not need to care of which is the underlying control protocol the modem uses. MM doesn't even configure the network interfaces, for what it's worth. MM only talks to the modem control port.

lynxthecat, 2022-11-04:
@desolated40 seems like a temporary fix is just to add the line 'ifup wan' to the new report-down script at /usr/lib/ModemManager/connection.d/10-report-down. Works for me in terms of immediately reconnecting post disconnect.

desolated40, 2022-11-08:
Thanks @lynxthecat this indeed works like you are describing in the happy flow. However there are cases where the LTE network is not in a ready state to connect to directly after a disconnection. 

I have tried to modify the script but I don't have the knowhow of modifying this script so netifd will keep trying to reconnect and stops when the connection has been restored.

lynxthecat, 2022-11-08:
You could try just adding 'sleep 2' right before this line:

`[ "${IFUP}" = "true" ] && ifup ${CFG}`

... but this is just a hack since the OpenWrt ModemManager implementation in OpenWrt is broken right now.

nemesisdesign, 2023-03-01:
> I am also facing this issue. Currently using ModemManager 1.18.12-5 in OpenWRT 21.02.
> 
> ModemManager netifd notifies that the interface is down but netifd does not reconnect ModemManager by itself.
> 
> @aleksander0m I am able to reproduce this problem consistently as I have a private LTE network in my lab where my device with OpenWRT installed is connected to.

@desolated40 this PR backported a number of fixes to OpenWrt 21.02.

lynxthecat, 2023-03-01:
Which PR? As far as I'm aware the broken behaviour described in this issue still remains in 22.03. In fact I don't think even the new modem manager with the dispatcher script functionality is even in 22.03. So LTE reconnection w.r.t ModemManager is borked.

lynxthecat, 2023-04-11:
Please, please can ModemManager be bumped to at least 1.18.12 for 22.03 so that we can at least users can use my temporary workaround for reconnects?

nemesisdesign, 2023-04-18:
@lynxthecat I was talking about https://github.com/openwrt/packages/pull/19648.

lynxthecat, 2023-04-18:
@nemesisdesign who should we ping to try to move the needle on this one?

somu1795, 2023-06-03:
One more major issue is the package umbim does not support proxy mode which is required by lot of lte/5G devices.

There's a commit in the umbim repo , however that feature does not exist in openwrt 22.03.5 

https://git.openwrt.org/?p=project/umbim.git;a=commit;h=ff8d35615153086f4b89443e511907f10ff059de

aleksander0m, 2023-06-05:
Has someone tried to work on the watcher process logic as suggested in https://github.com/openwrt/packages/issues/19794#issuecomment-1303601631?

suyuan168, 2023-11-18:
A year later, in the latest openwrt 23.05 and main line tasks.
There is still no change in the Modemmanager, and I cannot reconnect myself after disconnecting the link.
You must manually click Reconnect on the network port interface to work properly.
Is anyone willing to think of a solution?

lynxthecat, 2023-12-03:
@feckert might you be interested in trying out implementation for: 

https://github.com/openwrt/packages/issues/19794#issuecomment-1576338502

or other technique for ensuring that the netifd protocol for modemmanager actually brings the interface back up on disconnect rather than simply doing nothing and leaving the interface down, as per comments above?

maskimthedog, 2024-01-08:
> @desolated40 it seems to me like a reasonable temporary fix is to replace the existing '10-report-down' script with something like the following:
> 
> ```
> root@OpenWrt:~# cat /usr/lib/ModemManager/connection.d/10-report-down-and-reconnect
> #!/bin/sh
> 
> # Automatically report to netifd that the underlying modem
> # is really disconnected and reconnect if interface was up
> 
> # require program name and at least 4 arguments
> [ $# -lt 4 ] && exit 1
> 
> MODEM_PATH="$1"
> BEARER_PATH="$2"
> INTERFACE="$3"
> STATE="$4"
> 
> [ "${STATE}" = "disconnected" ] || exit 0
> 
> . /usr/share/ModemManager/modemmanager.common
> . /lib/netifd/netifd-proto.sh
> INCLUDE_ONLY=1 . /lib/netifd/proto/modemmanager.sh
> 
> MODEM_STATUS=$(mmcli --modem="${MODEM_PATH}" --output-keyvalue)
> [ -n "${MODEM_STATUS}" ] || exit 1
> 
> MODEM_DEVICE=$(modemmanager_get_field "${MODEM_STATUS}" "modem.generic.device")
> [ -n "${MODEM_DEVICE}" ] || exit 2
> 
> CFG=$(mm_get_modem_config "${MODEM_DEVICE}")
> [ -n "${CFG}" ] || exit 3
> 
> IFUP=$(ifstatus ${CFG} | jsonfilter -e '@.up')
> 
> logger -t "modemmanager" "interface ${CFG} (network device ${INTERFACE}) ${STATE}"
> proto_init_update $INTERFACE 0
> proto_send_update $CFG
> 
> [ "${IFUP}" = "true" ] && ifup ${CFG}
> 
> exit 0
> ```
> 
> Works for me in terms of immediately reconnecting post disconnect.

This does not work for me.

lynxthecat, 2024-01-08:
@maskimthedog you probably ought to elaborate on this if you can. What do you see in the system log?

maskimthedog, 2024-01-08:
We are using a MT7688 with BG96 over QMI, OpenWRT with Modem Manager 1.20.6 with the 10-report-down-and-reconnect
script.  Below quote is from my colleague performing signal degradation and modem recovery tests.

"when I add too much attenuation or disconnect the antenna completely, modem losts the connections
after installing back the antenna, modem quite quickly backs to good CSQ and state "registered", but openWrt cannot recover internet connection after that.. watchcat need to trigger restart (unless I send "ifup wan") "

maskimthedog, 2024-01-08:
In what logs are you interested @lynxthecat?

maskimthedog, 2024-01-08:
@lynxthecat my colleague is repeating his test to get one of our routers in to this state and dumping the log.  Will provide this afternoon.  Watchcat rebooted the router on the last iteration...we disabled it for this one.

lynxthecat, 2024-01-08:
Good. This is a hack (hence the outstanding issue), but it is certainly preferable (and should replace, not complement) watchcat, which is a much worse hack in my opinion.

maskimthedog, 2024-01-08:
@lynxthecat here you go.

[d66b_registered_log.txt](https://github.com/openwrt/packages/files/13864545/d66b_registered_log.txt)
[log (2).txt](https://github.com/openwrt/packages/files/13864547/log.2.txt)

maskimthedog, 2024-01-08:
Looks like mwan3 could be interfering?

maskimthedog, 2024-01-10:
@lynxthecat We have a script that runs as a service every 30 seconds to cache modem properties for query via a RESTful web server.  I put a hack in that script to test to see if modem manager was in state "disabled" or "registered" and issue and ifup for wan.  It works very well.

lynxthecat, 2024-01-10:
Ah glad to hear that (albeit it’s a form of polling). But the issue with OpenWrt remains. I hope one day some attention might be given to this by the right person(s).

maskimthedog, 2024-01-10:
We have other gateway products on which we employ Yocto with Network Manager/Modem Manager and do not see this issue.

yegorich, 2024-01-11:
Any idea how to provoke "disconnect" and `10-report-down` invocation without touching the antennas?

lynxthecat, 2024-01-11:
`mmcli -m any —simple-disconnect`

feckert, 2024-01-15:
A colleague of mine has staged a feature pull request [here](https://github.com/openwrt/packages/compare/master...osedlbauer:packages-fork:pr/240111-Add-mm-watcher) that might solve the problem. If someone has time and a test setup can some one please check it and report back if it worked.

maskimthedog, 2024-01-15:
> A colleague of mine has staged a feature pull request [here](https://github.com/openwrt/packages/compare/master...osedlbauer:packages-fork:pr/240111-Add-mm-watcher) that might solve the problem. If someone has time and a test setup can some one please check it and report back if it worked.

We can test in our RF screen room adjusting the signal level and see what happens.

maskimthedog, 2024-01-15:
> We can test in our RF screen room adjusting the signal level and see what happens.

The "workaround" we came up with watches for states "disabled" and "registered"...the states in which we see Modem Manager get "stuck" and issue an 'ifup wan' to remedy.

osedlbauer, 2024-01-16:
I had to make some minor adjustments and have now opened a PR. I tested the change with my setup. It works with simple-disconnect and antenna disconnect. Could someone else please check as well? (PR: #23111)

osedlbauer, 2024-01-17:
I made further changes because I noticed that the PPP method behaves differently. pppd already handles this, so it's neither necessary nor possible to start a second separate watcher here. For the other methods (static and DHCP), a watcher is now started when setting the 'keep_connected' option.
From my perspective, everything should be in order now.

lynxthecat, 2024-01-17:
@osedlbauer I'm keen to test, but I'm under a lot of time pressure in my work at the moment and my limited familiarity with compiling packages would result in it taking too long for me to compile my own from your fork. Any chance that you could compile a ModemManager package for my Zyxel NR7101 on 23.05 for me to test out?

feckert, 2024-02-12:
I have taken ModemManager changes by @dangowrt and opened a [merge request](https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/merge_requests/1131) to the Modemmanager upstream repository .

feckert, 2024-04-08:
This is not needed anymore, because the related pullrequest will not get merged. This was solved diffrently without a config option.
https://github.com/openwrt/packages/pull/23590/commits/77c16abb1a3021bba0db97cb4349e906c9bb6ecb

peterpt, 2025-07-27:
I had the same issue with a quetel EC25 modem card , the wwan0 after sometime was falling down due to most likely signal loss and i asked gemini pro 2.5 to make a script that works on startup of openwrt that checks if modem is present , in case not then uses sysfs to authorize the modem to make it active , then sets up the network connection up , also in case system is running is pinging google.com every 60 seconds in case it fails then it will check if i still have an ip address on wwan0 in case not then it will restart the network interface , then have a safe measure that if somehow modem disapears from lsusb then it will reboot the router automatically , here it is the script in case anyone needs it , if you are going to use it then make sure the vendorid and productid matches your modem in lsusb .

`#!/bin/sh

# --- Configuration ---
PING_HOST="google.com"
PING_COUNT=3
INTERFACE="wwan0"
SLEEP_INTERVAL=60
LOG_TAG="ConnMon"

# --- Advanced Failure Handling Configuration ---
MAX_HARD_RESET_FAILS=3
HARD_RESET_VERIFY_TIMEOUT=120

# --- Hardware Configuration ---
MODEM_VENDOR_ID="2c7c"
MODEM_PRODUCT_ID="0125"

# --- Global State Variable ---
HARD_RESET_FAIL_COUNT=0

# --- Main Script ---

logger -t "$LOG_TAG" "Connection monitor started. Patient mode enabled. Max failures before reboot: $MAX_HARD_RESET_FAILS."

# --- Helper Functions (these remain unchanged) ---

is_modem_present() {
    for dir in /sys/bus/usb/devices/*; do
        if [ -f "$dir/idVendor" ] && [ -f "$dir/idProduct" ]; then
            if [ "$(cat "$dir/idVendor")" = "$MODEM_VENDOR_ID" ] && [ "$(cat "$dir/idProduct")" = "$MODEM_PRODUCT_ID" ]; then
                return 0
            fi
        fi
    done
    return 1
}

perform_hard_reset_and_verify() {
    logger -t "$LOG_TAG" -p daemon.warn "Attempting HARD hardware reset of the modem via sysfs..."
    local device_dir
    device_dir=$(is_modem_present && for dir in /sys/bus/usb/devices/*; do if [ -f "$dir/idVendor" ] && [ "$(cat "$dir/idVendor")" = "$MODEM_VENDOR_ID" ] && [ "$(cat "$dir/idProduct")" = "$MODEM_PRODUCT_ID" ]; then echo "$dir"; break; fi; done)
    if [ -z "$device_dir" ]; then logger -t "$LOG_TAG" -p daemon.err "Cannot reset: modem not found."; return 1; fi
    logger -t "$LOG_TAG" "Found modem at $device_dir. Sending reset command."
    echo 0 > "$device_dir/authorized"; sleep 2; echo 1 > "$device_dir/authorized"
    logger -t "$LOG_TAG" "Reset sent. Verifying modem reappearance within $HARD_RESET_VERIFY_TIMEOUT seconds..."
    local wait_time=0
    while [ "$wait_time" -lt "$HARD_RESET_VERIFY_TIMEOUT" ]; do
        if is_modem_present; then logger -t "$LOG_TAG" -p daemon.notice "SUCCESS: Modem reappeared."; return 0; fi
        sleep 5; wait_time=$((wait_time + 5))
    done
    logger -t "$LOG_TAG" -p daemon.err "FAILURE: Modem did not reappear after timeout."; return 1
}

# --- Main Monitoring Loop (with NEW logic) ---

while true; do
    if ping -c "$PING_COUNT" "$PING_HOST" >/dev/null 2>&1; then
        # --- Connection is OK ---
        if [ "$HARD_RESET_FAIL_COUNT" -gt 0 ]; then
            logger -t "$LOG_TAG" -p daemon.notice "Connection OK. Resetting hard reset failure counter from $HARD_RESET_FAIL_COUNT to 0."
            HARD_RESET_FAIL_COUNT=0
        fi
    else
        # --- Ping Failed ---
        logger -t "$LOG_TAG" -p daemon.warn "Ping to $PING_HOST failed. Analyzing state..."

        # Check if the interface hardware exists at all.
        if ip addr show "$INTERFACE" >/dev/null 2>&1; then
            # Interface EXISTS. Now check if it has an IP address.
            # The space after 'inet ' is important to distinguish from 'inet6'.
            if ip addr show "$INTERFACE" | grep -q 'inet '; then
                # --- LEVEL 1 (PATIENCE): Ping failed, but IP is present. ---
                logger -t "$LOG_TAG" "Interface has IP, but no connectivity. Likely temporary signal/carrier issue. Taking no action. Will re-check next cycle."
            else
                # --- LEVEL 2 (SOFT RESET): Interface exists, but has no IP. ---
                logger -t "$LOG_TAG" -p daemon.notice "Interface found but has no IP address. Performing a SOFT interface restart."
                ifdown "$INTERFACE"; sleep 5; ifup "$INTERFACE"
                logger -t "$LOG_TAG" "Interface restarted. Waiting for connection."
                sleep 30
            fi
        else
            # --- LEVEL 3 (HARD RESET): Interface hardware is GONE. ---
            logger -t "$LOG_TAG" -p daemon.err "FATAL: Interface $INTERFACE not found! Hardware may have glitched."
            
            if perform_hard_reset_and_verify; then
                logger -t "$LOG_TAG" "Waiting 60s after successful hard reset for network init..."
                sleep 60
            else
                HARD_RESET_FAIL_COUNT=$((HARD_RESET_FAIL_COUNT + 1))
                logger -t "$LOG_TAG" -p daemon.err "Hard reset cycle failed. Failure count: $HARD_RESET_FAIL_COUNT/$MAX_HARD_RESET_FAILS."
                if [ "$HARD_RESET_FAIL_COUNT" -ge "$MAX_HARD_RESET_FAILS" ]; then
                    logger -t "$LOG_TAG" -p daemon.crit "CRITICAL: Max hard reset failures reached. Rebooting system in 10 seconds!"
                    sleep 10 && reboot &
                    exit 0
                fi
            fi
        fi
    fi

    sleep "$SLEEP_INTERVAL"
done

    
`

lynxthecat, 2025-07-27:
Post above is worse than a dog’s breakfast. Fix it boy!

maskimthedog, 2025-07-27:
> I had the same issue with a quetel EC25 modem card , the wwan0 after sometime was falling down due to most likely signal loss and i asked gemini pro 2.5 to make a script that works on startup of [...]

Don't reboot the router.  Any good device uses a USB hub with per-port power control with the modem sourcing power from the port.  You should be able to make use of uhubctl (https://github.com/mvp/uhubctl) to find which port on which the modem resides and then power cycle it.

peterpt, 2025-07-27:
actually the gemini wanted to use usbreset tool but it was not available in repository .

"Post above is worse than a dog’s breakfast. Fix it boy!"  
really !!! it is working like a charm , i just need to place the router in the desert and forget about it and i can be sure i will be able to connect to it , the reboot only happens if the modem disappear from the system , and if disappear then how can you find it . openwrt devs should build the modem firmwares correctly , i had to go to the desert to pick up the router because it was disconnected from celular network and all i needed was a restart to interface , but since the desert is far away from me then i want to make sure this does not happen again .

maskimthedog, 2025-07-28:
> actually the gemini wanted to use usbreset tool but it was not available in repository .
>
> "Post above is worse than a dog’s breakfast. Fix it boy!" really !!! it is working like a charm , i just [...]

usbreset will not be enough, it will have no effect.  The modem needs to be power-cycled./reset.

maskimthedog, 2025-07-28:
> actually the gemini wanted to use usbreset tool but it was not available in repository .
>
> "Post above is worse than a dog’s breakfast. Fix it boy!" really !!! it is working like a charm , i just [...]

Well, if rebooting the whole router is acceptable then good for you.  Though it is heavy-handed to reboot the whole system just to recover a sub-system...it is only a router.

peterpt, 2025-08-03:
well me an google gemini made the definitive script in case anyone needs it , it is universal for all modems in openwrt , these changes were mostly because i had sim card missing when is reality it was inserted in modem , so we found a way to fix it without rebooting the router by reseting the modem using the AT commands in case it is necessary , here it is the link : 
[https://github.com/peterpt/modem_manager_whatchdog](url)